sinara-hw/sinara #246: 8 channel PA - enclosure

gkasprow, 2017-08-06:
I finished RF amplifier enclosure design.
I assumed that:
- RF amplifier module can be quickly exchanged - user must only unscrew 2 SMA connectors on front panel and remove metallic bar that fixes all modules. It is attached using two screws.
- Power switch is integrated with IEC inlet and available on rear panel
- cooling flow is front rear side to front side.
- main control channel is via SCPI commands over Ethernet
- USB connector for firmware upgrade (DFU) is available from front side
- each RF channel has 3 LEDs that indicate operation mode, failure or warning (i.e. too high input power)
- there are 2 user buttons - one is reset of warning/alert, second is not yet defined.

Below are some screenshots.
[image]
[image]
[image]

hartytp, 2017-08-07:
@gkasprow That looks really nice! Great work.

I'm very impressed that you got it into 2U. 

Only one question: when can I have one ;)

klickverbot, 2017-08-07:
> cooling flow is from rear side to front side.

Hmm, so the cold air intake is on the front and the hot air is expelled at the back? This seems backwards to most common IT 19" equipment (and indeed, also the Schroff microTCA crate).

gkasprow, 2017-08-07:
@klickverbot good catch. At the moment cold air intake is on the rear side and hot air exhaust is at the front side.
But we can easily change it by mounting fans in reverse way.
I will to airflow simulation to see how it behaves in reversed way.

jbqubit, 2017-08-07:
@klickverbot The cold air intake is on the front and the hot air is expelled at the back?

Hot air out the back is most common in test and measurement devices. Physics labs are not setup like data centers.

klickverbot, 2017-08-07:
@jbqubit: Hot air out the back is most common in data centers as well – apparently I made a typo in my earlier comment (fixed).

gkasprow, 2017-08-08:
@klickverbot The main issue in PA design is thermal resistance between amplifier IC and heatsink which is pretty high. So with 50 degrees on heatsink we get something like 70..80 degrees on amplifier IC under full load
The cooling system is designed with large margin to keep the fans noise low.
Since fans operation is critical, I implemented complex protection system that detects fan wear before failure occurs. Fans speed is adjustable and depends on amplifier temperature. We cannot really stabilise temperature inside the box because each channel may dissipate differently, but at least we will keep the temperature of hottest module within a limit. To compensate temperature drift of amplifier gain, active compensation is used. Each channel has dedicated digital thermometer and digitally variable bias voltage so to some extend we can stabilise its gain.

hartytp, 2017-08-08:
@gkasprow Sounds like a good plan. 

> The cooling system is designed with large margin to keep the fans noise low.

Indeed. The total power dissipated by the 8 channel unit at max load is only (4W/0.35)*8=90W, which isn't that high given the large heat sinks and fans you're using. So, I don't think that airflow will be our issue here...

> To compensate temperature drift of amplifier gain, active compensation is used. Each channel has dedicated digital thermometer and digitally variable bias voltage so to some extend we can stabilise its gain.

I saw that in the schematics you uploaded, which look really nice by the way.

closed 2017-08-10 by hartytp