sinara-hw/sinara #247: [Sayma AMC] JTAG USB/SCANSTA master

jordens, 2017-08-07:
@gkasprow I am working on the JTAG for Sayma AMC and would like to to have the AMC FPGA present on the USB/FTDI JTAG bus.

Please explain the logic behind the FTDI/SCASTA arrangement around `MPselB1/B0`. AFAICT If I enable `EN_USB_JTAG` to drive the JTAG bus (IC32), that (by accident?) also **disables** that port (`B0`) as the SCANSTA's JTAG master (makes it a slave) and selects the AMC JTAG (`B1`) as the master (!). That sounds wrong to me.

Should I rip something out of that `MPselB1/B0` to be able to drive the JTAG bus from USB?

gkasprow, 2017-08-07:
@jordens This part was not tested.
The JTAG switch has two master ports.
So either by inserting JTAG probe or enabling EN_USB_JTAG you select port 0. Otherwise port1 (from AMC connector) is selected. There is T1 MOSFET that inverts this signal.
And now all depends how the Open OCD/URJTAG/Vivado implements IO control of JTAG output switch and TRST line.
There are several resistor options that can select one of operating mode.
I found a few different JTAG probes schematics that realise those functionality. At the moment PORTA.7 enables JTAG probe and PORTA.4 resets the target and also JTAG switch.

gkasprow, 2017-08-07:
@jordens what utility do you use to access JTAG over USB? I can install it and try to debug it with hardware.

jordens, 2017-08-07:
Ignore that. Wrong thinking.

jordens, 2017-08-07:
`openocd`. Your PORTA.7/4 logic is ok. Just needs configuring in the openocd configuration.

closed 2017-08-07 by jordens

jordens, 2017-08-07:
@gkasprow works so far. But only at 5 MHz. Could you confirm for me which of R498, R328, R489 and R496 are 0R and which are DNP? I can't find them on the board and the BOM does not match the schematic (or the schematic label placement is mangled).

jordens, 2017-08-07:
Also. AFAICT the second SPI flash CS is connected at a weird location. I don't think the FPGA bitstream loading logic will work with that (if we use the second flash). `QSPI1_CS_B` should be connected to `IO_L2N_T0L_N3_FWE_FCS2_B_65` on `G26` (like on the KCU105). With the current configuration we'd have to do a terrible special case for Sayma if we want to use that flash with proxy bitstreams. Could you add that to your errata if you agree?

gkasprow, 2017-08-07:
@jordens 
R328 = 0R
R498 = 0R
R489 = DNP
R496 = DNP

gkasprow, 2017-08-07:
@jordens yes, we can fix the second SPI connection.
This can be fixed right now with piece of wire but you'd loose programming capaility of RTM FPGA.
Do you prefer to program RTM FPGA from Sayma FPGA or it is more convenient to have dedicated FLASH on RTM?

jordens, 2017-08-07:
Ah you are daisy chaining them... That forces us to build support for 32 bit flassh addresses anyway. I.e. we have the choice between (a) implementing 32 bit flash address support in openocd/jtagspi (and maybe maintaining a special proxy bitstream for sayma) or (b) change it in the next revision and imeplemnt some other custom way to configure the RTM FPGA. IMHO (a) is a bit easier. Then we can live with the current configuration. But with 32 bit addresses, there is also little use for the second flash and it can be DNPed.

Anyway. JTAG, the proxy bitstream and both flashes work. This is the openocd configuration

`sayma.cfg`:
```tcl
interface ftdi
ftdi_device_desc "Quad RS232-HS"
ftdi_vid_pid 0x0403 0x6011
ftdi_channel 0
# EN_USB_JTAG on ADBUS7: out, high
# nTRST on ADBUS4: out, high, but R46 is DNP
ftdi_layout_init 0x0098 0x008b
ftdi_layout_signal nSRST -data 0x0080
ftdi_layout_signal nTRST -data 0x0010
reset_config trst_and_srst srst_pulls_trst srst_gates_jtag srst_push_pull trst_push_pull

adapter_khz 5000

transport select jtag

set CHIP XCKU040
source [find cpld/xilinx-xcu.cfg]
source [find cpld/jtagspi.cfg]
```

And some template to play with it:

```tcl
set JTAGSPI_IR 0x02
# set JTAGSPI_IR 0x03
source [find sayma.cfg]
init
# exit
jtagspi_init 0 bscan_spi_xcku040_sayma.bit
flash read_bank 0 sayma_f0.bin 0 1024
# flash read_bank 0 sayma_f1.bin 0 1024
xcu_program xcu.tap
exit
```

gkasprow, 2017-08-07:
@jordens at the moment RTM FPGA is connected to AMC FPGA pins.
So you can access it with your own configuration IP core.
It can be daisy chained but it is very slow, that's why I switched to alternative mode.

jordens, 2017-08-07:
@gkasprow Ah. Then why can't we just swap `G26` and `K21`? Or better yet, have `QSPI1_CS_B` on `G26` and move `RTM_FPGA_CFG_PROGRAM_B` somewhere else?

Also. Any idea why JTAG through the MOS switches and SCANSTA doesn't work above 5 MHz? All the usual FTDI interfaces I have seen work fine with 25 or 30 MHz

jordens, 2017-08-07:
With the `ftdi_tdo_sample_edge falling` tweak this works fine up to 25 MHz.

gkasprow, 2017-08-07:
@jordens this is what I will do - just swap these two pins. I already added this issue.
There is some stuff on a way of JTAG data signal so falling edge selection seem to be good choice :)