sinara-hw/sinara #248: MMC Code, Documentation

jordens, 2017-08-07:
@gkasprow Couple questions on the MMC:

* Is the MMC software open? Where can I find it?
* What can it do? Is there a manual/documentation?
* What are the serial console parameters?
* Specifically, how do I control the SCANSTA stuff?

gkasprow, 2017-08-07:
@jordens 
At the moment board you have has very simple init software that just does power sequencing.
It is [here](https://www.dropbox.com/sh/0m9xfr4dmwnwus1/AADjA7aKJxEhOH-bQBXpkcxEa?dl=0)
Creotech is porting Open MMC to Sayma platform and this is open source MMC developed with a few partners from LNLS and GSI.
There is some documentation : https://github.com/lnls-dig/openMMC
Serial console : 115200, 8,n,1 but so far there is only limited output there.
SCANSTA control is automatic - you don't have to do anything.
When you unplug RTM, MMC detects it and reconfigures SCANSTA.
However if you want to play with advanced modes like scanbridge, it's necessary to modify the code for LPC.
Temporarily you can modify my demo code. Open MMC also has support for Scansta chip.

jordens, 2017-08-07:
Ah. That's what's meant by "stitching". Good. Works as well. Now that forces me to fix jordens/bscan_spi_bitstreams#1 ;)
Looking forward to seeing the Open MMC port! How does that relate to the [CERN MMC](https://espace.cern.ch/ph-dep-ESE-BE-uTCAEvaluationProject/MMC_project/default.aspx) ([SVN](https://svnweb.cern.ch/cern/wsvn/ph-ese/be/mmc_v2/Release_2))?

Otherwise, could I use the MMC to force-disable JTAG slaves (e.g. the AMC FPGA) from the JTAG chain to not have to work with multiple TAPs?

gkasprow, 2017-08-07:
@jordens disabling second JTAG is modification of one line of code.
I can send you bin file that does it if you like.

jbqubit, 2018-01-26:
Glad to see some progress on running OpenMMC on Sayma. cf https://github.com/m-labs/sinara/issues/475

jbqubit, 2018-02-07:
How's the OpenMMC port coming? I suspect even a rough version would be preferred by M-Labs over existing MMC code. cf https://github.com/m-labs/sinara/issues/507#issuecomment-363254368

jbqubit, 2018-02-07:
@gkasprow

gkasprow, 2018-02-07:
It works, it seems that no essential issues are left. 
It supports full hot plug for both AMC and RTM. Jakub implements UART diagnostics of Exar and PHY chip and should finish tomorrow.

jbqubit, 2018-02-15:
That's great progress! @sbourdeauducq Have you given the OpenMMC a try?

hartytp, 2018-05-03:
Closing this issue as it's stale. Please file new issues for each problem you find with the MMC firmware.

closed 2018-05-03 by hartytp