sinara-hw/sinara #250: Kasli schematic review

jordens, 2017-08-10:
* [ ] Consider making IC5 MAX6642ATT90+T DNP. There is the XADC which can do all that. It removes one BOM line. Provide option to GND DXP/DXN instead.

hartytp, 2017-08-10:
Some items to do with cleaning up the schematic:
- [ ] Please can you add a power budget, as we did with other boards?
- [ ] Please remove the annotation on page 7 about "With 4 extensions ..." this is annotation is obsolete.
- [ ] Can you add an annotation giving the max total 12V current for the EEMs. i.e. the max 12V current is 5A (limited by the barrel connector). So, how much current does Kasli itself use up/how much is left for the EEMs (total/per EEM)? What current are the IDCs/ribbon cables rated to?

jbqubit, 2017-08-11:
What happened to XC7A100T? See [mailing list](https://ssl.serverraum.org/lists-archive/artiq/2017-June/thread.html).

- [ ] TOP.sch clock input: clock impedance, frequency, amplitude range
- DRAM missing from TOP.sch

Is there no JTAG header or USB?

jordens, 2017-08-11:
@jbqubit Except for the amplitude range that's all done.

hartytp, 2017-08-30:
@marmeladapk Thanks for the good work on the schematic.

I really like the power budget @gkasprow did for the 3U_DAC.pdf schematic. In particular, a few changes we should make to the Kasli power budget in the schematic:
- [ ] Add total current consumption for the 12V rail
- [ ] Add power dissipation for each rail
- [ ] Can we have the component name to the left of the current consumption, please? This makes it a bit easier to read. It'd also be good to give a brief (1-2 words e.g. "RAM"/"UART"/"DDR termination" etc) description of each component to make it easier to follow
- What does "Limit: 1A + 200mA for LDO" mean? Is the current limit 1A or 1.2A?
Otherwise, it looks great!

jordens, 2017-08-31:
@marmeladapk great work!

* Nice idea with the I2C on FTDI. That's also great for playing around/testing and writing EEPROMs of EEMs.
* [ ] If there is space, expose the last port (DDBUS) of the FTDI as a DNP 100 mil pin header row.
* [ ] One of R43/R42 needs to be DNP. Or do you want this as a required decision and required modification for the prototypes?
* [ ] CK_FPGAIO_N also needs biasing. And all four resistors should be in the 10k-100k range according to [UG471](https://www.xilinx.com/support/documentation/user_guides/ug471_7Series_SelectIO.pdf#page=93).
* [ ] SW1 can be DNP, saves one BOM line.
* Mental note for testing: the unused negative outputs of IC1 and IC3 [may not need termination at all](http://www.ti.com/lit/an/snva745/snva745.pdf). That would save significant power.
* [ ] On TX_DISABLE, add DNP 10k pull-down on the SFP cage side of IC12 (for if we ever DNP IC12).
* [ ] On RATE_SELECT, add DNP 10k pull-up on the SFP cage side of IC12 (for if we ever DNP IC12).
* [ ] @hartytp where do we stand on AC coupling the shield of the coax (SMA front panel and MMCX in-crate)? Currently they are all grounded.
* [ ] When you let Altium swap EEM pins during routing, ensure that all pins of an EEM connector end up within the same bank of the FPGA.
* [ ] When routing, observe all the [intricacies of decoupling the MGTs](https://www.xilinx.com/support/documentation/user_guides/ug482_7Series_GTP_Transceivers.pdf#page=233)
* [ ] Make R9  51 Ohm instead of 49.9 to save a BOM line.
* [ ] You have SFP0.MOD_DEF0 (doesn't exist) on IO_0_34/T3 and SFP0.MOD_DEF1 nowhere.
* [ ] You have SFP1.MOD_DEF1 (connected twice) on IO_25_16/F21 and SFP1.MOD_DEF2 nowhere.
* [ ] Are you sure about the R40/R41 divider on the OSC1? Is that a 100 Ohm transmission line?
* [ ] Do we need to add a fuse on P12V0 as a weak link dimensioned to protect the copper traces from the EEMs? I know this will use a lot of space. So drop it (and let the 12V PS protect us) if that's problematic or if we can have 12V copper planes wide enough.

hartytp, 2017-08-31:
> where do we stand on AC coupling the shield of the coax (SMA front panel and MMCX in-crate)? Currently they are all grounded.

IMO, we shouldn't worry about isolating the grounds on these MMCX connectors. It shouldn't be necessary to get a pretty decent clock, and one can always use the clock buffer EEM to get ultra-low noise performance if necessary.

NB the intended use for these MMCX is distributing clocks over short distances within the rack. As a result, ground loops shouldn't be too much of an issue anyway.

jordens, 2017-08-31:
That's my thinking as well. But there are two smas on the front panel.

hartytp, 2017-08-31:
Comments about the front panel SMA:
- Isolating the grounds would be kind of nice here. However, it would mean adding insulating washers around the SMAs, which is quite a lot of faff. So, in this case I really don't think it's worth the hassle/expense.
- If one really needs to break ground loops, one can always use the 3U clock buffer EEM board as an isolating buffer. This is a cheap and easy solution for the rare cases that it's required.
- When Kasli is used as a DRTIO slave, I anticipate that one will almost always want to use the CDR clock directly, with no external PLL, and distributing it to other cards in the same chassis via the MMCX connectors. Thus, the SMAs will normally not be used at all for this mode. The CDR clock noise should be pretty low (e.g. comparable to AD9910 with PLL on) so it should be fine for most cases.

jordens, 2017-09-05:
I moved the remaining open items into #283 #284 #285 #286 #287 #288

closed 2017-09-05 by jordens

hartytp, 2017-09-08:
@marmeladapk Thanks for doing all that! New schematics look really lovely -- very clear and clean! Power budget is excellent and really helpful. Great work.