sinara-hw/sinara #251: 8 channel PA -- translation

jbqubit, 2017-08-11:
Small thing... Please translate Polish into English for 8 channel PA schematics. Otherwise I agree with @hartytp that they're very clear. 👍

hartytp, 2017-09-01:
@gkasprow When will the prototypes for this arrive?

hartytp, 2017-09-04:
@gkasprow List of suggested tests posted above. Let me know if that sounds okay, and if you can think of anything else/different we should do...

thomasRF, 2017-09-05:
> Assigned #251 <https://github.com/m-labs/sinara/issues/251>  to @thomasRF
> <https://github.com/thomasrf> .
>
> -
> <https://github.com/m-labs/sinara/issues/251#event-1233822114> on GitHub, or
> mute
> 4Dvfm9Iks5sfDA7gaJpZM4O0IaF> the thread.
> <https://github.com/notifications/beacon/AVx8a-idxFl6IIbpRqJNInTXZKtimlhRks5
> sfDA7gaJpZM4O0IaF.gif>

Grzegorzu,

 

Poniżej zawieram wstępne komentarze i uwagi do zaproponowanych testów.
Punkty 6,7, 10, 11 nie były brane pod uwagę przy specyfikowaniu wzmacniacza
oraz określeniu zadań przy realizacji projektu wzmacniacza.

 

 

1.	Power Efficiency

Ok

 

2.	Input/output S-parameters (small signal and at 1dB compression)

 

Nie ma czegoś takiego jak Input/output S-parameters jest macierz
rozproszenia [s],

Obawiam się, że posiadanym analizatorem obwodów, będzie problem z pomiarem
macierzy s dla 1dB, potrzeby do tego celu jest specjalny analizator obwodów.

Zbyt duża moc wyjściowa, i pomiar za pomocą Twego VNA będzie niedokładny.

 

3.	Output small-signal gain and P1dB point as a function of frequency

 

Ok

 

4.	 Output IP3 point as a function of frequency

 

Tom przy specyfikowaniu twierdził, że IP3 go nie interesuje, będzie zmieniać
częstotliwość oraz amplitudę na zasadzie on/off, a ważne było by sprawność
energetyczna była duża.

 

Jełki miałby być zmierzony IP3 to trzeba określić dla jakich warunków

 

5.	Time domain step response, 20/80% rising/falling time. Small signal
and at P1dB use 200MHz reference signal or another convenient frequency.

 

Ok.

Uwagi: Czy generator w labie, daje możliwość wygenerowania takiego sygnału?
JEśłi nie trzeba będzie pożyczyć.

 

 

6.	Time domain step response, overshoot. Small signal and at P1dB,
convenient reference frequency.

Wymaga doprecyzowania: step response, overshoot, convenient reference
frequency ?

 

7.	Only if easy to measure: phase noise for P1dB input signal at a
convenient frequency (we can measure this at Oxford if easier). NB this is
not part of the specification, but just good to know!

 

W ZUiAM mają przyrząd do pomiarów phase noise.

 

8.	Check temperature regulation, and thermal gain compensation.

Ok, podam ch-tę, wstawimy do cyfrowej pętli regulacji pkt. pracy, 
 
Uwagi: mamy komorę klimatyczną? Czy bierzemy lodówkę z grzałką? ;)

 

9.	Channel-channel cross-talk as a function of input frequency (again,
small signal and at P1dB)

Ok, to dla dwóch kanałów da się zrobić, ale przy produkcji, wykreślenie tych
ch-tyk to mnóstwo pomiarów. Proponuję w produkcji max. kilka pkt.
kontrolnych i test czy izolacja nie jest gorsza niż.

 

10.	SFDR/non-harmonic spurs

Uwagi: Jak w pkt. 4.

 

11.	Noise figure\far-from-carrier noise floor (again, not part of the
specification, but good to know)

W ZUiAM mają przyrząd do pomiarów NF, pytanie w jakim paśmie mogą mierzyć.
Niektóre analizatory widma maja opcję pomiarów NF, ale są to średniej
jakości pomiary.

 

 
Proszę o Twoje uwagi, to odpiszę na emalia datowanego na dzisaj od Toma.
 
Pozdrawiam,
Tomek

  _____

jbqubit, 2017-09-05:
@thomasRF Please use English. For now, using google translate...
> 8.	Check temperature regulation, and thermal gain compensation.
Remarks: we have a climate chamber? Do we take a refrigerator with a heater? ;)

Testing at ambient lab conditions should suffice.

Some additional test suggestions:
- [ ] Test that protection circuit shuts down individual channels in case of: input power overload; reverse power overload (as due to open/short).
- [ ] Measure delay between application of too-high input power and switch-off of output.
- [ ] Calibration of per-channel remote monitoring of forward and reverse power. Choose a couple representative frequencies.

hartytp, 2017-09-05:
@thomasRF @gkasprow Note that these are not intended to be production tests that are performed on each unit that is shipped, but rather methods of verifying the design of the prototypes.

gkasprow, 2017-09-05:
Sorry, it was internal communication and Thomas replied to all :)

gkasprow, 2017-10-18:
Status update: mechanics is ready for integration
[image]

hartytp, 2017-10-19:
Looks good @gkasprow! Can't wait to see it in action.

hartytp, 2017-12-20:
Thanks to @gkasprow and his team!

[image]

dhslichter, 2017-12-21:
@hartytp does the amps amping?  I can has gains plz?

hartytp, 2017-12-21:
> @hartytp does the amps amping? I can has gains plz?

I believe so. This particular Christmas Present is still in the post. @gkasprow has the data I believe.

gkasprow, 2017-12-21:
Yes, the channel isolation is better than planned. Gain flatness is also better. Tomasz is processing all measurement and promised to send them all to @hartytp

gkasprow, 2017-12-21:
more photos:
[image]
[image]

hartytp, 2017-12-21:
Ace!

hartytp, 2017-12-21:
Apart from the electrical tests above, we should check mechanical things like:
- [ ] Are the fans easy to get to (these will probably need replacing at some point, so it's nice if they can be accessed/changed without too much pain)
- [ ] SMAs easy to plug and unplug

Anything else?

hartytp, 2017-12-21:
One final thing, it would be nice to have an actual name for this board rather than just "RF PA". Thoughts? No Russian names, please? Looking for a bombastic name implying "loud" and fitting with the "er" naming scheme...

First attempt:
- Gainer

thomasRF, 2017-12-21:
Etna?

hartytp, 2017-12-21:
As in the active volcano? Points for bombasticness. Maybe a little overly destructive for lab equipment?

@dtcallcock something from the audio world?

thomasRF, 2017-12-21:
I thought about that as a source of power or vitality. By the way, probably the volcano is the place where Hephaestus had one of his forges.

dhslichter, 2017-12-21:
I prefer Etna to Gainer.  Some other "-er" suggestions

- Pumper
- Spider
- Booster

Some non-"er" suggestions:

- Falcon
- Torpedo
- Erebus ;)

gkasprow, 2017-12-21:
a few ideas for acronyms (http://acronymcreator.net/):

   MoCHA  Multi CHannel Amplifier
   MUsCA  MUlti Channel Amplifier
   HaIFA  HIgh Frequency Amplifier
   MoCAP  Multi Channel AmPlifier
   MODAL  MODular AmpLifier
   MaCAo  Multi Channel Amplifier
   MUsCAt  MUlti Channel Amplifier
   ReFRAME  Radio FRequency AMplifiEr
   MARiA  ModulAR Amplifier
   MICAM  MultI Channel AMplifier
  MICRON  MultI Channel RadiO frequeNcy
   MaCRON  Multi Channel RadiO frequeNcy
   MARADONA  ModulAr RADiO frequeNcy Amplifier
   MUsICALE  MUltI Channel AmpLifiEr
   REPERA  Radio frEquency PowER Amplifier
   MARQUA  ModulAr Radio freQUency Amplifier
   REPEAL  Radio frEquency PowEr AmpLifier
   MARINA  ModulAr RadIo frequeNcy Amplifier
   REPAIrER  Radio frEquency Power AmplifIER
   MURIRAI  ModUlar RadIo fRequency AmplIfier

dhslichter, 2017-12-21:
Definitely Maradona, eh English folks? ;)

hartytp, 2017-12-21:
> I thought about that as a source of power or vitality. By the way, probably the volcano is the place where Hephaestus had one of his forges.

👍 for the classical references.

> I prefer Etna to Gainer. Some other "-er" suggestions

I also half thought of "Wailer" (also cf Bob Marley) but maybe that sounds too much like Wailing in agony.

Booster I get and don't dislike (but don't love either, like gainer). Spider and pumper I don't get. Can you explain? I also don't get Falcon, Torpedo or Erebus. If we go down the semi-opaque route, the idea would be to have something memorable without too much explanation.

@gkasprow Will have a think on some of the acronyms ;)

thomasRF, 2017-12-23:
@hartytp Please look the preliminarny test results and if you like to comment or dicuss what would you like to achieve in the final performance of the RF PA. It is possible to adapt and little change the characteristics of the RF PA to your requirements. 

[Preliminary report.pdf](https://github.com/m-labs/sinara/files/1584143/Preliminary.report.pdf)

hartytp, 2017-12-24:
Tomek, thanks for posting that! That looks really good!

Questions/comments:
1. There are a few other tests we discussed for these amplifiers. Will you be able to do those in the New Year?
2. Which amplifier design is this? Is it the Macom chip?
3. Gain flatness is slightly worse than we originally agreed (we agreed +-1.5dB max). That's okay if it helps improve the design in other ways.
4. Power efficiency looks great! 40% at 4.4W! Is that wall socket to RF, including SMPSs etc?
5. Is the abscissa in figure 4 supposed to be "Input power (dBm)"? (Not frequency)
6. "It is possible to adapt and little change the characteristics of the RF PA to your requirements." What kind of changes are you thinking of? Broadly speaking, that looks fine to me, but if you have any particular changes to recommend then do let me know.
7. The cross-talk looks good here! Out of curiosity, do you have a feeling for what limits it? Is this limited by cross-talk on the power supply rails? Our of curiosity, are we confident that we're doing as much filtering etc as can be done easily to minimise the cross-talk? (e.g. the current number is fine, but if we can throw in a couple of extra ferrets/capacitors to improve it a bit then that would be great).

Also, are you in a position to give me a more accurate cost estimate for these amplifiers yet? e.g. What would it cost if we bought 5 8-channel amplifiers (40 channels) or 10 amplifiers (80 channels)? If you'd rather not disclose that publicly then feel free to email it to me.

Thanks again for all the hard work on this.

PS @dhslichter @dtcallcock @jbqubit @jordens and anyone else who is considering buying one of these, do let me know if you have any other comments/suggestions. Or, if you have anything you'd like to see us test/check with the prototypes.

We hope to place a large order for these towards the end of January, so let me know if you want to join in and buy some at the same time.

hartytp, 2017-12-24:
Okay, name-wise, let's go for "Booster". Anyone who cares, feel free to express an opinion with 👍 or 👎 on this post.

@thomasRF I like your suggestion, but "Booster" is a simple name that I think physicists will find easier to remember, and it fits in better with the other names used in this hardware family.

dhslichter, 2017-12-26:
> Spider and pumper I don't get. Can you explain? 
Doesn't need an explanation, needs to be reasonably memorable.  If no good, just ignore.  They are brainstorms :)

> I also don't get Falcon, Torpedo or Erebus. 
Again, just names.  Erebus because it's a volcano and so I was joking around.

> If we go down the semi-opaque route, the idea would be to have something memorable without too much explanation.
The whole point is no explanation, just something that people say "oh, that sounds like a name for a piece of equipment".  

I am fine with Booster for now but if someone comes up with a better name I'm not wedded to it.

dhslichter, 2017-12-26:
@thomasRF this looks very nice!  I would say that we are definitely interested, based on these initial data.  I have a few questions as well:

- Figure 2 looks more like the 2.5 dB compression point -- at the 1 dB compression point, you should see that the gain is 1dB less than the small-signal gain, right?
- Other characterizations that would be interesting to have (@hartytp may already have discussed these):
    - Noise figure
    - RF pulse response (in particular any slower time constant behavior e.g. from thermal effects)
    - Temperature coefficient of gain
    - Crosstalk as a function of channel distance -- in other words, is crosstalk worse for nearest neighbors in the case and better as you move farther away?  Or is it more uniform from any channel to any other channel?
- Is there data on how the amplifier handles poorly matched loads?  How bad a mismatch can the amplifier tolerate on its output, especially at high power?

hartytp, 2017-12-27:
> Edit: the gain shown in Fig. 3 looks to have a nontrivial deviation from linearity over a long range (from -20 dBm to -4 dBm input powers, for example). Is this an artifact or real? If real, can it be improved?

I would be interested to hear from @thomasRF on that, but I do believe this is real. I don't think it can be improved without significant reductions in power efficiency, which we wouldn't want to do for this design. 

Remember that the main target for this design is driving AOMs and other things like that, which are **far** less linear than this amplifier design.

> Other characterizations that would be interesting to have (@hartytp may already have discussed these):

Yes, those are on our list of things to test.

> Is there data on how the amplifier handles poorly matched loads? How bad a mismatch can the amplifier tolerate on its output, especially at high power?

@thomasRF Your spec sheet says that the amplifier is unconditionally stable. Is that correct? What is the maximum reverse power it can take. i.e. what settings are you using for the reverse power interlock?

hartytp, 2017-12-27:
> Edit: the gain shown in Fig. 3 looks to have a nontrivial deviation from linearity over a long range (from -20 dBm to -4 dBm input powers, for example). Is this an artifact or real? If real, can it be improved?

Anyway, as always, if there is an easy way of improving this without any significant trade-offs in terms of power efficiency etc then obviously we should do that. But, @thomasRF put a lot of work into the design and I do not believe that there are any real improvements to be had here without large sacrifices elsewhere, which is not worth it IMHO.

hartytp, 2017-12-27:
@gkasprow Can you add a few things to the schematic for the main board, please?

For the connectors that go to the RF modules:
- [ ] Connector pin-out with description of each pin (e.g. "Forward power (?V/W, differential)" or whatever)
- [ ] Power budget for the different power supply rails.
  - What limits the current on the supply rails? The supplies themselves? the d-types? the PCB routing (maximum tolerable voltage drop on the PCB?)? The overall max power dissipated in the chassis?
- [ ] Power sequencing

Can you also add an overall power budget, please?

dhslichter, 2017-12-27:
@hartytp ack to all.

dhslichter, 2017-12-27:
@thomasRF some errata for the schematics:

Preamp protection:
- [ ] Input SMA should appear where it's currently labeled just TP1.
- [ ] Part numbers are invalid/garbled for many ICs. Amps and power detector IC6, IC7, IC8, IC9 all have part numbers that don't seem to correspond to any real parts that Google knows about.  What are the actual part numbers for these?  
- [ ] CP2 needs a part number
- [ ] Notation should be made about the purpose and connectivity of the signals on J4 and J5, since they are routed to a separate board (PA_Channel_SUP_CTRL).
- [ ] Any critical inductors and capacitors in the amplification chain whose properties are important (particular voltage/current ratings, dielectric material for caps, core material for inductors, etc) should be called out.

Power stage:
- [ ] IC3 part number doesn't correspond to a part Google knows about (MMWX1)?
- [ ] Notations should be made for voltage/current/power ratings of passive components as appropriate.
- [ ] J3 should have a notation as to its purpose since it is connected to another board not in this schematic.

Output protection:
- [ ] CP1 needs a part number
- [ ] IC1, IC2 again have an invalid part number
- [ ] Notations need to be made about the purposes of the signals on J1, since they go to another board.

dhslichter, 2017-12-27:
> @thomasRF Your spec sheet says that the amplifier is unconditionally stable. Is that correct? What is the maximum reverse power it can take. i.e. what settings are you using for the reverse power interlock?

If this is the Macom amplifier (NPTB00004A), the datasheet specifies no damage at a 15:1 VSWR, all angles (https://cdn.macom.com/datasheets/NPTB00004A.pdf).

hartytp, 2017-12-27:
> If this is the Macom amplifier (NPTB00004A), the datasheet specifies no damage at a 15:1 VSWR, all angles (https://cdn.macom.com/datasheets/NPTB00004A.pdf).

Thanks for the reminder (10:1 was from memory). I'd guess that the loss in the output stage is enough to bring that close to unconditional stability (but @thomasRF will be able to confirm that).

@dhslichter Thanks for the other points. You can get all the part numbers from the Altium files (e.g. using the free Altium viewer). They are mainly correct IIRC. But, you're right that we should add annotation on the schematic so that they are visible in the printed/pdf version.

Could you either add those comments to the top post of this issue, or open another issue, please? Comments in the middle of the issue will probably get forgotten. Thanks!

---

Related point @thomasRF please can you check that all the part numbers in the allium file are correct? For example, when I looked at the Atlium file before, it looked like the capacitance labelled on the schematic didn't always agree with the actual component part number when I opened up the component properties in Altium. This looks like a simple copy and paste error, but it's important to get it right (AFAICT, this will affect the BOM etc generated from the Altium files). @gkasprow Can you help Tomek go over the schematics and do some of this tidying up, please?

dhslichter, 2017-12-27:
@hartytp will repost the above to a new issue.  But the issues I have with part numbers *are* using the Altium files -- I am looking at the schematics in Altium and the part numbers listed on the .SchDoc files are not valid part numbers.

dhslichter, 2017-12-27:
Also @hartytp I didn't see any PDF schematics anywhere.

hartytp, 2017-12-27:
> Also @hartytp I didn't see any PDF schematics anywhere.

Apologies, mis-remembered. Anyway, I was referring to the difference between looking at the annotations/labelling on the schematic file, and selecting an individual component, right-clicking and going into the "properties or whatever" menu. My recollection was that I easily found all the ICs when I looked at the part numbers in the properties tab.

cjbe, 2018-01-11:
@gkasprow after shipping the two channel prototype the channel cards worked themselves loose from the backplane (see pictures). It seems that the shelf holding the backplane as well as the PSUs worked its way loose and shifted about 10mm towards the back of the enclosure. 
[image]
[image]

jbqubit, 2018-01-11:
Simple solution would be to have end users install the modules themselves in the field.

cjbe, 2018-01-11:
@jbqubit that was my first thought, but the shelf holding the backplane and PSU will still move around in shipping, meaning the user has set this shelf back into position - this is non-trivial, as if it is positioned incorrectly the channel cards can have intermittent connections.

gkasprow, 2018-01-11:
I noticed that during assembly. I will get rid of adjustable slots and make tight holes to prevent such movements.

cjbe, 2018-01-11:
@gkasprow great. I also noticed an intermittent problem that I cannot properly reproduce that I think was down to this same issue:

The second channel sometimes did not fully turn on (i.e. did not come out of standby). The channel card itself was fine, as if I swapped the position of the two channel cards in the enclosure both worked. The problematic channel card sits slightly lower than the other channel card, so I suspect the two prototype channel card enclosures are slightly different in size, and that the backplane was poorly positioned by me, leading to slightly intermittent connection on the D9.

gkasprow, 2018-01-11:
@cjbe once we collect all issues, we will fix them in second chassis we have at WUT and ship it to you and you will ship yours to us.

hartytp, 2018-01-11:
@gkasprow Sounds good.

FWIW, I wondered if this would work better if the RF amplifier modules actually screwed into the D-type connectors on the control PCB.

gkasprow, 2018-01-11:
We can drill 2 holes through top cover but module replacement would be more problematic.

hartytp, 2018-01-11:
Well, do whatever you think is best, so long as its robust and doesn't have intermittent connection issues when used in a "typical lab environment".

thomasRF, 2018-01-16:
Sorry for my absence. I will try to answer in chronological order the question, starting from December 21st.

thomasRF, 2018-01-16:
@hartytp 

>Apart from the electrical tests above, we should check mechanical things like:
Are the fans easy to get to (these will probably need replacing at some point, so it's nice if they can be accessed/changed without too much pain)
 >- SMAs easy to plug and unplug
 >- Is field replacement of the amplifier modules reasonably easy?
 >- Are the SMAs robustly mounted?
>
>Anything else?

I’m responsible for the RF PA modules design, and I discussed with Greg the issues of RF PA modules exchange and of the functionality of using RF modules. In the final version we will try to provide ease of replacement and usage convenience of the RF PA units. All comments, suggestions are welcome.

hartytp, 2018-01-16:
@thomasRF Good to hear from you, glad you're feeling better!

> I’m responsible for the RF PA modules design, and I discussed with Greg the issues of RF PA modules exchange and of the functionality of using RF modules. In the final version we will try to provide ease of replacement and usage convenience of the RF PA units. All comments, suggestions are welcome.

Sounds good; this is more an issue for Greg than you, as you're focusing on the RF part, with Greg doing the mechanics etc.

We've finished our testing and posted some suggestions/comments about this. See the other issues.

thomasRF, 2018-01-17:
@hartytp 

>1.	There are a few other tests we discussed for these amplifiers. Will you be able to do those in the New Year?

Yes, I ‘ll be able to do the discussed additional test, however, the only RF PA that I’ve made are in your possession. if you want, please send it (RF PA modules) along with the enclosure of the unit, which will be adjusted, and then I’ll carry out additional tests.

>2.	Which amplifier design is this? Is it the Macom chip?

At the outset, I would like to point out that during the project I have made changes of the schematics diagrams, but the schematics diagrams at the GitHub diagrams are from a few months ago.

The RF PA amplifier has been designed in the three stages structure in two variants:

- based on the NPA1003QA transistor you’ve proposed in the final stage,
               NPA1003QA: https://cdn.macom.com/datasheets/NPA1003QA.pdf

- and in an additional option with NPTB00004A instead NPA1003QA (there is possible to achieve           
              slightly higher efficiency with slightly lower output power is possible.
             NPTB00004A: https://cdn.macom.com/datasheets/NPTB00004A.pdf

The RF PA amplifiers that I have made, which you have, consists of NPA1003QA transistors in the final stage, and the two preamplifier stages are built based on the MAAM-010373 and MMG3012NT1 transistors respectively. The stages are design to work with NPA1003QA. In the additional option with NPTB00004A instead of NPA1003QA, if the NPTB00004A transistor should be used, is required to change matching circuits in the preamplifier.

The version that will be manufacturing in near future is depend on your choice. You are welcome to comment, and I’ll develop one optimal version for you. 

The schematics diagrams I’ll update, after that we determine the final RF PA parameters, and you accept the test results so that I don’t need to generate subsequent versions of the documentation.

>3. Gain flatness is slightly worse than we originally agreed (we agreed +-1.5dB max). That's okay if it helps improve the design in other ways.

The presented RF PA solution is one of the versions possible to obtain with some small spread of parameters in the mass production. It is possible to change:
- gain, up or down to the value specified and indicated by you,
- linearity of the gain characteristics (with worsening matching at the input and output of the RF PA). It is bearing in mind that the impedance mismatch, especially at the output, it can be significant as to the expected value of the output power, i.e. the output power is depends on the load impedance and its matching to the amplifier, which depends on the frequency, where the amplifier output is matched to the 50ohms  ( Fig.2 - [preliminary report.pdf](https://github.com/m-labs/sinara/files/1639764/Preliminary.report.pdf)),
- output power.

Please indicate what to do, what changes should be made (please remember that improving one parameter may worsen the other)

>4.	Power efficiency looks great! 40% at 4.4W! Is that wall socket to RF, including SMPSs etc?

The power efficiency was estimated on the basis of RF input/output power measurements and power consumption at RF PA power terminals (i.e. for supply voltages of 8V and 28V). It does not take into account the SMPSs' efficiency. In addition, RF PA modules require power supply for protection, measurement and logic circuits, which are not power amplifier constituent circuits, however the additional blocks' power consumptions in relation to the amplifier are negligible.

>5.	Is the abscissa in figure 4 supposed to be "Input power (dBm)"? (Not frequency)

Yes, the corrected report: [preliminary report v1.1.pdf](https://github.com/m-labs/sinara/files/1639776/Preliminary.report.v1.1.pdf)

>6. It is possible to adapt and little change the characteristics of the RF PA to your requirements." What kind of changes are you thinking of? Broadly speaking, that looks fine to me, but if you have any particular changes to recommend then do let me know.

As I presented in the answer to question no. 3, the presented RF PA solution is one of the versions possible to achieve with some small spread of parameters in mass production. It is possible to change:
- gain, up or down to the value specified and indicated by you,
- linearity of the gain characteristics (with worsening matching at the input and output of the RF PA). It is bearing in mind that the impedance mismatch, especially at the output, it can be significant as to the expected value of the output power, i.e. the output power is depends on the load impedance and its matching to the amplifier, which depends on the frequency, where the amplifier output is matched to the 50ohms  ( Fig.2 - [preliminary report.pdf](https://github.com/m-labs/sinara/files/1639764/Preliminary.report.pdf)),
- output power.

>7.	The cross-talk looks good here! Out of curiosity, do you have a feeling for what limits it? Is this limited by cross-talk on the power supply rails? Our of curiosity, are we confident that we're doing as much filtering etc as can be done easily to minimise the cross-talk? (e.g. the current number is fine, but if we can throw in a couple of extra ferrets/capacitors to improve it a bit then that would be great). Edit: you're probably already doing all this, but all power + data pins on those D-types need CMCs + CLC (or capacitor-ferrite-capacitor) filters on both the RF module and control PCB side.

[... cut ...] e particulary) on the main board and the isolation of the power supply for the individual amplifier channels. In the part for which I am responsible, I can make these changes, so that in the next version of the amplifier, improve the inter-channel isolation, however, whether the overall effect will be satisfactory, depends on refilling RF PA modules and main board. I think that, I with Greg will deal with this issue. The question is what value of interchannel cross-talk would satisfy your needs.

hartytp, 2018-01-17:
@thomasRF 

> Yes, I ‘ll be able to do the discussed additional test, however, the only RF PA that I’ve made are in your possession. if you want, please send it (RF PA modules) along with the enclosure of the unit, which will be adjusted, and then I’ll carry out additional tests.

Don't worry about it, @cjbe has already done these tests and posted the results on GitHub. So, no need to do that.

> At the outset, I would like to point out that during the project I have made changes of the schematics diagrams, but the schematics diagrams at the GitHub diagrams are from a few months ago.

Okay, good to know.

> The RF PA amplifiers that I have made, which you have, consists of NPA1003QA transistors in the final stage, and the two preamplifier stages are built based on the MAAM-010373 and MMG3012NT1 [...]

I see, so the NPTB00004A version hasn't been tested or built yet.

Well, unless there is a significant improvement from changing the final stage amplifier, let's stick with what we have (no point spending a lot of time testing a new power amplifier stage if the improvement is only marginal). However, if you feel that using the other transistor could offer a significant improvement then please explain what the benefits/trade-offs would be.

> Please indicate what to do, what changes should be made (please remember that improving one parameter may worsen the other)

We are happy with the overall performance of the amplifier. If there are easy improvements you can make "for free" then please make them. But, it sounds like you've already done that and are at the point where any improvement in one parameter will make another parameter worse. If that's the case then let's leave the design how it is.

hartytp, 2018-01-17:
To be clear: we are very happy with the cross-talk and noise performance. It meets our specification and is okay.

However, if this can be improved easily -- and **without increasing the cost of manufacturing these amplifiers in a significant way or significantly increasing the design time** -- then we should do it, as less cross-talk/noise is always better.

What I would like you to do is to write your report which has the current design and recommendations for how it could be improved, including reducing the cross-talk. These recommendations should include the cost of any changes you propose to make. If the changes are cheap and quick (e.g. adding a bigger capacitor or a ferrite bead somewhere) then we will implement them. If they are expensive (like using a much higher specification power supply) then we will not.

Our goal here is just to double check that there are no quick, cheap, easy changes we can make to improve the performance (which is already very good!).

jbqubit, 2018-01-18:
@thomasRF Great documentation BTW. :)

thomasRF, 2018-01-24:
@dhslichter 

> @thomasRF this looks very nice! I would say that we are definitely interested, based on these initial data. I have a few questions as well:
> - Figure 2 looks more like the 2.5 dB compression point -- [...]

The Fig.3 I’ve plotted again in Excel (for the same data – [data_Ox1.xlsx](https://github.com/m-labs/sinara/files/1660688/data_Ox1.xlsx)), and 1dB compression point is for c.a. -2dBm input power. At the point the gain is decreasing by 1dB in relation to the ideal characteristic.

>- Edit: Fig. 3 looks like Psat is 36.5 dBm, not 37.5 dBm. Is this just amp-to-amp variability?

You're right, the correct value of Psat is 36.8dBm, and may vary slightly for individual RF. The Psat is 37.5dBm for the other operation point of the transistor in last stage.

>- Edit: the gain shown in Fig. 3 looks to have a nontrivial deviation from linearity over a long range (from -20 dBm to -4 dBm input powers, for example). Is this an artifact or real? If real, can it be improved?

This is a real measured characteristic, of course with a certain error. I can try to improve the gain characteristics, so that the variability of gain would be smaller, depending on the input power, however at the expense of impedance matching. 

Nowadays, the correction of the characteristics is realize by using a digital pre-distortion technique. If we know how the gain changes as a function of the input power and frequency, it is sufficient to process the input signal (ie in this case its amplitude characteristic), so that we get a signal in the output of the RF amplifier, as though gained by a linear amplifier. If a source of a signal is a generator, it is sufficient to prepare specific source code for the program the generator. The designed RF amplifier module can provide information about the power of the output signal, which can be useful in a given solution.

>Other characterizations that would be interesting to have (@hartytp may already have discussed these):
>- Noise figure
>- RF pulse response (in particular any slower time constant behavior e.g. from thermal effects)

@hartytp has written, that the tests, apart from the ones I presented, were made by @cjbe 

>- Temperature coefficient of gain

Here I want to point out that in the current version of the RF PA, the option of temperature compensation of the gain is not turned on. Changes are required in the part of Greg (@gkasprow) , and I also need to change the RF layout, due to the change in the position of the temperature sensor. However, measuring blocks of input forward power and output forward and reflected power are implemented, which enable software compensation of gain changes as a function of temperature.

>- Crosstalk as a function of channel distance -- in other words, is crosstalk worse for nearest neighbors in the case and better as you move farther away? Or is it more uniform from any channel to any other channel?

As I've written, in the first place crosstalk limitation is a circuitry issue. The distance between the RF modules is of secondary importance. The presented values of inter-channel crosstalk are for the worst case, i.e. between neighboring channels (RF modules), measurements were made in a configuration as in the photographs presented by Greg (@gkasprow).

>Is there data on how the amplifier handles poorly matched loads? How bad a mismatch can the amplifier tolerate on its output, especially at high power?

In the case of mismatch, the RF amplifier behavior is typical, i.e. the output power is reduced by the reflected power, where 50 ohms should be taken as the output impedance of the RF amplifier. If the mismatch is observed in the input of the amplifier, the forward (input) power will be reduced by the power reflected from the input (where Zin = 50 Ohms), there the output power will also be reduced. If you need more details about this effect, please let me know by email.

thomasRF, 2018-01-24:
@dhslichter 

>Is there data on how the amplifier handles poorly matched loads? How bad a mismatch can the amplifier tolerate on its output, especially at high power?

In addition, to protect the amplifier from damage, I’ve implemented the following three protection stage:
- stage No 1 (hardware protection) - the input (forward) power is measured, and if the given power is greater than -1.5dBm CW (i.e. 1.5dBm above the maximum expected value of the input power), the RF input power is immediately switched off. The value is fixed in hardware for CW power, I can change it, or set it for a specific signal (eg using a peak value detector), this value cannot be changed by the user.
- stage No 2 (hardware protection) - the output power (reflected) is measured, and if the given power is greater than 30dBm (1W) CW (which can be additionally dissipated on the power transistor in the last stage of the RF PA), then the RF input power is immediately switched off, regardless of the of the measured VSWR value. The value is fixed in hardware for CW power, I can change it, or set it for a specific signal (eg using a peak value detector), this value cannot be changed by the user.
- stopień 3 (zabezpieczenie software) - - dokonywany jest pomiar VSWR na wyjściu wzmacniacza, na podstawie mocy padającej i odbitej. Dane zabezpieczenie działa w ten sposób, że procesor może wyłączyć moc wejściową dla określonej wartości VSWR. Wartość przy której następuje wyłączenie mocy wejściowej określa użytkownik dokonując jej zdefiniowania prze dedykowana aplikację do monitorowania oraz kontroli wzmacniacza. Obecnie w układzie VSWR jest określanie na podstawie pomiaru mocy CW padającej i odbitej. Mogę dostosować by pomiar VSWR był dokonywany dla konkretnego typu sygnału (np. impulsowego). 
- stage No 3 (software protection) - VSWR is measured at the output of the RF amplifier, based on the forward and reflected power. The protection stage works in the way that the processor can switched off the RF input power for a specific VSWR value. The value at which the RF input power is switched off is determined by the user by defining it through a dedicated application for monitoring and controlling the RF amplifier. Currently, the VSWR is determining on the basis of the measured CW forward and reflected power incident and reflected in the RF PA circuits,. I can adjust that the VSWR measurement should be made for a specific type of signal (eg pulse).
In addition, the amplifiers are protected against overload by monitoring of the voltage and current of power supply, as well as temperatures of the individual RF PA modules and the FR PA unit.

All comments, suggestions are welcome.

thomasRF, 2018-01-24:
@dhslichter 

> @thomasRF some errata for the schematics:
>
> Preamp protection:
>
> - Input SMA should appear where it's currently labeled just TP1.
> - Part numbers are invalid/garbled for many ICs. Amps and power [...]
 Notation should be made about the purpose and connectivity of the signals on J4 and J5, since they are routed to a separate board (PA_Channel_SUP_CTRL).
 Any critical inductors and capacitors in the amplification chain whose properties are important (particular voltage/current ratings, dielectric material for caps, core material for inductors, etc) should be called out.
>
>Power stage:
>
>- IC3 part number doesn't correspond to a part Google knows about (MMWX1)?
>- Notations should be made for voltage/current/power ratings of passive components as appropriate.
 J3 should have a notation as to its purpose since it is connected to another board not in this schematic.
>
>Output protection:
>- CP1 needs a part number
>-  IC1, IC2 again have an invalid part number
>- Notations need to be made about the purposes of the signals on J1, since they go to another board.

Thank you for the comments.
The schematics diagrams I’ll update, after that we determine the final RF PA parameters, and you accept the test results so that I don’t need to generate subsequent versions of the documentation.

thomasRF, 2018-01-24:
@dhslichter @hartytp 

>If this is the Macom amplifier (NPTB00004A), the datasheet specifies no damage at a 15:1 VSWR, all angles (https://cdn.macom.com/datasheets/NPTB00004A.pdf).

It is correct at the datasheet of NPTB00004A specifies no damage at a 15:1 VSWR, and for NPA1003QA specifies no damage at a 10:1 VSWR, however I suggested to protect the  RF PA by limiting acceptable reflected power to 30dBm (1W) CW, regardless VSWR value.

hartytp, 2018-01-24:
@thomasRF thank you for the detailed response.s

> This is a real measured characteristic, of course with a certain error. I can try to improve the gain characteristics, so that the variability of gain would be smaller, depending on the input power, however at the expense of impedance matching.

I wouldn't want to make the impedance matching significantly worse to improve the linearity. My view is that the current level of linearity is okay for our purposes (it's better than an AOM!).

> Nowadays, the correction of the characteristics is realize by using a digital pre-distortion technique. If we know how the gain changes as a function of the input power and frequency, it is [...]

Good point. We can implement that later on in software with no hardware changes if we want so let's not worry about it for now. However, FWIW, this may not work so well for most of our use cases because the signal bandwidth is quite high.

> stage No 1 (hardware protection) - the input (forward) power is measured, and if the given power is greater than -1.5dBm CW (i.e. 1.5dBm above the maximum expected value of the input power), the RF [...]

> stage No 2 (hardware protection) - the output power (reflected) is measured, and if the given power is greater than 30dBm (1W) CW (which can be additionally dissipated on the power transistor in the [...]

> In addition, the amplifiers are protected against overload by monitoring of the voltage and current of power supply, as well as temperatures of the individual RF PA modules and the FR PA unit.

That sounds like a very good idea. This ensures that the amplifier itself cannot be damaged from an excessive input power or over temeprature etc! (NB this stage mainly protects the amplifier, not the load).

> stage No 3 (software protection) - VSWR is measured at the output of the RF amplifier, based on the forward and reflected power. The protection stage works in the way that the processor can switched [...]

From a hardware perspective, that sounds great. We can discuss the software later on.

FWIW though, I do not think we need to have a software interlock for the VSWR. The amplifier itself is protected from reflected power by the hardware interlock in stage 2 and VSWR will not damage the load. It is useful to be able to monitor the output VSWR via software though, even if it's not interlocked.

AFAICS, the only use for the software interlock is to protect the load from excessive input power, so the software just needs to ensure that the max output power never exceeds a programmable threshold.

hartytp, 2018-01-24:
> It is correct at the datasheet of NPTB00004A specifies no damage at a 15:1 VSWR, and for NPA1003QA specifies no damage at a 10:1 VSWR, however I suggested to protect the RF PA by limiting acceptable reflected power to 30dBm (1W) CW, regardless VSWR value.

Sounds like a good idea!

[22 more replies]