sinara-hw/sinara #254: consistent board names

sbourdeauducq, 2017-08-17:
Sinara boards should use the same naming scheme everywhere. We need names for:

* VHDCI Carrier
* BNC DIO breakout
* SMA DIO breakout
* RJ45 LVDS breakout
* RF PA
* Medium speed DAC (#253)

hartytp, 2017-08-17:
Okay, we can give these names as well. But, ...

FWIW, most of the conversations I have with people about ARTIQ hardware still follow the lines of "Which board was that again? Oh, you mean the 3U DAC board/smart arb board/etc". From an empirical study of current/potential users: there is a lot of confusion about how to pronounce these names; people can't remember them/find them confusing; these names aren't used, and instead people refer to them by form-factor and functionality....

jordens, 2017-08-17:
We have discussed this at length before. I don't see an alternative. Whether you are buying a laptop, an EMCCD camera, a car, or a washing machine, there will always be opaque names or numbers. Are those any different? Encoding metadata in those names is failing at every turn for a non-trivial bpard. We already have three different DAC designs with one potentially being called "medium speed DAC" informally...
The breakout boards and the VHDCI carrier are IMHO trivial enough (they are 1:1 line drivers) that there is no doubt about functionality and no ambiguous meaning. I would make an exception for those. Also because BNC, SMA, RJ45, VHDCI are already opaque names.

sbourdeauducq, 2017-08-18:
How about:
RF PA -> Larino
medium speed DAC -> Kasargi

jordens, 2017-08-18:
Maybe there is a middle ground with semi-opaque names like for phaser or nu-servo. 

Medium speed dac: **Shuttle(r)**
RF PA: **Valve** (or stick with RF pa)

dhslichter, 2017-08-18:
At the risk of revisiting the very long and well-worn discussion we had before, I do vote for something only semi-opaque as Robert mentions.  Things like "PDQ" and "phaser" are opaque but seem for most people to be easier to remember than a lot of the random Russian city names that have been chosen for the other boards -- and this seems to hold for non-native English speakers too in my discussions.  I can't keep Zotino and Novogorny straight.  I like the idea of "Shuttler" and "Valve"; carefully chosen names are important.  Yes, commercial products use opaque names, but they are chosen extremely carefully to evoke associations in the customer.  I also would argue that since English is the lingua franca of ARTIQ/Sinara users and developers (recognizing that for many it is not their native language!), choosing English or English-seeming words as board labels is more likely to allow people to remember/associate than the current names.  

I am happy keeping the names SMA/BNC/RJ45/VHDCI, although I point out that they are definitely NOT "opaque" names, they refer to a specific type of connector, and if someone makes a new board breaking out to SMA in a different way, for example, you would have a potential name collision (or at least confusion).  This was one of the points that was belabored a great deal in our previous discussions.  In this case, the solution might be to append a number (e.g. SMA2) for a second-generation design.  Since I think it's unlikely there will be many (if any) second generation designs for these boards, it's probably fine to leave them as they are.

sbourdeauducq, 2017-08-19:
"Valve" isn't very differentiated. How about "Octovalve"?

jordens, 2017-08-19:
Not being very differentiated is the idea and als means not encoding metadata (like channel number). That thing will run just fine with 1-7 channels.

dhslichter, 2017-08-21:
I have tentatively named the 12 GSPS superconducting-qubit-oriented RTM card for which there is no funding/spec yet as "Blaster", just as a placeholder.  If there are strong objections to this nomenclature, please say so.

I would like, if we can, to rename some of the existing Eurocard boards (in particular, Zotino and Novogorny, potentially Urukul as well), along the lines of what has been discussed above.  If we are going to do this, it should be done ASAP before production boards are made.  May I suggest as a couple of options (if you don't like these, please suggest your own!  This is more intended to get the conversation going):

Zotino -> Beehive, Metro, or ???
Novogorny -> Spider, Rabbit, or ???
Urukul -> Dancer, Violin, or ???

sbourdeauducq, 2017-08-21:
Let's not rename those.

dhslichter, 2017-08-21:
I don't think Novogorny has really caught on as a name with anyone, and Zotino hasn't either -- even @gkasprow and his team call them the 3U ADC and 3U DAC.  Changing now, it would be relatively easy to go back to modify the appropriate names in design files/issues (perhaps with a slash so the old and new names are both there).  I am NOT proposing we change the names of Sayma, Kasli, or Metlino, all of which have had MUCH more written about them, and seem to be easier to remember and work with.

sbourdeauducq, 2017-08-21:
There are already enough problems with board names (and there are the old/fast Novogorny ...). Let's not change those.

dhslichter, 2017-08-21:
I will also point out that if you want people to buy in to ARTIQ, having names for equipment which they can't remember/have trouble with for whatever reason (even if that seems in the abstract to be an "invalid" reason), it WILL have a real effect on whether people go for it or not.

sbourdeauducq, 2017-08-21:
Ikea and their weird Swedish names is probably the best counter-example.

dhslichter, 2017-08-21:
> There are already enough problems with board names (and there are the old/fast Novogorny ...). Let's not change those.

Agreed, but I feel like this is because we either went with the Russian town names style or the "name it what it does" style, both of which seem to be troublesome.  My point is that it is painful to rename, but considerably less painful if we do it now than if we wait any longer, and that the "semi-opaque" a la phaser/shuttler/nu-servo/pdq seems to work much better for most people.

dhslichter, 2017-08-21:
> Ikea and their weird Swedish names is probably the best counter-example.

Point taken, but a) those names actually mean something in Swedish and b) for a lot of non-Swedish speakers, some of those names are really really confusing.  What sells a lot?  Poang.  Malm.  Billy.  Very short and sweet.  In the same way, I think that Sayma and Kasli are OK, but Novogorny is not (and Zotino is a bit marginal, but not as bad).

sbourdeauducq, 2017-08-21:
nu-servo stands for **N**ovogorny **U**rukul **-servo**, which is another argument against the renaming of Novogorny and Urukul.

dhslichter, 2017-08-21:
Yes, but again this is one more relatively new/minor nomenclature that could itself be renamed, or one could simple choose new names starting with the same letter.  Or just shorten "Novogorny" to "Novo", even.

sbourdeauducq, 2017-08-21:
Here are a few other random Ikea product names: RASKMÖLLE, LAPPLJUNG RUTA, EIVOR CIRKEL, SILKEBORG, AVSIKTLIG.
Novogorny isn't worse than that.

dhslichter, 2017-08-21:
> Here are a few other random Ikea product names: RASKMÖLLE, LAPPLJUNG RUTA, EIVOR CIRKEL, SILKEBORG, AVSIKTLIG.
Novogorny isn't worse than that.

No, it's not.  But those are _**horrible**_ product names.  Unless you're Swedish, they are a disaster to try to remember (and maybe even then they still are).  Let's aim a bit above the lowest common denominator is my point :)

klickverbot, 2017-08-22:
Everybody also talks about "that white Ikea couch/shelf/…" rather than the actual product name. Only a few classics like the BILLY bookcase stand out.

jordens, 2017-08-22:
Are there any examples where non-opaque names work? Not a single car, not a single laptop, not a single washing machine, not a single fighter jet or airplane has a non-opaque name. Just deal with it. The prevalence of opaque names to me is evidence enough that there is no other useful solution.
Cases where non-opaque or semi-opaque naming schemes with varying levels of metadata in them are merely amusing for the reasons cited above are aplenty. Take telescope names for a nice afternoon read.

jordens, 2017-08-22:
And how does anyone remember what a "Hamiltonian" or a "Laplacian" is? A few minutes of learning hardware names are an acceptable investment.

dtcallcock, 2017-08-22:
Factory Records even gave their cat an opaque name: https://en.wikipedia.org/wiki/Factory_Records_discography

dhslichter, 2017-08-22:
@jordens I am _**not**_ trying to argue for a non-opaque name here.  I am trying to argue for one that is more memorable, and shorter, than "Novogorny" -- thus the suggestion of "Novo", for example.  I am further trying to argue that the current method of picking opaque names, namely "choose a random Russian town", has so far provided names which have been of widely varying degrees of successful adoption among a number of the people working with them every day.  This is a plea at the very least to choose shorter/simpler Russian town names such as "Sayma" or "Kasli", and give space for (and listen to) feedback on a proposed name for a device before settling on it and charging ahead.  "One might suspect that the failure or unwillingness to do so has other reasons," as you put it yourself.  Changing the name of a piece of hardware which has already had it for a long time is another matter, with many additional complications, so there I am much more able to understand the desire not to change it (although I floated the idea, because I think the benefits would outweigh the costs for Novogorny, at least).  

On the business/adoption side, almost every discussion I've had about the Sinara ecosystem with people other than @sbourdeauducq and @jordens has included some unsolicited commentary from the person I'm talking to about "what is the deal with all the weird names?"  I am NOT suggesting that we try to name things after what they do (this is clearly bad, agreed!), just that the opaque names be chosen to as to generate less confusion/apprehension among potential adopters of the system, as well as the designers of the system.  I find myself on the defensive at the start of so many conversations about ARTIQ and Sinara because of this, and as dumb as it is that people don't see past this, the fact is that a lot of them don't, and that is not what we want.  If the source of the issue is that these words are unfamiliar or difficult to pronounce for non-Russian speakers (because I have had this reaction from native speakers of a variety of languages, not just English), I don't think that alters the validity of the fact that it puts an unnecessary obstacle in place to adoption for some potential users.

I think that Shuttler, Valve, Sayma, and Kasli are all fine examples of opaque names for hardware; if I had to pick the common thread in these names, I would say that they are pretty short, and that as a native English speaker I have a preference for English words in terms of what I personally find easy to remember.  Neither Sayma nor Kasli is English or particularly English-sounding to me, but they generally have a nice "ring" to them for me and they seem to contain sounds which people speaking different language also find easy to say/remember.  See the comment from @klickverbot above about BILLY bookshelves, if you need to be convinced that these perceptions of what makes a good, memorable opaque name are not just confined to native English speakers.

Anyway, at the end of this we are back where we have been after previous iterations of this argument, namely an internet shouting match.  What I'd like to achieve from this, this time around, would be to change to a process whereby when a piece of hardware is being named, it is done by proposing a set of names as an issue and getting some "market research" feedback from others working on the Sinara project before charging ahead.  This has seemed to work OK for Shuttler and Valve just now.

sbourdeauducq, 2017-08-22:
Is there something wrong with Larino?

dhslichter, 2017-08-22:
Personally, I prefer Valve.  I find that Larino and Zotino and Metlino all rhyme, and so I am more prone to confuse them.  "Valve" doesn't sounds like any of the other board names, while also being relatively short and simpler.  There are many, many other possible names that would work as well if for some reason "Valve" is not good in the opinion of others.

gkasprow, 2017-08-24:
what about using names which mean something, like names of CERN
experiments? Alice, Atlas, Isolde are the examples.
There is a webpage that helps generating such acronymes:
http://acronymcreator.net/ It's quite powerfull tool, where many variants
of every word can be used.
Examples: SADE: Slow Analog to Digital conveter in Eurocard format
ETIC: Eurocard Real Time Controller
FONDA: Fast Octal Digital to Analog converter
or DANCE   Digital ANalog ConvErter
or FANCy   Fast ANalog Converter
I had some fun long time ago when was trying to find suitable name for hard
real time control system. Among acronyms the web poposed were: HETMAN and
ORGASM.
Since Hetman means something historical in polish (military grade), I kept
this name.
I propose playing with this or similar tool, choose some names and then
vote which ones to finally choose.
The front panels are not yet manufactured and I can postpone it by a few
days and place these names on them.

>

sbourdeauducq, 2017-08-24:
Yes, that's what many research groups are doing. Let's change this a bit. Russian names are good.

klickverbot, 2017-08-24:
It's just silly to use Russian names if that's not even the native language of a significant part of the contributor/potential user base (and the transliteration into latin script is not even unique).

hartytp, 2017-08-24:
> Russian names are good.

...??? Because no one in the community speaks Russian or has any intuitive feeling for the spelling or pronunciation of Russian words (making them hard to remember or talk about)?

dhslichter, 2017-08-24:
@gkasprow this is another fine way to choose an opaque name.  I really don't care how people end up coming up with names -- acronyms, words from some language, Russian towns, etc -- as long as:

1. The name is pretty short.
2. The name is opaque or semi-opaque -- doesn't try to encode restrictive metadata (e.g. "FastDAC" is a bad name).  
3. The name is easy to pronounce and remember for those involved with Sinara, to be judged by the people involved with Sinara.

sbourdeauducq, 2017-08-24:
> no one in the community speaks Russian

At least three M-Labs employees/contractors do.

hartytp, 2017-08-24:
> At least three M-Labs employees/contractors do.

Good to know, but still: none of the current/prospective AOM users speak Russian, so Russian names don't help them (us!).

hartytp, 2017-09-03:
Is this complete? Consensus seems to be to name the DIO boards using the pattern BNC_DIO_EEM etc. 

Unless anyone objects, I'll close this issue...

closed 2017-09-04 by hartytp