sinara-hw/sinara #255: review FTDI chip USB signals

sbourdeauducq, 2017-08-18:
The Sayma card seems unusually picky about what USB cables it will accept, and which ones will cause the device not to be detected or get detected and then disconnected immediately.

jbqubit, 2017-12-14:
Has this been explored further? @sbourdeauducq What did you see that suggests this is a cable issue and not something else? With a single USB cable I sometimes see strange behavior #872 and sometimes not.

gkasprow, 2017-12-14:
@sbourdeauducq if you observe MMC reset which causes full power cycle, this is probably caused by embedded MMC FLASH programmer. You can disable this feature by removing a jumper that connects Source of the T23 is missing GND. So I placed a piece of wire that connects pin 2 of T24 with pin 2 of T24.  But still, the capacitance between D and G of T23 may cause resets, so it's better to lift pin 3 of T23 and isolate it.

jbqubit, 2017-12-14:
> So I placed a piece of wire that connects pin 2 of T24 with pin 2 of T24. 

Is that a typo?

gkasprow, 2017-12-14:
edit: T23 with T24 :)

gkasprow, 2017-12-14:
Here is drawing which will help
[image]

hartytp, 2018-05-03:
No action points here. This doesn't add anything to the other issues discussing USB JTAG.

closed 2018-05-03 by hartytp