sinara-hw/sinara #262: Rename Novogorny?

dhslichter, 2017-08-23:
This is an issue to discuss whether or not Novogorny should be renamed, if we are too far along with the name to do so or not, and if we do end up renaming it, what the new name should be.  

I propose, as an initial place for discussion, that we simply shorten the name to "Novo".  This seems to me to be a short, memorable name (others should give their opinions), and one that allows for minimal adjustment of names in issues/Wiki/etc -- in all, more of a middle ground solution.  

I would point out that in the actual code, there is still no reference to "Novogorny" (or "Urukul" or "Zotino", for that matter), it is still referred to as "3U_ADC", which is a placeholder nomenclature already, so no matter what option we choose, it will be basically the same amount of work in terms of updating the schematics/silkscreens/etc.  Changes would need to be made to the Wiki and potentially to issues if the name were changed.  

Looking for comments on the options below from all involved parties, so we can get a sense of what path most people would prefer @gkasprow @marmeladapk @WeiDaZhang @sbourdeauducq @jordens @hartytp @jbqubit @cjbe @klickverbot @jmizrahi @dtcallcock, and please pass to others I have forgotten in this list:

1. Keep the name "Novogorny".
2. Shorten the name to "Novo".
3. Choose an entirely different name (please make a suggestion if so).

sbourdeauducq, 2017-08-23:
1

sjmackenzie, 2017-08-23:
1

michallgaska, 2017-08-23:
1

dhslichter, 2017-08-23:
2

sjmackenzie, 2017-08-23:
How about navogamy?

klickverbot, 2017-08-23:
2 (= 3 > 1)

sjmackenzie, 2017-08-23:
My proposed name has a deeper more funny meaning and isn't meant to be taken seriously. :-)

dhslichter, 2017-08-23:
Nine wives seems like eight too many, @sjmackenzie  :)

jordens, 2017-08-24:
[Refreshingly](https://en.wikipedia.org/wiki/Condorcet_method) [distracting](https://en.wikipedia.org/wiki/Schulze_method).
1 > 3a(Meter) > 2

gkasprow, 2017-08-24:
2

dtcallcock, 2017-08-24:
3 - Spud

klickverbot, 2017-08-24:
(@jordens: Also check out [Arrow's impossibility theorem](https://en.wikipedia.org/wiki/Arrow%27s_impossibility_theorem)).

dhslichter, 2017-08-24:
@hartytp @marmeladapk @jbqubit @cjbe @jmizrahi what are your votes here?

cjbe, 2017-08-24:
2,3 >> 1

hartytp, 2017-08-24:
2 (3 if someone else comes up with a decent name).

jordens, 2017-08-24:
I think we have to yield to Tom/Oxford here. If they are ok with a free for all vote then I'll run the SSD/condorcet stuff setting a deadline at 13:00 UTC (Friday). You can edit your vote until then, with preference, and currently we have 3a: Meter and 3b:Spud (@dtcallcock what's the background ?  ;)

hartytp, 2017-08-24:
> I think we have to yield to Tom/Oxford here.

Thanks Robert. As I said, I don't have a strong personal/idealogical objection to the current board names. It's just that I've found that they have actively hindered my discussions with current/prospective Sinara users by causing confusion (far too much "wait, which board was that again? Oh, you mean the one with the funny name that I couldn't remember/pronounce?")

Within reason, I'd be happy with any name that people can remember and will actually user (rather than reverting to "Oh, the Eurocard ADC board." etc every time).

Vote is fine for me.

dtcallcock, 2017-08-25:
> 3b:Spud (@dtcallcock what's the background ?

Super Potential Undulation Detector

sbourdeauducq, 2017-08-26:
Since there are two Novogornys (the current one, which likely will end up unsupported, and a newer one with a faster DAC chip) we could also call the current one Novogorny and the new one Spud.

jordens, 2017-08-27:
```
Rename Novogorny
https://github.com/m-labs/sinara/issues/262

A. Novogorny
B. Novo
C. Meter
D. Spud
E. Larino

V: 14-32 sbourdeauducq
V: 1---- sjmackenzie
V: 1---- michallgaska
V: 21111 dhslichter
V: 21111 klickverbot
V: 132-- jordens
V: -1--- gkasprow
V: ---1- dtcallcock
V: -1222 cjbe
V: -1222 hartytp

        A    B    C    D    X
   A    -    4    4    4    4
   B    5    -    4    4    4
   C    4    1    -    1    1
   D    5    2    2    -    1
   X    4    1    1    1    -

option A eliminated (B beats A but A does not beat B)
option C eliminated (B beats C but C does not beat B)
option D eliminated (B beats D but D does not beat B)
option X eliminated (B beats X but X does not beat B)
the Schwartz set is { B }

result: option B wins
```

Novo it is.

jordens, 2017-08-27:
But I am also happy to go with one of C,D,E for the "fast/current" Novogorny. Or even do another vote. I am already looking forward to all those people complaining about Novo being hard to remember. ;)

jordens, 2017-08-28:
https://github.com/m-labs/sinara/wiki/Novo

closed 2017-08-28 by jordens