sinara-hw/sinara #263: stress on AFE from SMA

jbqubit, 2017-08-24:
SMA barrel on AFEs is long enough that the threads extend to the back side of the RTM panel. If user
tightens nut on SMA barrel, SMPs and DIO header are stressed. We should advise users to /not/ use SMA nuts. 

Attachment of a SMA cable will put stress on SMPs and DIO header. The stress on the connectors is reduced if standoffs are used to secure the AFEs to the AMC_RTM. We should advise use of a SMA torque wrench. It looks like 8 in-lbs is [standard ](https://www.pasternack.com/preset-torque-wrench-5-16-bit-sma-2.92mm-3.5mm-8in-lbs-pe5011-1-p.aspx) torque. 

- [ ] @gkasprow Please include these standoffs (and screws) in the BOM. Add same for clock mezzanine too. These standoffs should be installed by default at the factory. 

@filipswit please update manual to 
- [ ] advise users to use standoffs and not use SMA nuts
- [ ] advise users as to recommended SMA torque 
- [ ] include standoff and screws in list of user-replaceable parts

dhslichter, 2017-08-24:
For the time being, is there enough room to put an SMA nut on the inside of the panel as well as outside?  This is a standard way of addressing this issue.  In the long term having mechanical standoffs connecting RTM to AFE cards sounds good, if there is space.

jordens, 2017-08-24:
there need to be nuts on the inside in any case for the longer connectors because of length and "flush-ness"
.

gkasprow, 2017-08-30:
@dhslichter @jordens @jbqubit  the RTM is missing machined plastic panel which takes care about SMA isolation and  positioning of SMAs in the panel holes. It also reduces stress on AFE. We planned to use dedicated isolation washers but their assembly would be painful for so many SMAs.

dhslichter, 2017-08-30:
@gkasprow ack, is this documented in the wiki or in the repository?

gkasprow, 2017-08-30:
@dhslichter Design of such panel is on a way. I was waiting for MTCA panels to check how it fits. There was discussion about it but I cannot find it right now.

gkasprow, 2017-08-31:
@dhslichter design is ready. 
[image]
[image]

gkasprow, 2017-08-31:
[here](https://github.com/m-labs/sinara/tree/master/MECH/PANELS) are the files

jbqubit, 2017-08-31:
@gkasprow Is there room for standoff’s between AFEs and RTM? There are holes on both PCBs to support this. Some ICs look very close to where standoffs would attach in the current layout...

gkasprow, 2017-08-31:
Yes, each AFE is positioned using 4 standoffs.
The holes are 2.55mm, pad is 4.55mm in diameter.
we will use these standoffs
http://www.farnell.com/datasheets/1231941.pdf?_ga=2.141285745.271167733.1504158512-162313899.1464604133

jbqubit, 2017-08-31:
I looked to check if the standoffs are in the Allaki AFE BOM but can't find a BOM. Doesn't look like there are BOMs for the EEM designs but not Sayma-related designs.

gkasprow, 2017-09-18:
@jbqubit I added the BOM files for Sayma/Metlino boards

jbqubit, 2017-09-19:
@gkasprow Thanks for adding the BOMs. :)

hartytp, 2018-05-03:
Not relevant now that we're planning to move to an FMC form-factor for the next version (if the AFE has its own FP, this isn't an issue anymore).

closed 2018-05-03 by hartytp