sinara-hw/sinara #265: Clock distribution EEM

hartytp, 2017-08-28:
Proposal for clock distribution EEM (request for comments).

- 1 SMA clock input on front panel
- At least 4 SMA clock outputs on the front panel
- Low noise LVPECL clock buffer (no mux required). Probably an ADCLK946 or ADCLK950 depending on the number of SMAs we can conveniently fit on the front panel while leaving room for an SMA spanner.
- If the clock buffer has spare outputs, add a few MMCX connectors on the PCB (not front panel) for distribution to other EEMs in the same rack
- All SMAs have isolated grounds: use same washers as Sayma to isolate grounds from front-panel; connect to circuit ground with 100pF in parallel with 100k.
- TCM2-43X+ baluns on clock input and all outputs for single-ended to differential conversion
- +12V barrel connector on front panel. Also have an IDC to allow this board to be powered from same supply as Kasli/VHDCI carrier.
- Generate 3V3 from SMPS followed by low noise LDO (same used on Sayma) (as well as the usual ferrites/LC filters between the SMPS and LDO).
- Screening can for clock buffer and baluns.
- 1 LED for power on.
- No EEPROM/I2C required.

gkasprow, 2017-08-29:
@hartytp how many of them could you need? What is your timeline?

hartytp, 2017-08-29:
> how many of them would you need?

Probably between 3 and 5, although a few other people also expressed interest at NACTI as well.

> What is your timeline?

Say, delivery (at least of prototypes) by the end of this calendar year if possible.

jbqubit, 2017-08-31:
Yes. I’m interested in buying 2-3 copies of this board.

gkasprow, 2017-09-01:
@hartytp we can arrange SMAs like this: 
[image]
Please have a look at the [schematic.](https://github.com/m-labs/sinara/blob/master/ARTIQ_ALTIUM/Kasli/3U/PCB_3U_CLK_Dist/3U_CLK_DIST.PDF)

jordens, 2017-09-01:
Please don't pack the SMAs that tightly. The ones on Sayma for example are too close together to be screwed in even with the most compact tools we had available.

hartytp, 2017-09-01:
@gkasprow That looks great, thanks! A few comments:

- I agree with @jordens: the SMA spacing is too tight. What about dropping to 4 SMA outputs? That should be plenty for most cases IMO.
- If possible can you slightly separate the input SMA from the rest to make it easier to see (otherwise, we'll just put a box around it or something on the panel)?
- Can you add a title to the schematic ("Low noise clock buffer" or similar)
- Can you add an annotation with the current consumption from the 12V rail, please?

gkasprow, 2017-09-03:
It seems I found perfect insulating washers for SMAs. PN is Keystone  7687
One can install them on both sides of panel and fix using only one screw.
[image]

SMA arrangement is now as below
[image]

hartytp, 2017-09-03:
@gkasprow Good job! Washers look nice and cheap (8p in quantity of 100!), and the pitch on those SMAs, looks pretty manageable. Let's use those on Allaki as well.

Thanks for finishing. This all looks good to me. 

Let's set a deadline for additional comments at the end of Monday. After that, let's close this issue and send the prototypes for manufacture!

jordens, 2017-09-03:
If those were acrylic, they would make useful and intuitive lightpipes as well.

gkasprow, 2017-09-03:
@jordens Good point, at least it's worth testing. In case of Urukul there are 2 LEDs close to each input. I can place there RGB LEDs :) The question is what functionality would such LEDs have?

jordens, 2017-09-03:
I meant more for Urukul. There they are also AC-coupled, would need those washers and could just use the LEDs through the washers as lightpipes. Since the clock distribution thing is pretty static, I don't think it makes much sense to have more than the two leds.

Also re the current schematics:

* [ ] The meanders for the clock lines should have more clearance to each other.

jbqubit, 2017-09-05:
Looks good!

hartytp, 2017-09-05:
Thanks again for doing such a nice job of this Greg!

closed 2017-09-05 by hartytp

hartytp, 2017-10-31:
@gkasprow Did this go for manufacture yet? What's the timeline on prototypes arriving? We can do the testing on this board if that helps.