sinara-hw/sinara #266: Urukul schematic review

hartytp, 2017-09-02:
Thanks for posting the updated Urukul schematic @gkasprow! Starting a review of it...

- [ ] DDS_CLK1_P/DDS_CLK1_N and similar for other DDSs should be zero-indexed
- [ ] RFSW_CTRL_1 etc should be zero-indexed
- [ ] ...

hartytp, 2017-09-02:
@jordens Can you remind me what the plan for synchronising DDSs is for the "wide bandwidth" case?

gkasprow, 2017-09-02:
@hartytp Thanks for quick review.
 There is 4 position DIP switch. So can be used for clock selectrion, IDC utlisation, LED assignment, etc.
Instead of "DNP" I use variants, check PDF file now.
The LTC regulators go up to 500mA. We need much more than that. We can use dedicated LDOs per DDS chip but we would need plenty of them.
For CPLD update we would need to write bit-bang driver over I2C. I treat this CPLD as hardwired logic which used don;t need to touch. If there is some volunteer who will write such update software, I can add I2C extender that controls JTAG lines.

hartytp, 2017-09-02:
@gkasprow Likewise -- thanks for the quick fixes!

> There is 4 position DIP switch. So can be used for clock selectrion, IDC utlisation, LED assignment, etc.

Thanks.

> The LTC regulators go up to 500mA. We need much more than that. We can use dedicated LDOs per DDS chip but we would need plenty of them.

Okay, so long as the LDOs you've chosen are reasonably low noise, let's stick with them.

If we have issues with noise or supply-line-related cross-talk, we can consider moving to 1 LDO per DDS in the next revision.

> For CPLD update we would need to write bit-bang driver over I2C. I treat this CPLD as hardwired logic which used don;t need to touch. If there is some volunteer who will write such update software, I can add I2C extender that controls JTAG lines.

@jordens may feel differently, but I think it's worth adding the I2C extended, even if it's DNP by default. That way, at least we have the option of programming over I2C if we decide to write the loader later on.

jordens, 2017-09-02:
Wide bandwidth case?

jordens, 2017-09-02:
Awesome!

* I like the cpld idea.
* We should agree to work hard to keep the cpld logic purely combinatorial. And it should be written, agreed upon, and unittested before the prototypes arrive. In that case I am ok with JTAG only.
* Did you go for the Si fanout just because it is lower power and cheaper than another adclk? Is it worth the additional bom line?
* PS design looks ok to me but needs a thermal sim.
* Also the (or at least two of the) first prototypes need to be with ad9912.
... More coming.

gkasprow, 2017-09-02:
@jordens I used this LVDS fanout because it has individual output enables and I need LVDS not LVPECL. However I cannot use it as clock fanout because it runs up to 700MHz.

hartytp, 2017-09-02:
> Wide bandwidth case?

The mode used in the NU servo (that's what it's called on the Wiki atm).

jordens, 2017-09-02:
@hartytp use the ad9910 sync mechanism and either drive that sync from the servo or sync the servo to it. AFAICT either should work.

hartytp, 2017-09-02:
@gkasprow On the new version of the schematic, all the repeated blocks are now included multiple times (e.g. 4 identical pages of DDSs). In the past, there was just one copy of each, which was better IMO. Can we go back to just one copy of each identical block, please?

Also, obvious point, but have you double checked the termination/matching required for all the DDS inputs/outputs? (IIRC, there are a few quirks about how the various pins are terminated/need to be terminated/matched).

hartytp, 2017-09-02:
> @hartytp use the ad9910 sync mechanism and either drive that sync from the servo or sync the servo to it. AFAICT either should work.

@jordens My memory of the AD9910 is a little rust now, but IIRC:
- To sync the AD9910 with the servo, one needs to supply it with a SYNC_IN signal generated by the servo
- To sync the servo with the AD9910, one needs to feed the SYNC_OUT signal to the FPGA.

However, as currently described on the Wiki, neither of those signals are connected to the IDCs in the "wide bandwidth" case. Or, am I missing something?

gkasprow, 2017-09-02:
@hartytp notice that pages differ by component designators.
Altium prints each channel separately when assembly variants are used because each channel can differ.
Another option is to use DNP marking for not mounted components.
SYNC_OUT from DDS0 can be routed via IC9 to LVDS1 directly to avoid TTL->LVDS buffer delay

hartytp, 2017-09-02:
> Altium prints each channel separately when assembly variants are used because each channel can differ.
Another option is to use DNP marking for not mounted components. But then we cannot have variants for 9912 and 9910.

Okay, that makes sense. If the variants option is useful for maintaining multiple population options then I don't object to using it...

> SYNC_OUT from DDS0 can be routed via IC9 to LVDS1 directly to avoid TTL->LVDS buffer delay

My question was more about pin usage on the IDC connector; for the "wide bandwidth" case, where we have lines for each MOSI and for each switch, there aren't any lines spare for SYNC_IN/SYNC_OUT (although, these could be multiplexed with the attenuator control lines using a CS and the PLD).

jordens, 2017-09-03:
* [ ] Wire `SYNC_SMP_ERR[3:0]` and `PLL_LOCK[3:0]` to the CPLD as well. Then we don't necessarily need testpoints there.
* [ ] As mentioned before, don't wire `EXT_PWR_DOWN` as that's well configured directly over SPI to the DDS. Doing it over SPI through the CPLD does not make much sense to me.
* [ ] Put two DDS (0/1) SYNC_IN on Bank A (of IC9)  and the other two (2/3) on Bank B for symmetry, equal loading, and aesthetics. ;)
* [ ] Please allow for individual control (separate CPLD pins) of EN of IC4 (FIN1101K8X) and OE4 of IC9.
* [ ] Why do you feed the single ended DDS0_SYNC_CLK also to the negative input (1) of IC4?
* [ ] When routing please do a reasonable (100ps) length match within each of DDS_SYNC_CLK[3:0], DDS_UPDATE[3:0], RFSW_CTRL[3:0], IO[15:11] (the RFSW_CTRL lines), DDS[3:0]_SYNC_IN, and a really good one on DDS_CLK[3:0]_[PN].
* [ ] I don't think that idea of muxing the SYNC fanout onto IDC1:LVDS1 works. The Si53304 says: "When OE is low, Q is held low and Q is held high for differential output formats". The drivers are not disabled, just the pulses. Do we need another FIN1101K8X here? Is there a better suited mux/fan-out chip (CDCLVC1106?) then (since we can't use/don't need the OE feature)?
* [ ] Should the shield on CK_SMA_IN (J2) also be AC-coupled? @hartytp 

* [ ] On the DDS pages, there are several components that could/should be shared between the variants. Like R70/R102 and a couple of the chokes.
* [ ] What's the function of the RCs (R41, C26 etc) on SDI/SCKI? Noise?
* [ ] Document the loop filter component choices (and design Icp) for both AD9912 and AD9910 and confirm that they are designed for 1 GHz from  100 or 125 MHz. For the AD9912 the 10x numbers are 470 Ω, 820 pF, 56 pF.
* [ ] Take really good care placing and choosing DAC_RSET (R70 and R102) as those are stability relevant.
* [ ] If possible, connect I/O_RESET to a separate CPLD pin (all DDS in common).

* [ ] You still have the ERA-4XSM+ with 75/65 mA in the power budget and in the annotations (instead of the ERA-3+ with 35mA)
* [ ] TR1 [should be](https://github.com/m-labs/sinara/issues/199#issuecomment-304343253) a 1:sqrt(2) transformer. Does minicircuits have anything closer to that (e.g. 1:1.5)? Or is 1:2 already optimized for some reason I don't see? I know it is on the eval board...
* (Maybe) feel free to drop the discrete reconstruction filter (C164 etc) to save space.

* [ ] Delete R45 and R81 and DNP C194, C81 in that variant. The negative innputs of IC17 are already taken care of.

* [ ] Why are the SYNC_CLK lines on GTS (global tristate) pins of the CPLD? I'd put SYNC_CLK0 (only) on GCK2. Leave GCK0 and GCK1 as is.

* If you need to save CPLD pins: fewer LEDs (4) is fine, or/and group the RE_DE[15:0] wires.

@hartytp re synchronization. we would use the CPLD to enable driving SYNC_IN from the FPGA during the startup phase.

hartytp, 2017-09-03:
> @hartytp re synchronization. we would use the CPLD to enable driving SYNC_IN from the FPGA during the startup phase.

Thanks for clarifying, that sounds good. Do you think that, once synchronised, synchronisation will be maintained for months? Do we need a way of checking for sync errors?

- [ ] To do: add this to the Wiki.

hartytp, 2017-09-03:
- [ ] @gkasprow Consider making the DDS current set resistors decent temp co (say, 25ppm/K) for improved amplitude stability (e.g. if this is ever used open-loop to drive quantum logic gates).

hartytp, 2017-09-03:
@jordens Re isolated GNDs on Urukul SMAs: I think it's nice to have the pads on the PCB so that floating the grounds is easy (just means altering the FP) if required. However, I'm not sure it's worth the hassle of actually fitting isolated washers by default (depends a little on how much it adds to the cost, and how good it is mechanically). Isolating the grounds is certainly not actually required for anything I'm planning to do with these boards...

hartytp, 2017-09-03:
> TR1 should be a 1:sqrt(2) transformer. Does minicircuits have anything closer to that (e.g. 1:1.5)? Or is 1:2 already optimized for some reason I don't see? I know it is on the eval board...

@sbouhabib Can you remind me why we didn't do this for Allaki? Shouldn't that also have been a 1:sqrt(2) rather than 1:2?

jordens, 2017-09-03:
I agree on the shield AC coupling. But it should be done consistently for the SMA clock input and the SMA DDS outputs. I.e. also make the clock input AC-couplable.

hartytp, 2017-09-03:
I have a vague memory (but didn't see anything in a quick skim through the data sheet) of the SYNC_CLK output of the AD9910 only having a really weak current drive. Needing something like 1k min termination. Does that ring any bells @cjbe?

hartytp, 2017-09-04:
@jordens does it matter that IO_UPDATE and SYNC_IN/IO_UPDATE_RED will be on different IDCs, while Kasli/VHDCI_CARRIER currently only do length matching between lines on the same IDC? (i.e. IO_UPDATE and SYNC_IN paths will not be length matched)?

gkasprow, 2017-09-04:
@hartytp It says only that CMOS outputs deliver 0.4 or 2.8V at 1mA load.
Which means that the output resistance is roughly 400...500OHms.

hartytp, 2017-09-04:
Thanks @gkasprow!

So, given that SNC_CLK is a low current CMOS output:
- we need to be careful with capacitive loading on this pin, so connecting it directly to the IDC without a buffer is probably not a good idea
- We should probably not 100Ohm terminate it at the input to the buffer (as in the current schematic)
- If we do buffer this signal, but not the other synchronisation-related signals, will the propagation delay of the buffer be an issue?

jordens, 2017-09-04:
The fanout (andf the CPLD) are there to do the buffering.
SYNC_OUT only needs to be matched on the branches to the DDS (SYNC_IN, as I described above). The rest doesn't matter.

hartytp, 2017-09-04:
> The fanout (andf the CPLD) are there to do the buffering.

ACK, but let's increase the termination resistance a bit.

gkasprow, 2017-09-04:
@hartytp the SYNC_CLK is delivered to the IC4 via level conversion resistive network.
One can increase R31 to even 1kOhm. We need only to make sure that LVDS buffer see signal amplitude at least 200mV.

hartytp, 2017-09-04:
@gkasprow Okay, if you're happy this will work reliably, that's fine then.

jordens, 2017-09-05:
I split the pending open items from this issue into #270 #271 #272 #273 #274 #275 #276 #277 #278 #279 #280 #281 #282

closed 2017-09-05 by jordens