sinara-hw/sinara #267: Urukul MMCX cables

gkasprow, 2017-09-03:
For Urukul and Kasli we would need MMCX to MMCX cables.
Would be nice to have them delivered together with 3U boards.
[Here](http://pl.mouser.com/Wire-Cable/Cable-Assemblies/RF-Cable-Assemblies/_/N-bkrix?P=1yzsqmlZ1yzsqp1&Keyword=147913546&FS=True&Ntk=P_MarCom) are example cables.
What lengths should be delivered?

jordens, 2017-09-03:
60cm is fine IMHO. Should match the IDC length. But package them with Urukul, not with Kasli.

klickverbot, 2017-09-03:
Wouldn't the most common application be to span up to ~half a 19" rack in width, with some leeway for proper cable management? From the Mouser table you linked, I'd just go for the 24" RG316 option and not worry about any details.

hartytp, 2017-09-03:
24''/60cm RA-RA MMCX works for me.

gkasprow, 2017-09-03:
THe question is if all IDCs and MMCX cable should have same length. For connection of adjacent boards one need 10..15cm of cable. 60cm would need to be folded several times and take some space.
Maybe it makes sense to order IDCs and MMCX cables separately in 2 or 3 lengths.

closed 2017-10-09 by gkasprow