sinara-hw/sinara #268: HV supply

gkasprow, 2017-09-04:
Some time ago I developed quite interesting HV supply modules with floating secondary side and current limit. They are supplied from single 12V, have RS485 or TTL control interface and galvanic isolation of 10kV. Output is floating which enables several channels to be connected in series. Each channel has adjustable current limit, fast discharge and disable switch. The project was intended for GEM detectors.
Outputs go from 0 to 1.5kV, 0-1mA but can also support 0-8kV.
Now I have a request for anther detector project to put them on 3U board with simple backplane.
And the question is if quantum community would be interested in such board. I can simply add EEM connector to it. As I wrote, control interface is trivial - just 12V, enable signal and UART. Several modules can be connected together using same interface - they are addressable (RS485).

jordens, 2017-09-04:
Quantum stuff usually ends around 1 kV. But maybe some Penning people or something more exotic could use that.

sbourdeauducq, 2017-09-05:
How much do they cost?

gkasprow, 2017-09-05:
Component cost is roughly 60$ per channel. But this is low noise design based on double resonant converter.

jbqubit, 2017-09-05:
Penning application is ~5000V, 100e-6A, ~16 channels. What is noise of the current design? Is interface for setting voltage analog ("12 V") or digital (UART)? This would be a nice addition to Sinara.

gkasprow, 2017-09-05:
Noise is below 100mV. 12V is supply, control interface is single IO line and UART.
We can make 5kV easily and pack 2 channels per 1 3U EEM.

hartytp, 2017-09-05:
Given how many designs we have on the go at the moment, it seems like our resources are starting to get stretched a bit thin. Better to avoid starting any new designs until some of the current ones are finished and tested?

gkasprow, 2017-09-05:
we are talking about one year perspective:)

>

hartytp, 2017-09-05:
Okay. But, I for one don't have enough brain space left to even begin thinking about new devices right now.

jbqubit, 2017-09-05:
Current NIST Penning uses Canberra 3125 HV . Noise spec is
- < 3 mV peak-to-peak with 100e-6 A load
- long term stability is 1e-4/hr

We also used a ~ 10 Hz low pass filter to further decrease noise.

jbqubit, 2017-09-05:
Current Penning experiments are not really limited by commercially
available HV sources so a new PCB isn't really needed. But for reference,
NIST Penning uses Canberra 3125 HV . Noise spec is
- < 3 mV peak-to-peak with 100e-6 A load
- long term stability is 1e-4/hr

We also used a ~ 10 Hz low pass filter to further decrease noise. Penni

I agree with @hartytp. Let's not get distracted from designs currently in
the pipeline. Each of these PCB's warrants careful consideration and
there's not enough collective bandwidth for it right now.

gkasprow, 2017-09-05:
OK, guys are building such supply anyway, once needed we can easily integrate it with EEM.

closed 2017-09-05 by gkasprow