sinara-hw/sinara #269: Allaki rev1 errata

gkasprow, 2017-09-04:
- [ ] replace AD8363ACPZ-WP  with AD8363ACPZ-R7CT-ND 
- [ ] replace ADT7420UCPZ-R2 with ADT7420UCPZ-RL7TR-ND

sbourdeauducq, 2017-09-05:
As explained in detail several times already, it would be better to have one issue for each item instead of one big errata, and use milestones and labels.

jbqubit, 2017-09-22:
@gkasprow asked in email
>To do tests of Allaki, I'd need sine wave at the DAC output. The sooner, the better.
What about clocking DAC from 100MHz source?

jbqubit, 2017-11-16:
- Allaki as-shipped to UMD do not have serial numbers. Please add serial numbers so end-users, WUT and Technosystem can track.

- back-side silkscreen double printing 
[image]

- As-shipped to UMD insulating front-panel missing. @gkasprow is sending the right panel. Photo from @gkasprow below so everybody can see what this will look like. 
[image]

hartytp, 2017-12-04:
Does anyone object to removing the power detector from BaseMod v1.1? cf #397

dhslichter, 2017-12-04:
The point of the power detector is to noise-eat at the output of the DACs.  What is it costing us?

hartytp, 2017-12-04:
Yes, but only when the switch is off, and only via the 24-bit ADC.

Main cost is space: as @sbouhabib pointed out elsewhere, this is the largest component on BaseMod and fitting it on was actually quite challenging. If we remove it then we can make the layout a bit less tight.

Not a huge deal, but it's not totally cost free so unless you **really** plan to use it then we should scrap it.

gkasprow, 2018-08-18:
This issue was moved to sinara-hw/BaseMod#6

closed 2018-08-18 by gkasprow