sinara-hw/sinara #270: AD9910 PLL divider

jordens, 2017-09-05:
Could someone with a AD9910 eval board check that a PLL divider of 10 works? The datasheet mentions 12 as the minimum and the PLL loopfilter spreadsheet only does 12.
Is the REFCLK divide-by-2 in front of the PLL or not? There are conflicting statements on the internet.
If the divider is in front of the PLL, a PLL divider of 20 would be the one.

hartytp, 2017-09-05:
Good point @jordens. @gkasprow can you look into this with the eval board I posted you? Otherwise, I can buy another and get someone here to test this next week...

T

hartytp, 2017-09-05:
Also, let's check this with a divider of 8. At least in our labs, we'll probably aim to clock the DDS boards from the 125MHz CDR RTIO clock for convenience.

hartytp, 2017-09-05:
> Is the REFCLK divide-by-2 in front of the PLL or not? There are conflicting statements on the internet.

I don't think so, no.
[image]

hartytp, 2017-09-05:
From the data sheet, it looks like you're right and the minimum divider is 12x. 

Maybe the best bet is to put a low-noise divider before the clock buffer. Maybe something programmable like an HMC983, so that we can set it to 1 if not needed? We can always remove it after the prototyping round if it turns out not to be required.

jordens, 2017-09-05:
As I said there are conflicting statements about the position of the REFCLK input divider.

* Please no additional PLL here. That's mission creep.
* There is little space on the board, there is little time, massive added complexity debugging and programming it and there is little use. I would suggest building an HMC830 into the clock distribution or have it separate.
* The most I'd be willing to incorporate is a discrete divide-by-two in front of the Urukul clock fanout.

jordens, 2017-09-05:
Oh sorry. I had read HMC830. A divider is ok if it is really small, and simple. I.e. a fixed divide-by-two. You can then then populate it if you drive it without the PLL.
But the AD9912 variant won't have that populated.
But even the HMC983 is not something that I want to have here.

hartytp, 2017-09-05:
How about HMC432(E)? Would have to think about how to fit this in with the clock muxes. Also it needs a square wave input.

jordens, 2017-09-05:
Sure. But do we really need it? Can't we just use the dividers in the SI5324? It has two independent outputs. We could use one just for the fan-out and the other for the rest (FPGA+GTP+CDR_OUT)

Also #287

hartytp, 2017-09-05:
@jordens The downside of that is a lack of flexibility: what if one wants to use two different EEMs that need different clock frequencies, connected to the same Kasli? Then one has to start mucking around with external dividers etc, which is a bit of a pain. 

IME, it's generally best to distribute a single reference frequency and design all boards to run off it.

But, if you strongly object to putting a divider on Urukul then yes, changing the Kasli EEM clock frequency is another option.

jordens, 2017-09-05:
Let's meet in the middle. Put the internal fan-out from Kasli on a programmable divider (#287) and put the footprint for a fixed, non-configurable, divide-by-two into the clock path. But let's keep it really simple, needs to be really cheap (<10$), no voltage standard changing required, and not something we have to bother with for the first prototypes.

hartytp, 2017-09-05:
> Put the internal fan-out from Kasli on a programmable divider (#287)

That sounds like a smart move in any case, independent of what we decide in #287.

hartytp, 2017-09-05:
@jordens If you've found conflicting information about this on the web, have you tried asking ADI for clarifications? No point worrying about this if it's just a data sheet mistake/unclear comment...

Otherwise, @gkasprow is the minimum divider ratio for the AD9910 something you can check quickly using the eval board I sent you?

Issues if we have to use a divider:
- We should change the XO frequency to 200MHz or similar
- Where do we put the divider in relation to the mux? Do we want it only the MMXC arm of the divider, or do we want it after the divider?
- The divider only works with square-wave signals below ~200MHz, so may need to be after a clock buffer IC if we want to use it with an external oscillator input.

jordens, 2017-09-05:
I haven't. Just found that this morning.

hartytp, 2017-09-05:
> I haven't. Just found that this morning.

Are you okay to follow this up with them?

jordens, 2017-09-05:
https://ez.analog.com/message/316249

gkasprow, 2017-09-06:
I will test it.

jordens, 2017-09-06:
That sounds more like the AD9912...

hartytp, 2017-09-06:
> So, the eval software enables me to set multiplier value between 1 and 127, but with values lower than 8 it displays red frame around it.

Limitation of the eval software?

> One can also enable frequency doubler
 
I didn't think the AD9910 had one...

gkasprow, 2017-09-06:
@hartytp when I enable the doubler, I observe twice higher output frequency. So it seems to work. I connected 25 MHz input clock.
The sofware also does not like division of 11, but due to the fact that VCO cannot tune to such frequency.
The doubler option does not resolve it. But when I set to 22 it works.
So when I set to 10 or 11 with doubler enabled, it loads same value.
So I started playing with it and was increasing divider from 18 do 23.
And values of 20,21 and 22 give exactly same frequency at the output. So it seems it is some bug in the software. later on values 23, 24, 25 also give same value.
It seems that PLL does not lock and what I do is simply switching VCO ranges.

gkasprow, 2017-09-06:
@hartytp 
correct, I ignore multiplier and switch VCO range and get same results.
Did you manage to lock the PLL?
The lock indicator is off all the time.

gkasprow, 2017-09-06:
When I click the doubler field, it simply changes VCO ranges. So I had impression that it worked.
I noticed that I get same result with and without external clock :)

hartytp, 2017-09-06:
> The sofware also does not like division of 11, but due to the fact that VCO cannot tune to such frequency.

What if you put in a clock at 125MHz, and set a division of 8?

> Did you manage to lock the PLL? 

Is the loop filter populated by default?

gkasprow, 2017-09-06:
I didn't touch the loop filter so it should be.

gkasprow, 2017-09-06:
Once I force it to division 7, the chip locks somehow and I have to reset it to get any output.

gkasprow, 2017-09-06:
OK, so the PLL does not lock but it tries. When I touch with my finger to the loop components, it  locks and lock indicator goes green :)

hartytp, 2017-09-06:
@gkasprow What were in reference and DDS clock frequencies for that test?

gkasprow, 2017-09-06:
I used 125MHz reference and 1GHz clock.
But have same issue with 25MHz ref and 500Mhz clock.

hartytp, 2017-09-06:
Great! So long as 125MHz reference and 1GHz DAC clock locks fine, I'm happy and we can close this issue.

jordens, 2017-09-06:
I am confused. Does the relevant configuration work (with certainty and not with a finger) or not?

gkasprow, 2017-09-06:
@jordens It means that the loop filter needs some skin resistance/capacitance to lock. Otherwise it does not lock in any conditions. There are not mounted loop filter components by default: C13,C15, R37.

jordens, 2017-09-06:
I'd be interested in seeing the test of the two relevant cases (multipliers of 10 and 8) described in the top post. Has that been done? Maybe I am misunderstanding you.

* populate C13,C15,R37 with suitable values for 125 MHz external clock input (or 100 Mhz) and 1000 MHz DDS system clock
* note those values in #274 
* test the populated board with a PLL multiplier of 8 (or 10) and 125 MHz input (or 100 Mhz)
* check with a scope that the PLL stays permanently locked (to confirm that it does not just lock temporarily)
* set the DDS to some frequency and confirm that that frequency is output (to confirm that the PLL locks to the correct frequency)

gkasprow, 2017-09-06:
@jordens  I soldered loop filter components and it locks stable now
[image]
On the scope i see sine wave with 10.02 MHz

gkasprow, 2017-09-06:
@jordens it works with 100MHz and div=10 as well.

jordens, 2017-09-06:
Excellent. Thanks! Then we can skip the divider. But I am surprised that the PFD works well at that high frequency. My guess it that that's why they only spec it up to 60 MHz.

gkasprow, 2017-09-06:
@jordens it works in this particular chip . May have issues with others due to production parameters dispersion.

hartytp, 2017-09-06:
@gkasprow Well, let's try it without the divider for the prototype round. If we have issues, we can add one later (and dead-bug a divider onto any proto boards we build).

hartytp, 2017-09-06:
Thanks again for testing that so fast @gkasprow!

Will be interested to hear what ADI have to say about this though...

gkasprow, 2017-09-06:
You can post it to ADI blog...

hartytp, 2017-09-06:
Thinking about this a little more, I'm not happy about leaving this divider out for the prototype round: @jordens is probably correct that we'd be running the AD9910 PFD outside it's specified range, opening the door to potential PVT issues.

Can I make a request for the prototype round:
- If there is space on the PCB, please add a second ADCLK948 between the SMA/MMCX inputs and the DDS clock pins. 
- The second ADCLK948 drives the DDS clock inputs.
- Two outputs of the first ADCLK948 should connect to the inputs of the second ADCLK, all other outputs should be NC (so that power dissipation due to the extra IC is low).
- Between the two ADCLK948s, add an HMC432 (or similar high-performance, low cost divider). NB this needs a square-wave input for frequencies <~200MHz. Use a 3V3 supply rail for this divider.
- Use the DIP switch to control the second ADCLK, to give a switchable divider for the input clocks.

My thinking here is:
- yes, it's a little ugly, but I don't want to rely on using the AD9910 PLL out of its specified operating range. Adding this divider gives us an important extra piece of flexability, allowing us to run this board directly from the standard RTIO clock or from our standardised 100MHz reference frequency without having to have an external divider. I see that is a major benefit.
- Extra power consumption due to this is pretty small.
- Extra cost is only ~15USD, so negligible compared to the BOM cost.
- Once we've tested the prototypes, we can consider removing this for the production round if we want...

jordens, 2017-09-07:
No. Let's not do that. Just the divider after the internal mmcx seems absolutely sufficient to me. And dnp it for the ad9912 variant. If you feeding the external sma you are expected to be able to supply something suitable anyway. If you use the internal clock distribution going for 62.5 MHz is sensible. We don't have any other user for the internal clock tree and it already has a configurable divider.

hartytp, 2017-09-07:
AFAICT, we definitely should NOT rely on using the 100MHz/125MHz input for the AD9910. [This](https://ez.analog.com/thread/38735) post points out that it is not guaranteed to work -- no details about why, but it doesn't seem like a good thing to rely on in our design.

As a result, this input divider is a high priority for us.

> If you feeding the external sma you are expected to be able to supply something suitable anyway. If you use the internal clock distribution going for 62.5 MHz is sensible. We don't have any other user for the internal clock tree and it already has a configurable divider.

I would really like to move away from this kind of attitude for Sinara. In my experience, if one doesn't take the view that all boards should work with a common reference frequency (100MHz/125MHz in this case), the result is a gradual proliferation of clock frequency requirements as the system grows and new designs are added. 

This leads to users having to supply multiple clock sources/PLLs, along with the accompaniment of cabling, buffering, etc. (e.g. on your suggestion, users now have to get an additional 62.5MHz clock from somewhere to use the front panel SMA, which is inconvenient). That means another signal generator to add and distribute. Not to mention the requirement of documenting (and users understanding) the clock requirements of each board.

In the long-run, life is MUCH easier if there is just a single reference clock that works with everything!

> Just the divider after the internal mmcx seems absolutely sufficient to me.

I would strongly prefer to have the extra mux as well as the divider:
1. Having the divider only on the MMCX arm breaks the symmetry between the two clock inputs. This introduces an extra gotcha that must be documented and understood by users (you can plug 100MHz into here, but not there).
2. Clocking Urukul from 125MHz/100MHz on the FP SMA seems like a reasonable use case, which we should support. e.g. if Urukul is connected to a carrier other than Kasli via the VHDCI_breakout. In that case, there will probably not be a convenient clock source in the rack that can be connected to the MMCX. So, the most convenient clocking option is to supply either 100MHz/125MHz reference (which we will already have for other boards) via the FP.
3. At this relatively low frequency, the dividers only work with clean square waves, so it's good to have a buffer on Urukul to square the signal up (allows a sinusoidal reference to be used, and also makes us less sensitive to attenuation of high-frequency components of the square-wave clock in cabling).
4. While the divider and mux might seem like mission creep, it's pretty negligible: total cost of both is  around $15, which will likely be 2% of the total cost of an Urukul (~$500 to $1k); the power consumption is also a pretty negligible fraction of the total board power (remember, we're only using two outputs of the LVPECL buffer/mux); board area consumed is also pretty tiny, as is increase in schematic complexity; no extra power supply rails required/ICs to program/etc. So, I don't even see a need to make this DNP for your A9912 variant -- you don't need to use it, but the money etc you'll save by removing it isn't worth the hassle of having more differences between the two variants IMO...

jordens, 2017-09-07:
Correct. The attitude of trying to cram in more and more features and accommodating for ever increasing number of external constraints and idiosyncrasies has been very bad for Sayma. Just look at the hundreds of possible wishlist clocking options that were wedged in there just because it was possible and because nobody was responsible for signing off.

That's exactly why I don't want to support 100/125 **or** 50/62.5. That flexibility is definitely not worth it. And that's why I don't see the need for another clock mux to support operation with and without a divider. If you say, the ad9910 needs the divider, then we will add the divider for the ad9910. But we need to do it in a way that does not increase the options, has acceptably low effect on routing and BOM cost, and zero additional things to control and debug. I can't imagine that another 2:6 LVPECL differential mux, plus a single ended divider feeding another differential LVPECL mux, plus another LDO to make 3V is the best option here.

@gkasprow What do you think? How can we map the 125 MHz from either of the two clock inputs to 62.5 MHz at the four DDS with minimal resources?

gkasprow, 2017-09-07:
@jordens I'd use clock distribution IC with mux, divider and fanout, like [Si53312](https://www.silabs.com/documents/public/data-sheets/Si53312.pdf)

hartytp, 2017-09-07:
> That's exactly why I don't want to support 100/125 or 50/62.5. That flexibility is definitely not worth it. 

ACK. However:
- While we could replace one of the muxes with a fanout buffer, that would introduce an additional BOM line. Lesser of two evils? If you want, we can just use the second ADCLK948 as a fanout buffer, and only use one input.
- IIRC, you originally wanted to leave the door open to using a 1GHz input clock (no DDS PLL). This would require a bypass for the divider.
These were my motivation for using the mux/divide combination.

> If you say, the ad9910 needs the divider, then we will add the divider for the ad9910.

From what I can tell from the posts linked above, it seems the divider is needed. But, I'd be happy if someone could convince me otherwise.

> I can't imagine that another 2:6 LVPECL differential mux, plus a single ended divider feeding another differential LVPECL mux, plus another LDO to make 3V is the best option here.

Only one of these muxes is new. The other one is already in the schematic. Also, AFAICT, the divider will run fine from 3V3, so no new LDO required.

gkasprow, 2017-09-07:
@hartytp Si53312 seem to solve the issue.

jordens, 2017-09-07:
@hartytp In the original design the SMA panel input would be exclusively to drive the DDS directly (i.e. 1 GHz). While the internal one would be "not ultra-low noise anyway" and would go with the DDS PLLs.

But AFAICT Si53312 gives us what we need in noise, division, IO levels, BOM, price. Thanks @gkasprow .

hartytp, 2017-09-07:
> @hartytp Si53312 seems to solve the issue.

Yes, sorry, I saw that after posting.

From a quick skim over the data sheet, it seems fine to me.

Phase noise is something like 10dB worse than the AD9910 with the PLL off at most frequencies. Which doesn't particularly bother me (for ultra-low noise, people should use Sayma or similar).

If you and @jordens think this is simpler than using a divider and a couple of muxes then I'd be completely happy with this solution.

hartytp, 2017-09-07:
How do you want to wire the divider? Use a DIP switch to go between DIV 1 and DIV 2 (open and pulled down)?

jordens, 2017-09-07:
I see only 2-5dB degradation of the AD9910 noise at 10kHz and 1MHz comparing the 156 MHz numbers with the 201 MHz.

hartytp, 2017-09-07:
> I see only 2-5dB degradation of the AD9910 noise at 10kHz and 1MHz comparing the 156 MHz numbers with the 201 MHz.

Maybe I'm doing a bad job reading plots on a small screen. 

Either way, I don't think we care about it.

gkasprow, 2017-09-07:
@hartytp we can connect it to free CPLD pin. Then in the code we can generate 3 output states.

hartytp, 2017-09-07:
> @hartytp we can connect it to free CPLD pin. Then in the code we can generate 3 output states.

ACK. But, a DIP switch is probably easier for the user. Software control is not required for this pin IMO, so it's better just to use a switch.

hartytp, 2017-09-07:
@gkasprow Sure, but let's connect it to the dip switch as well. That way, using the switch it can be set to either div=1 or div=2 (which are the important cases).

gkasprow, 2017-09-07:
@hartytp We will do it via CPLD :)

closed 2017-09-08 by jordens