sinara-hw/sinara #272: Urkul output BALUN

jordens, 2017-09-05:
We should [probably](https://ww2.minicircuits.com/app/AN20-002.pdf) go for the standard center-tapped transformer instead of the trifilar TLT. The original design came over from Allaki which wants to go to higher frequencies. But for the usual DDS applications and frequencies the center-tapped trafo seems more suitable.

gkasprow, 2017-09-06:
@jordens  Minicircuit specifies impedance ratio, not turn ratio. So 1:2 should be fine.

jordens, 2017-09-06:
Oh. Thanks for clarifying.  But then is the eval board wrong? And what about the 100 ohm resistor on the ad9912 board?

dhslichter, 2017-09-06:
IME one shouldn't take the balun ratios and populated resistances too literally, especially because the output impedances of the DDS chips can vary substantially over frequency.  When working with the AD9914, I found that doing the "expected" thing gave worse performance than the way it was set up on the eval board at higher frequencies, although better at lower frequencies.  

Usually there are design files with measured output impedances that exist, which one can combine with measured S parameters of baluns to do a simulation.  Looks like the AD9912 has an internal 100 ohm differential termination already (they claim on the datasheet), and the eval board has [optional resistor pads](http://www.analog.com/media/en/evaluation-boards-kits/evaluation-board/AD9912_Schematic.pdf).  Someone with a network analyzer should look at the return loss looking into the output of the DDS via the balun with and without those resistors populated and see what you get.

gkasprow, 2017-09-07:
@dhslichter I can check with network anayzer at the devkit output. I assume that DDS needs to be powered on.

dhslichter, 2017-09-07:
@gkasprow yes, power on the DDS, turn on the output to some reasonable value and check the return loss at the devkit output.  Then play with balun choice and/or termination resistors at DDS output until you find something that matches reasonably well in the band of interest.

gkasprow, 2017-09-07:
@dhslichter I'm not sure if measurements taken with signal at the DDS output would be representative.
I can make measurements quickly but optimisation would have to wait.

dhslichter, 2017-09-07:
You can set the amplitude to zero if you want.  If the amplitude is nonzero, you'll get some spike in your S11 at the signal frequency but just ignore that spike and look at the frequencies to the side.  I would at least check with zero amplitude vs nonzero amplitude to see if it makes a difference to the impedance match (it shouldn't, but there could be higher order considerations).

Agreed that right now we just want to see if the eval board solution is good enough, and if so, we won't worry about optimizing.

jordens, 2017-09-07:
Also here @gkasprow let's focus on getting prototypes out. We can optimize the baluns later on the prototypes.

gkasprow, 2017-09-17:
I changed it to ADT1-1WT+ used on 9910 devkit which is footprint compatible with ADT2-1T-1P used on 9912 devkit.

closed 2017-09-17 by gkasprow

jordens, 2017-09-17:
Could we also get updated BOMs?

commit ee43099c (2017-09-17): fixed #272, reduced pdf size