sinara-hw/sinara #274: Urukul loop filters

jordens, 2017-09-05:
* [ ] Document the loop filter component choices (and design Icp) for both AD9912 and AD9910 and confirm that they are designed for 1 GHz from  100 or 125 MHz. For the AD9912 the 10x numbers are 470 Ω, 820 pF, 56 pF.
* [ ] Why do you (AC-) connect the AD9910 loop filter to P1V8 and not to GND? Why do you handle the AD9912 differently?

gkasprow, 2017-09-06:
@jordens I do as it is done  in the devkit

gkasprow, 2017-09-11:
@jordens what loop bandwidth and phase margin do you want?

jordens, 2017-09-17:
1.6 MHz bandwidth and 65 deg margin. Those seem to be the AD9912 numbers.

hartytp, 2017-09-17:
At some point, we should try to optimise this for some input clock noise spectrum (e.g. the CDR clock), but that sounds like a good starting point for the prototyping boards.

jordens, 2017-09-17:
@gkasprow oh yeas. and if possible make them 0603 for solderability (if they aren't already).

gkasprow, 2017-09-17:
@jordens they are already 0603.
I added separate RSET and loop filter components  due to layout constraints. Otherwise traces would need to be long and pickup noise.

jordens, 2017-09-18:
Very good!

hartytp, 2017-09-18:
- Is the AD9910 loop filter missing from the current PDF, or did I just miss it (I saw it for the AD9912)?

gkasprow, 2017-09-18:
@hartytp It was common for both ICs, but recently I made them separate.

jordens, 2017-10-10:
@gkasprow You currently have the same values for the AD9910 and AD9912 loop filters. From what I can see that won't work. Those two a very different on multiple fronts. The AD9912 has a ~2pF internal shunt, will use 100 MHz PFD freq (and not the divided 50 MHz) and should get the datasheet values for 10x, 250µA, 800 MHz/V (470 Ω + 820 pF || 56 pF). The AD9910 has a 1 kΩ series and 1.1 pF shunt and will use the 50 MHz (20x), 387 µA, 850 MHz/V. I am getting 643 Ω + 698 pF || 36 pF from their AD9910 spreadsheet (for 1.6 MHz, 65 deg). @hartytp should confirm the AD9910 values.

gkasprow, 2017-10-10:
@jordens I used ADI spreadsheet to calculate these values, not datasheet.

jordens, 2017-10-10:
Is there an ADI spreadsheet for the AD9912 PLL?
And the parameters are different for the AD9912 and AD9910. I don't understand how you can arrive at the same values for both.

hartytp, 2017-10-10:
Point from @WeiDaZhang: 
- AD9910 data sheet, table 1 comments that the max REF CLK input frequency with the reference clock multiplier on is 60MHz. Thus, we will need to use the divider in divide-by-4 mode. For 125MHz RTIO_CLOCK, this gives an input clock of 31.25MHz.

jordens, 2017-10-10:
Or even that. But my guess is that 62.5 MHz will be fine.

hartytp, 2017-10-10:
> But my guess is that 62.5 MHz will be fine.

Most likely, yes. But it's always nicer to operate these things inside specification where possible.

Note that if we do go for a 150MHz RTIO clock frequency then divide-by-2 gives us 75MHz, which is starting to get a bit far outside the specified operating range for comfort. Anyway, the advantage of the programmable divider hooked up to the CPLD is that we can play with this once the prototypes arrive...

gkasprow, 2017-10-10:
@jordens I simply assumed that this spreadsheet works in both cases.

jordens, 2017-10-10:
@hartytp Sure. But you still have to decide on one set of loop filter values before production.
@gkasprow It most likely does. But the parameters are all different.

hartytp, 2017-10-10:
> @hartytp Sure. But you still have to decide on one set of loop filter values before production.

@WeiDaZhang is on it. Will get back to you asap.

hartytp, 2017-10-10:
AD9910 PLL plans courtesy of @WeiDaZhang :
- DDS clock: 1GHz
- Reference clock: 125MHz/4=31.25MHz
- DDS VCO range setting: 5 (920-1030MHz, 850MHz/V)
- Charge pump current (Icp): 387uA (max)
- Loop multiplication factor (N): 32
- Loop bandwidth: 3MHz (assumption is that the Kasli clock should be lower noise than the AD9910 VCO, so want a wide bandwidth)
- Phase margin: 45deg

Component values:
- R90B + R92B = 2k2 (this is the nearest E24 value, more exactly: 2k2367) (this is R1 on AD9910 datasheet figure 35)
- C72B||C99B = 56p (this is the nearest E12 value, more exactly: 55p879) (this is C1 on AD9910 datasheet figure 35)
- C112B = 12p (this is the nearest E12 value, more exactly: 12p281) (this is C2 on AD9910 datasheet figure 35)

@gkasprow What tolerance capacitors are you using for this? Is there any point having two series resistors/parallel capacitors? Why not just have one resistor and one capacitor?

@WeiDaZhang can you check this (and compare with current Urukul DDS schematics to check I've got the right components, please).

gkasprow, 2017-10-16:
@jordens I used 5% capacitors. I used double R and C to give better flexibility. It's not always easy to select values so close to E24. Once we verify, we can replace with 1 R and 1C

jordens, 2017-10-17:
@gkasprow Thanks. And the design input parameters are added to the schematics?

gkasprow, 2017-10-17:
Yes, I just updated pdf

jordens, 2017-10-17:
@gkasprow This is still wrong. Please take the AD9912 loop filter from the [datasheet](http://www.analog.com/media/en/technical-documentation/data-sheets/AD9912.pdf#page=24) (470 Ω + 820 pF series, 56 pF shunt). And please fix the bubbles. Or better remove the wrong and redundant component values from the bubbles.

closed 2017-10-18 by hartytp