sinara-hw/sinara #275: DDS RSET

jordens, 2017-09-05:
* Take really good care placing and choosing DAC_RSET (R70 and R102) as those are stability relevant. Consider making the DDS current set resistors decent temp co (say, 25ppm/K) for improved amplitude stability (e.g. if this is ever used open-loop to drive quantum logic gates).

gkasprow, 2017-09-06:
10K 0.1% 25ppm

hartytp, 2017-09-07:
@gkasprow Let's not bother with the 25ppm/K resistor. Just put a standard resistor here.

General comment: I don't like putting low temp co components on the PCB unless we have a good reason to think they are doing good.

dhslichter, 2017-09-07:
My suggestion here would be to ensure that the resistor footprint and location are such that users could resolder a new one with lower tempco if desired.  Basically 0402 minimum, prefer 0603 if there is space, and with reasonably accessible solder pads.

closed 2017-09-11 by gkasprow