sinara-hw/sinara #276: Urukul SYNC fanout

jordens, 2017-09-05:
* [ ] I don't think that idea of muxing the SYNC fanout onto IDC1:LVDS1 works. The Si53304 says: "When OE is low, Q is held low and Q is held high for differential output formats". The drivers are not disabled, just the pulses. Do we need another FIN1101K8X here? Is there a better suited mux/fan-out chip (CDCLVC1106?) then (since we can't use/don't need the OE feature)? Maybe just stick with the [AD design](http://www.analog.com/media/en/reference-design-documentation/reference-designs/CN0121.pdf). That one has another output we can use.
* [ ] Put two DDS (0/1) SYNC_IN on Bank A (of IC9)  and the other two (2/3) on Bank B for symmetry, equal loading, and aesthetics. ;)

gkasprow, 2017-09-06:
The ADCLK846 that ADI is using:
- do not have input mux, so we won't be able to use LVDS2 signal to sync DDSs.
- is supplied from 1.8V so needs another regulator.

jordens, 2017-09-06:
I guess the CDCLVD1208 is also out of the race because it needs 2.5V. Maybe best to stick with the Si53304 and slap a FIN1101K8X onto that output.

closed 2017-09-11 by gkasprow