sinara-hw/sinara #278: Urukul synchronization

jordens, 2017-09-05:
When routing please do a reasonable (100ps) length match within each of `DDS_SYNC_CLK[3:0]`, `DDS_UPDATE[3:0]`, `RFSW_CTRL[3:0]`, `IO[15:11]` (nominally the RFSW_CTRL lines), `DDS[3:0]_SYNC_IN`, `DDS_RESET` and a really good one on `DDS_CLK[3:0]_[PN]`.

hartytp, 2017-09-17:
- [ ] Should DDS_SYNC_CLK0 be AC coupled to avoid current draw into biasing resistors?

gkasprow, 2017-09-17:
@hartytp I can increase the bias resistor values. I used 220R since they were already in the design.

jordens, 2017-09-17:
IIRC that is only a pretty weak output from the DDS.

But I think @hartytp is not so much referring to the bias resistor value and more to the possivble addition of an ac coupling to convert weak single ended 3.3V CMOS (?) to LVDS. There, I guess, are different options to achieve that. With AC (and zero common mode at the LVDS input) and with DC (and some common mode swing at the LVDS inputs) coupling...

gkasprow, 2017-09-17:
Anyway, 1k load is really light and every CMOS can handle it.

hartytp, 2017-09-17:
> Anyway, 1k load is really light and every CMOS can handle it.

Okay, feel free to leave it as is then.

dhslichter, 2017-09-18:
I am not sure sure about that, check the IBIS models @gkasprow ...the SYNC pins are pretty pathetic on these DDS chips.

jordens, 2017-10-17:
- [ ] You didn't match `DDS_RESET` AFAICT. The branch to the AD9910 does not need to be matched. But the one to the AD9912 does.
- [ ] ~For the RF switch action to be coincident with the DDS signal, the traces from the DDS (both) outputs to the baluns would need to be matched as well...~ But let's delay that for v1.1.

gkasprow, 2017-10-17:
1cm of trace is roughly 50ps. CPLD skew is higher than that.
In case of Reset line I had to add passive source termination. Otherwise this would not have much sense.
I spit it to reset dedicated to 9912 and 9910

jordens, 2017-10-17:
Please elaborate how you implemented this. The `DDS_RESET` (not `DDS_I/O_RESET`) needs to be **coincident at all AD9912** (the AD9910 don't matter). It isn't AFAICT. It comes in at the top and travels to the bottom. The 10cm board height would be 500ps skew between the top and bottom DDS.

gkasprow, 2017-10-17:
@jordens Sure, I just fixed it.
I source-terminated all resets that go to 9912. Otherwise you will get ringing.
9910 reset is routed to another CPLD pin to not load 9912 reset pin.

jordens, 2017-10-17:
Ok. That sounds good. I had interpreted your comment as something that you had already implemented.

gkasprow, 2017-10-17:
@jordens can we produce it as it is? There will always be some issues and every board can be routed better.

jordens, 2017-10-17:
Yes. Please go ahead. AFAICT you addressed all pertinent issues.

hartytp, 2017-10-17:
Thanks for finishing this off Greg! Looking forward to the prototypes arriving.

hartytp, 2017-10-18:
@jordens Can we close this?

jordens, 2017-10-18:
From my view yes. Generally @gkasprow should close them as the one who fixes them. The syntax is [a bit more restrictive](https://help.github.com/articles/closing-issues-using-keywords/).

closed 2017-10-18 by jordens

gkasprow, 2017-10-18:
Well, I used syntax: fixed and then series of issues.

jordens, 2017-10-18:
Correct. You need to repeat the keyword for each issue.