sinara-hw/sinara #279: Urukul Amplifier

jordens, 2017-09-05:
You still have the ERA-4XSM+ with 75/65 mA in the power budget and in the annotations (instead of the ERA-3+ with 35mA). Bias resistor looks ok.

dhslichter, 2017-09-06:
Has phase noise performance of the ERA-3+ been tested?  ERA-4XSM+ has been tested and verified excellent at NIST.

hartytp, 2017-09-06:
Not as far as I know. However, phase noise is less of a priority in this design than power consumption IMO. My preference is to build the prototypes with the lower-powered amps. We can then measure the phase noise and make a change before the production round if necessary.

dhslichter, 2017-09-06:
This is acceptable as long as we are not relying on the lower power consumption of the untested amplifier variant in order to meet the power budget.  I recommend that we plan on using the ERA-3+ but keep the power budget numbers for the higher-current ERA-4 to guard against this possibility.

hartytp, 2017-09-06:
> This is acceptable as long as we are not relying on the lower power consumption of the untested amplifier variant in order to meet the power budget. 

The difference is only ~100mW per channel, so it doesn't make a huge difference to the power budget either way -- enough of a power saving to be worth it IMO, but not enough that we couldn't live with the ERA-4.

If everyone is happy with ERA-3+ for the protoround, shall we close this issue?

closed 2017-09-06 by dhslichter