sinara-hw/sinara #282: Urukul schematics clean up and style

jordens, 2017-09-05:
- [ ] Can you put titles on each page, so it's easier to track what's going on when reading the schematic?
- [ ] There are a few bits of overlapping text (e.g. component values). Just needs a bit of a tidy up at some point.

hartytp, 2017-09-17:
Is there any way to make these Altium PDFs smaller? I always have problems working with them on my laptop...

gkasprow, 2017-09-17:
@hartytp Yes, when I disable Table of content, component parameters, net information and global bookmarks for components and nets, the size drops to 1.5MB. But navigation is slightly more problematic.

marmeladapk, 2017-09-17:
@hartytp Then you're gonna love 3D pdfs Altium can create...

hartytp, 2017-09-17:
> @hartytp Then you're gonna love 3D pdfs Altium can create...

:)

> @hartytp Yes, when I disable Table of content, component parameters, net information and global bookmarks for components and nets, the size drops to 1.5MB. But navigation is slightly more problematic.

Okay, well I'll leave it to you to do what you think is best as I don't have much experience with allium PDFs. I'd guess that a TOC and component parameters are useful to have and don't take up too much space, but the nets are probably not worth it.

hartytp, 2017-09-17:
PS thanks for the quick fixes to those issues!

hartytp, 2017-09-18:
- [ ] The two big red annotations "Place it close to ..." look a bit odd. Please either remove them from the final version, or replace them with a small annotation saying something like "IC4 & IC9 placed close to LVDS buffer."

- [ ] Can we make the "When routing please do a reasonable length match" annotation a bit more succinct (or remove it). Something like "Trace lengths matched within each of DDS_SYNC_CLK[3:0], DDS_UDPATE[3:0], RF_SW_CTRL[3:0], IO[15:11] (RF_SW ctrl lines), DDS_SYNC_IN[3:0], DDS_RESET[3:0], DDS_CLK[3:0]" (please double check this is correct).

- [ ] If any of the LEDs have functions assigned to them already, can we mark them on the schematic (e.g. PWR on, PLL locked etc).

- [ ] Please remove the annotation "This module connects to Kasli ..." and replace it with something like "EEM connector: IO are LVDS, I2C is 3V3 LVCMOS, P3V3_MP up to 20mA, P12V up to 1A." (please copy this to the other EEMs at some point). Motivation for this is that "EEM connector" is not something of a standard, so it shouldn't be tied to Metlino/Kasli anymore.

- [ ] A few symbols misplaced: IC12 SDA, SCL, P3V3_MP; IC15 GND; GND near R54 overlaps with component name; "Digital Attenuator" no longer next to the digital attenuator; GND for R138 a long way from the GND symbol (this is in a few places); 

- [ ] Quite a few of the allium IO symbols (yellow diamond type things with text in) have the text outside the box. e.g. RE_DE, CLK_DIV, DDS_SYNC etc on the CPLD page, IN_SEL and CLK_DIV on the clock page, etc

- [ ] Two resistors labelled R? near IC15.

- [ ] Power budget still has the ADCLK948, rather than the Si chip

- [ ] Annotation "One of two RF filters..." copied from Allaki doesn't make sense here! (No ADC). I'd just say "Jumpers R57/R59 and R58/C28 select between filter options." (And, put this in a nice yellow bubble like the other annotations.)

- [ ] R57 & R58 should be 0R by default to avoid floating metal

- [ ] Can we make the large red annotations like "F clk max = 30MHz" "Amplifier ~23dB gain" either a little smaller, or just put them in yellow bubbles like everything else.

- [ ] Consider removing R33A. I know that it was me who advocated adding this. However, last time I did this using one of these MCL MMICs and an 0603 resistor jumper, the thing oscillated at ~10GHz due to input-output parasitic couplings! (I ended up having to remove the pads and traces with a scalpel). Not sure if it will be a problem for this amp (depends on gain, BW, pad size, etc).

- [ ] C190 and C196 don't seem necessary, as the signals are AC coupled on the other side of the balun.

- I'm still surprised how much power the LVDS interfaces take up. Isn't there a slightly less power hungry solution we could have used?

hartytp, 2017-09-18:
Other than the above (pretty minor) points, I'm happy to sign off on this schematic -- although, I haven't looked too closely at the CPLD/sync etc.

Good work!

hartytp, 2017-09-18:
Initial layout also looks good to me. 

Re screening though: based on our experience with Novo tests, let's aim to test this without screening first. If internal (cross-talk) and external pickup look good then let's aim to operate the prototypes without screening enclosures. If we're happy with it, we can DNP/remove the screening clips for the production hardware.

jordens, 2017-09-18:
C190/C196 are there for the balun case. Not worth the trouble optimizing them away IMHO

hartytp, 2017-09-18:
> C190/C196 are there for the balun case. Not worth the trouble optimizing them away IMHO

ACK. After re-looking at the topology for the TCM2-43X+, I remembered why they were needed. Good catch!

jordens, 2017-09-22:
@hartytp do you want to go through your list of items and check whether they have been resolved and if yes close this?

hartytp, 2017-09-22:
> @hartytp do you want to go through your list of items and check whether they have been resolved and if yes close this?

These haven't been fixed AFAICT. Assuming @gkasprow will tick them off as he completes them as usual.

closed 2017-10-09 by gkasprow

hartytp, 2017-10-11:
@gkasprow Thanks for doing all that!

A few final things (low priority, can be done for the second revision to avoid delays):

- [ ] there is still quite a bit of text in the wrong place in the current version of the schematic. e.g. on the CPLD page, the signal names aren't actually in their yellow boxes (they're offset from them)...

- [ ] Can we rename "EEM A" and "EEM B" to "EEM 0" and "EEM 1" to be consistent with other boards.
 
- [ ] Might be better to change the annotation on page 1 from "Clock from Kasli" and "Internal clock input". That makes the annotation name more consistent with the net names, and makes it clear that this clock could come from other sources, such as clocker.

The layout looks good from the quick look I've had at it. Nice job!

hartytp, 2017-10-17:
@gkasprow If you have time, can you make the above trivial changes to the schematic, please?

gkasprow, 2017-10-17:
I don't have time but I did it :)
So far I only fixed such cases by looking at schematics, not pdf files.
Altium does wrong font scaling. Source document looks fine but on pdf it is shifted. No idea how to fix it.
I enlarged all windows so it should look better

commit 9e5deb5c (2017-09-17): fixed #282, #312, component placement done, routing started