sinara-hw/sinara #284: Kasli EEM power specification

jordens, 2017-09-05:
* [ ] What current are the IDCs/ribbon cables rated to?
* [ ] What current can the copper plane and the barrel connector take?
* [ ] Add a stated limit to the schematics.

gkasprow, 2017-09-05:
@jordens Current rating is [here](https://multimedia.3m.com/mws/media/920107O/3m-tech-brief-current-rating-of-common-3mtm-flat-ribbon-cable.pdf). 1A per wire is standard value and we have 2 wires in parallel. IDC is also specified at 1A per pin.
Barrel connector is specified for 5A. 
In case of copper plane there is no limit - after power integrity simulation one iterates until current density is lower than let's say 50A/mm2.

jordens, 2017-09-07:
Thanks. I updated the [wiki](https://github.com/m-labs/sinara/wiki/EEM#eem-idc-electrical-interface).

commit 6314b8cb (2017-09-17): Kasli fixes
Changed position of some comments and designators - check #304
Updated power budget, fixed #297 fixed #284
Added capacitors for #303
closed 2017-09-17 by marmeladapk