sinara-hw/sinara #287: Kasli CLK_FPGA_IO driving and biasing

jordens, 2017-09-05:
CK_FPGAIO_N also needs biasing. I am a bit worried about the LVPECL swing driving the 1.8 V IOs of the FPGA. Apparently also "all four resistors should be in the 10k-100k range" according to [UG471](https://www.xilinx.com/support/documentation/user_guides/ug471_7Series_SelectIO.pdf#page=93)... 

But IMHO it would be much better to drive CK_FPGAIO from CKOUT2 of the SI5324 (IC2). That output can be switched to LVDS (less complicated divider and biasing) and allows us to leave one output of IC1 (ADCLK944) unterminated saving power. CK_FPGAIO is also more of a safety net (for if IBUFDS_GTE2 has too much delay variations) that we may not need. If we don't need the safety net, we could then just deactivate CKOUT2.

marmeladapk, 2017-09-05:
@jordens: @gkasprow noticed, that IC1 adds non-deterministic delay, so moving CK_FPGAIO to IC2 may be troublesome.

jordens, 2017-09-05:
But that's just 50 ps non-determinism. I don't think that would be problematic.
But anyway, looking at #270, I think it's best to put the fan-out (IC3) on CKOUT2 of (IC2) while keeping everything else (IC1: CK_FPGAIO, CK_GTP, CDR_OUT) on CKOUT1.

jordens, 2017-09-05:
OK. Let's go with
* the crate clock fanout (IC3) driven from IC2 (CKOUT2 -> CLK0) and CLK_BUF_IN (CLK1)
* the CDR fanout (IC1) driven from IC2 (CKOUT1) and driving CLK_CDR_OUT, CK_FPGAIO (with proper biasing, level matching, and termination), and CD_GTP
* Leave the fourth output of IC1 unconnected and unterminated
* Leave all unused LVPECL output pins unterminated

@marmeladapk Maybe extrapolating from [XAPP696](https://www.xilinx.com/support/documentation/application_notes/xapp696.pdf) is helpful for connecting the CDR clock fanout to CK_FPGAIO. In any case, I'll leave the details to you.

marmeladapk, 2017-09-08:
Done in latest commit, but we dodged a bullet here. I had to change 1V8 to 2V5.

closed 2017-09-08 by marmeladapk

jordens, 2017-09-08:
Because of the matching/biasing constraints of CD_FPGAIO?

jordens, 2017-09-08:
@gkasprow could you sign off on the CK_FPGAIO biasing/temrination/level matching?

marmeladapk, 2017-09-08:
@jordens No, turns out 7 series FPGAs support 1V8 LVDS only on HP (High Performance) banks, and Artix 7 only has HR (High Range) banks that support 2V5 LVDS. 1V8 LVDS could be received in HR banks with use of external term. resistor, but LVDS output to EEMS could have wrong levels, current or would not work.

Now we can use internal terminating resistor.

jordens, 2017-09-09:
ACK.

jordens, 2017-09-09:
I just now realize what kind of bullet we dodged here!
But from what I can see both 1.8 V and 2.5 V LVDS on the Kasli/Metlino/Sayma side are compatible with the the different things that we have on the other side (sn65mlvd040, Si53304, FIN1101K8X etc) in both directions.

jordens, 2017-09-09:
As soon as @gkasprow signs off on the schematics, we are good to go ahead with layout and routing. I don't see any outstanding issues.