sinara-hw/sinara #290: Urukul T-pad before amplifier

jordens, 2017-09-05:
With about 1dBm out of the DDS, 0.5 dB insertion loss from the Balun, 0.5 dB from the lowpass, 1.5 dB from the attenuator, we need a 9dB T-pad to attenuate that before the ERA-3+ with 23 dB gain and P1dB of 13 dBm at our frequencies.

gkasprow, 2017-09-05:
@jordens Would [such](https://industrial.panasonic.com/cdbs/www-data/pdf/AOJ0000/AOJ0000CE2.pdf) attenuator do the job? Or you prefer discrete one made of resistors?

hartytp, 2017-09-05:
Out of curiosity, what's the benefit of using that kind of attenuator? Is that cheaper than using resistors? Or, just more accurate/lower temp co?

jordens, 2017-09-05:
resistors are fine.

gkasprow, 2017-09-06:
Such attenuator gives better frequency response because has less parasitics. It's hard to make good resistor attenuator working in a few GHz range mainly due to pad capacitance.

jordens, 2017-09-06:
We are looking at 500 MHz Max here.

dhslichter, 2017-09-06:
We don't want the attenuator to have reduced attenuation at harmonics/Nyquist images if possible.  Unless cost is prohibitive, I would advocate for packaged attenuator.

hartytp, 2017-09-06:
> We don't want the attenuator to have reduced attenuation at harmonics/Nyquist images if possible. Unless cost is prohibitive, I would advocate for packaged attenuator.

A 9dB T-pad made with 0201 resistors will still be pretty close to 9dB at 1GHz, so this isn't a concern IMO. Given the filters and other components, I wouldn't expect an 0201 T-pad to be the main cause of lack of gain-flatness in the pass-band either.

Argument against the packaged attenuator is mainly that it's one more non-standard BOM line to worry about.

dhslichter, 2017-09-06:
Most standard resistors have terrible tempco.  If we care at all about amplitude stability of the output, we'll want to use thin film NiCr instead of thick film anyway.  Plus placing 0201 parts adds considerable expense.  I guess it seems to me like a shame with all the engineering that is going into this to cut corners on something pretty simple like this.

hartytp, 2017-09-06:
> Most standard resistors have terrible tempco

True, tempCo is more of a reason to use this component that frequency response IMO. My guess is that a packaged attenuator (10c in quantity) is cheaper than ~25ppm/K resistors. As these boards could get quite hot, this is a good reason to consider using these attenuators.

>  Plus placing 0201 parts adds considerable expense. 

Really? I though this was pretty standard for pick and place machines these days, but maybe I'm wrong?

dhslichter, 2017-09-06:
> My guess is that a packaged attenuator (10c in quantity) is cheaper than ~25ppm/K resistors. As these boards could get quite hot, this is a good reason to consider using these attenuators.

That's my feeling for sure.  Plus you will probably get something that is co-fabricated on one die so you have better matching of tempco (and thus more stable relative attenation) for free.  

>>Plus placing 0201 parts adds considerable expense.

>Really? I though this was pretty standard for pick and place machines these days, but maybe I'm wrong?

Most of the shops I order from grumble about 0201, and you usually need to supply 100% component overage instead of 20% overage for the pick and place.  They can certainly be routinely done, but they will often charge more.  

A reasonable comparison is something like [the YAT series](https://www.minicircuits.com/WebStore/dashboard.html?model=YAT-9%2B), which clock in at ~$2.50 each for our quantities.  These Panasonic attenuators may not have great tempco because they use thick film resistors.  For low tempco we want thin film like the YAT attenuators.  Small footprint, 2 mm x 2 mm.

jordens, 2017-09-07:
Let's leave this to @gkasprow to decide what's best value for money.
Regarding tempco there are multiple components which need to all be good: RSET, DDS, switchable attenuator, fixed attenuator, amplifier bias.

hartytp, 2017-09-07:
Amplitude temperature sensitivity analysis:

**AD9910** (DDS)
- Not specified on data sheet (also not specified for other DDSs like the AD9912 and AD9914). Didn't find much info out there, but see e.g. [https://ez.analog.com/thread/70661](https://ez.analog.com/thread/70661).
- Assume it's similar to AD9154 so ~100ppm/K?
- Also depends on Rset

**HMC542** (step attenuator)
- Not specified on data sheet, didn't find anything on the web

Other components...

Conclusion:
- It's far from clear that putting low temp co components (by which I mean better than the standard +-100ppm/K) on this PCB will make any improvement to the overall temp co
- We could measure these numbers using eval boards but, given that amplitude stability is not a critical design goal for this project, I don't think it's worth the hassle!
- My recommendation: stick with standard temp co components for everything and don't worry about this for the time being (can always improve in a future revision).

hartytp, 2017-09-07:
> Let's leave this to @gkasprow to decide what's best value for money.

Agreed, let's just go for the cheapest option with a temp co <~100ppm/K and be done with it.

dhslichter, 2017-09-07:
I am fine with this, as long as we picking better than ~100 ppm/C.

dhslichter, 2017-09-07:
As was just mentioned in #275, I would suggest we try to pick an attenuation circuit here such that users could have the ability to replace with a lower tempco part after the fact if they desire it for their particular application.  I am envisioning someone using this board as a one-off in a small temperature-stabilized case, with additional heat sinking, as we do for the magtrap, for example.  Basically this just means picking an attenuator with a footprint such that other parts with lower tempco could be substituted in if desired.

dtcallcock, 2017-09-07:
The ERA-3+ amplifier gain variation is -0.005dB/C.  This is about 1200ppm/C so will dominate.  Some other Minicircuits parts are a factor of 2-5 better.  

We could also use this T-pad attenuator to compensate for the amplifier's tempco (and the tempco of the rest of the RF chain).  See,
[Example1](http://www.thin-film.com/uploadedcontent/documents/ATV_Presentation.pdf), [Example2](https://www.nidec-copal-electronics.com/e/catalog/attenuator/e-ta.pdf), [Example3](https://media.digikey.com/pdf/Data%20Sheets/Susumu%20PDFs/PXV_PBV_Series.pdf), & [Example4](http://www.emc-rflabs.com/Passive-Components/Thermopad).

dhslichter, 2017-09-07:
I like this idea, cheap/simple way to mitigate the other thermal drifts.  I think even if amplitude stability is not the main design goal of this board, it's nice to have for many potential users if we can implement simple/cheap things that help it.

dhslichter, 2017-09-07:
If we go this route, I would vote for the 0805 sized attenuator packages (4 pads) because there seem to be many options there.

gkasprow, 2017-09-07:
@dhslichter This would work providing that we thermally couple  such attenuator with amplifier die.

dhslichter, 2017-09-07:
I think physical proximity on the board will be enough for our purposes.

gkasprow, 2017-09-11:
I used available PAT1220 0805 attenuator.

closed 2017-09-11 by gkasprow