sinara-hw/sinara #291: Urukul thermal simulation

jordens, 2017-09-05:
According to the current power budget Urukul is burning 10-12 W max.
Are we convinced that convection across 100mm and 4HP is enough?

gkasprow, 2017-09-05:
@jordens In MTCA 100W is the limit. Don't thing we would have issue in 3U chassis with 10W, but forced airflow may be needed when we pack a few DDS boards in adjacent slots.

hartytp, 2017-09-05:
Do we need heat sinks on the board? Or will the airflow over the PCB be enough?

dhslichter, 2017-09-06:
Heat sinks on DDS chips seem like a good idea if there is room.

hartytp, 2017-09-06:
> Heat sinks on DDS chips seem like a good idea if there is room.

Not sure that will be very effective once the screening cans are in place...

dhslichter, 2017-09-06:
Forgot about screening cans.  If done properly, DDS should be well heat sunk with thermal vias under the pad, then you can put heat sinks on the back side of the board to make good thermal contact.

dhslichter, 2017-09-07:
It would be nice to have a few mounting holes and clear (no soldermask) metal on the backside to allow efficient placement of a custom machined heat sink/stabilization block that users can implement themselves.

jordens, 2017-09-07:
Thanks for the suggestion While nice conceptually, I'd like to leave that for the next revision. I don't want to risk our deadline for this.

dhslichter, 2017-09-07:
I am OK with waiting for the next revision, just wanted to suggest it would be good to keep in mind when placing components.

jordens, 2017-09-24:
#321 
Needs 50 cm/s flow worst case.

closed 2017-09-24 by jordens

gkasprow, 2017-09-24:
We can use small[ heat sinks glued directly to the PCB](http://www.farnell.com/datasheets/1448934.pdf?_ga=2.56563819.1752915978.1506280963-162313899.1464604133). From bottom side we have max 3mm of height, so their efficiency would be very limited. Mechanical mounting would take too much space.

jordens, 2017-09-24:
I'd just donate 2 or 4 HP space on the left for a bigger heatsink.

gkasprow, 2017-09-24:
@jordens I assumed worst case current consumption for all components with their max ratings. So it won't be that bad.
Anyway, I will do temperature measurements after we make them running full speed.