sinara-hw/sinara #292: Urukul CPLD prototype code

jordens, 2017-09-05:
Before we send of the boards to manufacture, we should have a protoype of the CPLD logic ready and compiling so that we know it can in principle implement all the required functionality. We also need to know typical timing delays through the CPLD and the variation (pin-to-pin and jitter) on those.

We would want to have the CPLD code in Migen. There is a basic template for this [here](https://github.com/m-labs/cpld_test). Some integration into Migen is required still.

gkasprow, 2017-09-05:
@jordens I treat CPLD as part of hardware - I use it to replace bunch of discrete logic ICs.
I can write core in VHDL or even draw schematics using  graphical symbols of TTL chips and compile using xilinx ISE.
It is trivial - a few registers and gates.
I don't know Migen, never played with it and don't want to invest my time at this stage of the project because have much more important tasks to do.
For debuging purposes I will anyway prepare my own version of CPLD code. You are free to do your own Migen code later on.

jordens, 2017-09-05:
ACK. But we still need a way to define and agree on the logic.

gkasprow, 2017-09-05:
I will draw schematic in ISE so everybody would be able to quickly get the intention.

jbqubit, 2017-09-05:
Schematically, what is the reason for using a CPLD for Urukul? I gather it will be implementing #273. Is this simply an alternative to using discrete logic components on the PCB?

gkasprow, 2017-09-05:
@jbqubit Exactly. CPLD chip is cheaper and mitigates the risk.

gkasprow, 2017-09-08:
@hartytp @jordens [here](https://github.com/m-labs/sinara/blob/master/ARTIQ_ALTIUM/Kasli/3U/PCB_3U_DDS/CPLD_code/Urukul_CPLD/Urukul_CPLD.vhd) is _very_ initial CPLD code. Just to check if it fits and compiles.

jordens, 2017-10-26:
I implemented the CPLD functionality the way I'd like to see it used in the beginning. There were a few changes to the NU-Servo pinout to simplify the logic but nothing dramatic. The uses 76% of the CPLD macro cells.

https://github.com/m-labs/urukul

https://github.com/m-labs/urukul/releases/download/v1.0rc1/urukul_v1.0rc1.tar.gz

closed 2017-10-26 by jordens