sinara-hw/sinara #294: Move SYNC_IN EEM drive from EEMB:LVDS2 to EEMA:LVDS7

jordens, 2017-09-07:
I consolidated the EEM pin assignments and cut them down to what's really needed and what appears to be the best way of achieving the goals. [Urukul Wiki](https://github.com/m-labs/sinara/wiki/Urukul#eem-connector-usage)

Most of it is just soft CPLD programming. The only change that I'd like to see in the current schematics is moving the SYNC_IN drive of IC9:CLK0 from EEMB:LVDS2 (IDC1:LVDS2) to EEMA:LVDS7 (would be IDC0:LVDS7).

Rationale: in the one-EEM, "simple" case, for both AD9912 and AD9910 the relevant line driving synchronization (DDS_RESET and SYNC_IN) from the FPGA is now on the first EEM. Also all driven synchronization-relevant signals are now on EEMA and all received signals on EEMB. This should allow using the AD9910 "simple" variant fully synced as well.

hartytp, 2017-09-07:
Happy to leave these details to you.

Out of curiosity:
- Does SYNC_OUT connect to the AD9910 SYNC_OUT pin (i.e. it's an input on the FPGA)? If so, no SYNC_IN in the final version for AD9910? So, plan is to reset FGPA gateware to sync with DDS, rather than other way around?
- No DDS MISO for the servo use-case? I thought you were very against this?
- What does AD9910 (b) CS do?

jordens, 2017-09-07:
The FPGA should (and can) drive sync. Otherwise we can't do a proper drtio slave for example.

Yes. There won't be DDS miso in that wide interface. But it's a dip switch away. And I would like to have it available on the spi bus on the second EEM. That also resolves the issue of initial configuration of the DDS. As you can seen there is pin shortage.

The spi bus in the second eem talks to the cpld which then acts a bit as a spi router. It routes access to the attenuators, implements the switch/led/clock sell shift regs etc.

hartytp, 2017-09-07:
> The FPGA should (and can) drive sync. Otherwise we can't do a proper drtio slave for example.

Am I missing something? In the current version of the wiki, only SYNC_OUT is connected to the EEM connector (not SYNC_IN). This is an output from the DDS, not an input that the FPGA should drive. 

> Yes. There won't be DDS miso in that wide interface. But it's a dip switch away. And I would like to have it available on the spi bus on the second EEM. That also resolves the issue of initial configuration of the DDS. As you can seen there is pin shortage.

> The spi bus in the second eem talks to the cpld which then acts a bit as a spi router. It routes access to the attenuators, implements the switch/led/clock sell shift regs etc.

Okay, that sounds sensible (although, I'd need to see a full plan of the CPLD routing to check that I fully understand you).

gkasprow, 2017-09-07:
@hartytp I started sketching CPLD schematic but VHDL code would be much easier to maintain. Do you have any objection?

hartytp, 2017-09-07:
> @hartytp I started sketching CPLD schematic but VHDL code would be much easier to maintain. Do you have any objection?

No objections at all! Use whatever language you like.

In the end though, @jordens has thought the most about this, so I'm happy to defer to him on CPLD usage.

jordens, 2017-09-07:
@hartytp It's the old unresolvable labeling issue. An "IN" to one thing is necessarily an "OUT" to something else. The long form description defines directions.

@gkasprow Editable textual representations of the code are strongly preferred. And if you don't want to learn Migen, then Verilog. In the end we will probably end up porting it to Migen anyway during testing...

jordens, 2017-09-12:
That should go to EEMA:LVDS7 (IDC0:LVDS7 in your notation), not EEMB/IDC1.

gkasprow, 2017-09-17:
Right!

commit d4a4cd9d (2017-09-11): fixed #294 #276 #275 #293 #290
closed 2017-09-17 by gkasprow