sinara-hw/sinara #295: Use Si53312 as clock fanout/mux/divider

jordens, 2017-09-08:
from #270 

There are two wiring options here: 

- A:
  * All AD9912 to the A-bank,
  * All AD9910 to the B-bank,
  * terminate only the ones in use in that variant
  * DIVA open, DIVB GND

- B
  * All AD9912 and AD9910 to the A-bank on top of each other
  * Wire DIVA to the CPLD
  * LEave the B-bank unused

A has the advantage of zero stubs on the clock lines but uses a bit more routing. And it frees up a CPLD line and possible source of errors. B uses less routing, has stubs (either on the clock pair or on the analog output).

hartytp, 2017-09-08:
@jordens I'd definitely like a way of disabling the divider for the AD9910 e.g to allow a 1GHz clock input. So, even in option A, I'd still route DIVB to the CPLD.

hartytp, 2017-09-08:
Other than that, I'd go for option A if we can route it without having to add additional layers.

jordens, 2017-09-08:
Oh yes. I forgot about that.

gkasprow, 2017-09-10:
option A implemented. closing.

commit 7a05a549 (2017-09-10): fixed #295, #277, #273, #271, #270,
closed 2017-09-10 by gkasprow