sinara-hw/sinara #296: Kasli component choice

jordens, 2017-09-10:
* in the standard BOM, DNP J20 and J24
* around IC6, AFAICT the compensation component values are not critical and can be chosen with tolerances
  * drop R64, R80

hartytp, 2017-09-10:
@jordens Can we keep J20 (BP connector) in the standard BOM, please?

For our 16-channel NU servo, we need 2xNovo=4 EEM connectors and 4xUrukul=8 EEM connectors. So, we need a total of 12 EEM connectors, meaning that we _have_ to use the IO lines on the BP connector.

Otherwise, we'll have to order a slightly custom version of Kasli with the BP connector populated (and the barrel connector DNP) for NU servo.

jordens, 2017-09-10:
You have to choose between the barrel connector and the DIN connector. Since we don't have a backplane designed, we also don't have a way to power Kasli over that backplane yet.

You can just desolder the barrel connector and then solder in the DIN connector. That's hand-solderable.
Also this is the prototypes run.

jordens, 2017-09-10:
Novogorny only takes one EEM by the way.

hartytp, 2017-09-10:
> You have to choose between the barrel connector and the DIN connector.

Yes, I'd forgotten that we removed the barrel connector from the front panel due to lack of space.

Out of curiosity, are we definitely happy not having a barrel connector on the FP? I haven't got much experience using EuroRacks but AFAICT the rear barrel connector may be quite hard to access in some racks.

> You can just desolder the barrel connector and then solder in the DIN connector. That's hand-solderable.
> Also this is the prototypes run.

Okay, we can do that for our boards.

> Novogorny only takes one EEM by the way.

I thinking of the "fast Novogorny" (or whatever we're calling it these days). IIRC, that's 2 IDCs.

dtcallcock, 2017-09-11:
> Out of curiosity, are we definitely happy not having a barrel connector on the FP? I haven't got much experience using EuroRacks but AFAICT the rear barrel connector may be quite hard to access in some racks.

If this is a problem you could add a panel-mount barrel connector (or any other connector) to a 2HP filler panel and run an internal cable to the Kasli.

jordens, 2017-09-11:
The barrel connector is just as hard to access as the DIN connector. We'll place the power supply in the back of the rack as usual.

jordens, 2017-09-11:
Correct. Fast Novo occupies two EEMs. What's the reason you can't use two Kasli?

NB: You appear to be unaware (or don't care) what you voted for, what the consensus on the naming issue was, and where the responsibility for the current naming lies now after changing the name. That again strengthens the argument that the names were/are already as good as they get and that the problem can only be with "not learning names".

hartytp, 2017-09-11:
> NB: You appear to be unaware (or don't care) what you voted for, what the consensus on the naming issue was, and where the responsibility for the current naming lies now after changing the name. [...]

"Forgot"/"Slipped your mind" would have been more accurate :)

FWIW, I thought the plan was to obsolete the current Novo once the fast version is out -- price should be similar for the two boards, and the slow version doesn't have any features the fast version is lacking. Thus, when I refer to "Novo" I'm generally always thinking of the fast version. (But, if you feel strongly about this, let me know and I'll adopt whatever convention everyone else is using).

> What's the reason you can't use two Kasli?

Why use two if we can use one? More money, rack space, power consumption, wiring, etc. But, yes, if we have to, we can use 4 Kasli for our 32 channel NU system, I'd just prefer to use 2 if possible.

> The barrel connector is just as hard to access as the DIN connector. We'll place the power supply in the back of the rack as usual.

Okay, that's certainly an option. FWIW though, the more I think about it, the more I think a FP barrel connector would be a really useful convenience feature.

Putting an idea out there: what if we dropped the clock fanout input SMA from the Kasli front panel? We could replace it with an MMCX on the PCB near the fanout buffer. This might give us enough space to add a barrel connector on the FP.

My thinking here is that the main motivation for this SMA was distributing a very low noise clock (either an externally supplied 100MHz, or the CDR clock after passing through an external PLL). However, for these purposes, the low-cost, low-noise fanout EEM we've built is probably better suited. So, IMO, there isn't a strong argument for this SMA taking up valuable FP real estate any more.

That might free up enough space for the power supply connector.

I wondered if we could also relegate the micro-USB from the FP to being internal only to free up some more space? Depends on how much people expect to use it...

> If this is a problem you could add a panel-mount barrel connector (or any other connector) to a 2HP filler panel and run an internal cable to the Kasli.

Definitely an option. But, wastes some rack space, requires a new panel to be supplied, and is generally a bit of a faff. Let's avoid this unless really necessary.

jordens, 2017-09-11:
Names being easily forgotten or slipping peoples minds was the main complaint about them. There is no way out.

I'm ok with "Novogorny" referring to the current single lane SPI bus and "Novo" referring to the upcoming one with quad lane source synchronous data.

I am ok with killing the entire clock fanout on Kasli. That would mean only one SMA on the front (*either* CDR_IN or CDR_OUT with jumpers). But having the fanout input internal would force us to use the other clock module to drive that fanout. That doesn't make sense to me.
To make the clock fanout really low noise, it would likely need a dedicated LDO anyway.
The micro-USB should remain on the FP. And I prefer to supply Kasli from the back of the rack or from within the rack.
To make the frontpanel @dtcallcock is referring to you'd just buy a blank one and drill a hole into it. It also allows you to supply both Kaslis with one front panel.

hartytp, 2017-09-11:
> I am ok with killing the entire clock fanout on Kasli. That would mean only one SMA on the front (either CDR_IN or CDR_OUT with jumpers). But having the fanout input internal would force us to use the other clock module to drive that fanout. That doesn't make sense to me.

Sorry, my post wasn't clear. It's a mux, so it has two inputs. My suggestion was to keep the current situation where one input comes from the Si clock recovery chip (as in the current situation). That would serve (what I anticipate being) the most common use-case of Kasli distributing the RTIO/CDR clock. IMO, it's worth keeping the clock buffer on Kasli to serve this common use-case, whatever we do with the second input.

The second input to the mux is less important, but could be an MMCX on Kasli. Or we could kill it.

hartytp, 2017-09-11:
> And I prefer to supply Kasli from the back of the rack or from within the rack.
> To make the frontpanel @dtcallcock is referring to you'd just buy a blank one and drill a hole into it. It also allows you to supply both Kaslis with one front panel.

Okay. I still think that a FP power input would be a useful convenience feature, that would make life easier for people. e.g. if you put a single Kasli in a box, it's nice for all connections to be on the same face.

But, I'll leave the final decision to you.

If we don't go for the FP power input then we might as well keep the fanout buffer clock input SMA on the FP.

jordens, 2017-09-11:
Let's leave it and reconsider after the prototypes. Then we know more.

gkasprow, 2017-09-11:
@hartytp @jordens When I was testing VHDCI carrier, I realised that the rear side power connector was not very convenient, so I added front panel one without asking.

marmeladapk, 2017-09-11:
We may do something like this:
[image]
and leave connector on the back.

hartytp, 2017-09-11:
> @hartytp @jordens When I was testing VHDCI carrier, I realised that the rear side power connector was not very convenient, so I added front panel one without asking.

Great! Good idea @gkasprow!

My vote (FWIW) is:
- Leave the barrel connector on the back. My preference is to DNP the real barrel connector and populate the 96-pin DIN, but I'm not too fussed about that, as this is easy to change when the boards arrive.
- Add a second barrel connector on the front.
- Leave the clock fanout SMA input on the FP if there is room. If not, remove it and replace it with an internal MMCX. (As I said before, the main use-case I want to cover with this buffer is conveniently supplying the RTIO/CDR clock, so this SMA/MMCX is an optional-extra IMO).

hartytp, 2017-09-11:
@marmeladapk For convenience for the user, it might be nicer not to put the power connector between the SMAs if possible. Instead, I'd have the power connector near the bottom edge of the board, and the two SMAs together. Otherwise, LGTM.

jordens, 2017-09-11:
For the prototypes I'd like:
* the barrel connector in the back
* no DIN connector
* you may add a barrel connector on the front panel IFF you show that it does not interfere with the SMA mounting tool minimum dimensions, both of the SMA connector mounted, and with regular size Micro USB connector mounted

But please let's not unnecessarily loose time on this.

hartytp, 2017-09-11:
> no DIN connector

To clarify, assuming this means: DNP the DIN connector (i.e. don't remove the pads!).

jordens, 2017-09-11:
Yes.

hartytp, 2017-09-11:
> you may add a barrel connector on the front panel IFF you show that it does not interfere with the SMA mounting tool minimum dimensions, both of the SMA connector mounted, and with regular size Micro USB connector mounted

Given Greg's comments, I still think the SMA is significantly less important to have on the FP than the power connector. But, agreed, one way or the other, it's time to make a decision on this and move onto layout.

gkasprow, 2017-09-11:
What would be clock source for KASLI? The clock buffer or some external source?
Anyway, when somebody needs SMA connector for ext clock, can add SMA to MMCX jumper and fix it to 3U panel.
So I'd use just internal vertical MMCX connector for clock input and then leave one SMA only.  One can add another MMCX instead of solder jumper.

hartytp, 2017-09-11:
> What would be clock source for KASLI? The clock buffer or some external source?

To check we're all on the same page: there are currently 2 SMAs on Kasli.
1. External input to Si chip/output from Si chip. **This one is the one I want to keep.** When Kasli is used in "master" mode, this input is required to supply it with a 100MHz reference clock.
2. External input to the ADCLK948 clock fanout buffer (other input to this buffer is an output from the Si chip). **This is the SMA I propose we drop from the FR and replace with an internal MMCX**. This will not be used in many cases, but can be used to supply an external clock to the fanout buffer.

> Anyway, when somebody needs SMA connector for ext clock, can add SMA to MMCX jumper and fix it to 3U panel.

That sounds like a good compromise to me! My guess is that the FP power connector will be used more commonly than the clock buffer external input, so it makes more sense to but the power connector on the FP by default.

@jordens Would you be happy with this solution?

jordens, 2017-09-11:
That amounts to:

1. make J19 an internal MMCX
2. add a locking barrel connector to the FP in parallel with the one on the back (J17)

I am ok with that.

hartytp, 2017-09-11:
Great! Thanks. Sounds like we all agree on the solution to this.

marmeladapk, 2017-09-16:
@jordens I wanted LD1 to be red because it indicates loss of pll lock on Si5324.

jordens, 2017-09-16:
That doesn't warrant a new BOM line IMHO. Just make it green. Good silk screen will do the rest.

gkasprow, 2017-09-16:
the LG L29K-G2J1-24-Z LED is no longer available.
PLCC LEDs are large, you should use 0603 or 0402 LEDs everywhere.
In CERN lib there are KPH-1608CGCK but they have reversed footprint, I asked CERN to fix it, you can also modify the footprint.

sbourdeauducq, 2017-09-16:
The firmware is already monitoring that 5324 lock continuously (and reporting on UART), it doesn't really need a LED.

jordens, 2017-09-17:
@marmeladapk But J25 should be placed. Otherwise there is a hole in the FP.

marmeladapk, 2017-09-17:
Oh, I thought we wouldn't include it in FP in versions without backplane connector (so with rear power connector).

jordens, 2017-09-17:
Designing, manufacturing, selling, maintaining etc two front panel types sounds more complicated than having that unused connector.

hartytp, 2017-09-17:
I agree with @jordens: definitely keep the FP power connector in both versions.

hartytp, 2017-09-17:
Why is there still an "ADD LEDS" annotation on the front page? Can that be removed now?

jordens, 2017-10-04:
@marmeladapk How is the layout going? Anything we can help you with?
@gkasprow Do we have a timeslot in the assembly line for Kasli and Urukul?

gkasprow, 2017-10-05:
I can reserve assembly line for Urukul in 2 weeks. For Kasli it's too early.

jordens, 2017-10-05:
@gkasprow Ok. We are really under a deadline to get the boards back (in our hands) by the end of the month. Can we do anything for Urukul or Kasli to accelerate?

jordens, 2017-10-05:
And from a tactical perspective, we should focus on Kasli because on my end doing anything with the other boards without Kasli will be mostly pointless.

gkasprow, 2017-10-06:
@jordens We need support in HDL.
Before routing is started, one must make sure that current pin assignment is fine and won't cause troubles like it was in case of Sayma. Pawel spent plenty of time building design only to make sure that i.e. SDRAM controller will work with current memory connectivity.
I asked him to skip this stage and finish routing ASAP, but someone must implement some example design to be sure we won't have troubles i.e. with missing clock domains.

jordens, 2017-10-07:
But you said that the sdram pinout can be just taken from one of the existing boards.

Please tell me earlier next time that something like this needs to be done and is pending.

I'll do an example design this weekend. What needs to be checked? And what issues with sayma are you referring to? The Ethernet phy clocking issue won't show up here.

gkasprow, 2017-10-07:
@jordens We can copy from devkit, but there is different chip in different package. So theoretically should work. It's not that urgent - we need to have it verified before board goes to production.
Mainly clocks would be an issue. Recently I had to rewire some board because wanted to use general purpose clock routed to SDRAM bank and the memory controller consumed all clocking resources...

jordens, 2017-10-11:
@marmeladapk @gkasprow I wrote an initial [Kasli](https://github.com/m-labs/migen/blob/master/migen/build/platforms/sinara/kasli.py) [design](https://github.com/m-labs/misoc/blob/master/misoc/targets/kasli.py) to test some basic things like clocking and sdram. Apart from the currently open issues, we are ok with freezing it and pushing hard for manufacture. Let us know if there are other items.

hartytp, 2017-10-23:
@marmeladapk It looks like all items here are resolved, can we close this issue?

commit 1e9e545f (2017-10-26): PCB routing
Fixed #331
Closed #296
closed 2017-10-26 by marmeladapk