sinara-hw/sinara #298: DDR termination

jordens, 2017-09-10:
The [Arty development board](https://reference.digilentinc.com/_media/arty:arty_sch.pdf) from Digilent uses the same DDR3L chip class (just smaller) and an Artix 7 but:

* runs at 1.35V
* does not use termination resistors (uses ODT termination presumably) or a termination voltage regulator

Maybe given the crowding this is a good option?

gkasprow, 2017-09-10:
DDR3L can be run without termination providing that it is verly close to the FPGA.
With Artix chip one won't be able to get full speed anyway, so this may work.

marmeladapk, 2017-09-11:
> ODT is only applied to each of the following balls: DQ[7:0], DQS, DQS#, and DM 
Lines that are terminated right now are not covered by ODT, we'll leave them, those resistors won't take much space.

We may change 1V5 to 1V35, but it shaves off just 90 mW.

marmeladapk, 2017-09-11:
@jordens IT means you're fine with leaving 1V5 and termination resistors?

jordens, 2017-09-12:
Yes. You are more qualified to decide that.