sinara-hw/sinara #299: Kasli VCCO

jordens, 2017-09-10:
Right now we have 2.5 V IO voltage.
Would it be wise or possible to move to 3.3 V IO voltage?
I know this is pretty late, but the change from 1.8 V to 2.5 V just happened.

* We'd save a bunch of 2.5 <--> 3.3 level translators.
* It would free up one power supply rail, making room for VCCAUX/VCC_XDAC (#297)
* LVDS is not affected AFAICT.
* Maybe a bit higher power consumption.

marmeladapk, 2017-09-10:
> Assigned #299 <https://github.com/m-labs/sinara/issues/299> to
> @marmeladapk <https://github.com/marmeladapk>.

> The "LVDS_33" I/O Standard that was available in some older FPGA
families, is not supported in 7 series.
> Neither High Range (HR) banks, nor High Performance (HP) banks can have
their VCCO pins powered at 3.3V (if using LVDS outputs).

[Source](https://www.xilinx.com/support/answers/43989.html)

jordens, 2017-09-10:
Ah. Never mind then.

closed 2017-09-10 by jordens