sinara-hw/sinara #303: AC coupling on CLK_CDR_IN

gkasprow, 2017-09-10:
CLK_CDR_IN clock input is not AC coupled - R9 and R13 make DC loop

jordens, 2017-09-13:
On the other hand, CKIN2+- are probably internally biased (I couldn't tell from the datasheets but they are only used AC coupled). You can't short that with R13. That one should be DNP. And without the balun, there should be a 0R to GND in front of C14.

marmeladapk, 2017-09-15:
@jordens You're right, though [this document](https://www.silabs.com/documents/public/reference-manuals/si53xx-reference-manual.pdf#G1251443) mentions: 

> AC coupling the input clocks is recommended because it removes any issue with common mode input voltages. However, either ac or dc coupling is acceptable.

I'll add capacitor before balun for sma and mmcx connectors.

jordens, 2017-09-16:
But the implied statement is that you have to obey the common mode voltage i.e. the biasing.

marmeladapk, 2017-09-16:
That's absolutely right.

commit 773f740d (2017-09-13): Kasli fixes
Moved R9 and R13 after C5 and C14, fixes #303
Moved R51 to pullup Power Good signal, fixes #301
Fixed #300
Replaced R40, fixed #305
Fixed #302
Added clock input annotation (for #306 )
Added 1V8 buck (IC23), fixed #297
closed 2017-09-17 by hartytp