sinara-hw/sinara #306: Kasli clock input annotations

hartytp, 2017-09-11:
- [ ] Add a brief annotation for the solder jumpers on the SMA clock input for Kasli. Annotation should say something like "For clock input ("master" mode), populate ..., for clock output ("slave" mode) populate ..."

jordens, 2017-09-13:
"DRTIO master" and slave are well defined.
All correct. Just that in master mode Kasli gets the clean clock over that connector. The jitter attenuation effect is not important in that case. In slave mode Kasli recovers the clock from the serial link to the upstream DRTIO master and that needs cleaning. And it is convenient to then output that clock on the connector for other purposes.

marmeladapk, 2017-09-13:
Please check if current note is verbose enough.

hartytp, 2017-09-14:
Current annotation looks good to me. Thanks!

closed 2017-09-14 by hartytp