sinara-hw/sinara #309: Variant DNPs

jordens, 2017-09-12:
Make the DNPs on the DDS_OUT_Channel and CLK_INPUT pages the same in both variants.

jordens, 2017-09-17:
* [ ] I don't think it makes sense to populate the series parts of the discrete filter with 0R. Just DNP them.
* [ ] The loop filter components (C107, C113, R90, R92) should be placed in both variants (but with different values).
* [ ] DNP the DDS_CLK_P/N termination resistors in the unused variant.
* [ ] Place L9, C101, C102 by default to make adding the oscillator easier.
* [ ] Either beef up the the 35 mA bias resistor R137 (burns 120 mW) or shrink the R18||R19 temination (burns 20 mW).
* [ ] The bias current annotation near the amplifier is still outdated.

gkasprow, 2017-09-17:
@jordens I left these 0R just in case somebody needs direct path. I made them DNP
R137 is already 0.5W one

jordens, 2017-09-18:
ACK.

closed 2017-09-18 by jordens

hartytp, 2017-09-18:
>  I don't think it makes sense to populate the series parts of the discrete filter with 0R. Just DNP them. (When used, there would usually be inductors).

Also, I generally wouldn't want to leave these bits of metal floating, as you can get funny resonances if you're not careful (the amps work into the GHz regime). So, keeping the 0R resistors is a good idea IMO.

jordens, 2017-09-18:
What are you envisioning here? You get resonances whether you have open ends or grounded ends. There are loads of floating things there. Mounting holes, fiducials, NC pads... The way the discrete filter is wired now (the two mux-jumpers half-DNP and the rest series 0R) makes things worse than with everything DNP.

hartytp, 2017-09-18:
> There are loads of floating things there. Mounting holes, fiducials, NC pads... The way the discrete filter is wired now (the two mux-jumpers half-DNP and the rest series 0R) makes things worse than with everything DNP.

ACK. The resistors should be populated here to ground this metal. IME grounded metal in/above a ground plane causes fewer problems than floating metal.

jordens, 2017-09-18:
Well. With all the 0Rs you have exactly the shape of a directional coupler bypassing the filter. With everything DNP you'd have just a bunch of islands with fF coupling. But I'm happy with the 0Rs if you feel strongly about them.

hartytp, 2017-09-18:
> Well. With all the 0Rs you have exactly the shape of a directional coupler bypassing the filter. With everything DNP you'd have just a bunch of islands with fF coupling. But I'm happy with the 0Rs if you feel strongly about them.

I don't feel too strongly about it, but somehow leaving this floating seems wrong to me. Happy to be convinced otherwise ;)