sinara-hw/sinara #310: Clock input baluns

jordens, 2017-09-12:
Use the same baluns for the clock inputs as for the DDS outputs (once those have been changed to center-tapped #272 ). That makes the ones at the clock inputs so cheap that we should just populate them in both variants.

hartytp, 2017-09-12:
ACK.

Can't remember which PN we're using, but if we're going to put them on the clock inputs, let's make sure they are very broadband (<<100MHz to >1GHz) to preserve our nice sharp rising edges on the clock inputs.

dhslichter, 2017-09-12:
Agreed with @hartytp, I worry about the ability of center-tapped baluns to pass sharp clock edges.  In general a transmission line/Guanella balun would have considerably better bandwidth, and would likely be a better choice.  

Right now the output balun on the schematic is a 2:1 transmission line balun.  We were planning to change that to a center-tapped balun (#272) as on the [AD9910 eval board](https://static5.arrow.com/pdfs/2009/9/10/19/9/32/656/adi_/auto/ad9910_schematics.pdf).  That part is a 1:1, [ADT1-1WT](https://www.minicircuits.com/pdfs/ADT1-1WT.pdf) which only goes to 800 MHz -- fine for the output, but suboptimal for a clock input.  That same eval board uses an [ETC1-1-13](https://cdn.macom.com/datasheets/ETC1-1-13.pdf) for the clock input, which goes to 3 GHz.  I think it's definitely worth the extra BOM item in this instance.

hartytp, 2017-09-12:
The balun is not essential here, and my guess is that an 800MHz balun will do more harm than good when supplied with a nice square-wave clock.

My vote is to either leave the balun out (DNP + add 0R jumper to bypass) or add an extra BOM line (let's stick with the TCM2-43X+ we use everywhere else).

jordens, 2017-09-12:
We need to run the DDS also at low frequencies and don't want to have TLTs on the outputs. Those will be center-tapped. In that case leave the clock inputs with the baluns DNP and single ended.

closed 2017-09-12 by jordens