sinara-hw/sinara #311: Kasli schematic clean up

hartytp, 2017-09-17:
- [ ] I find the annotation "Replace 0R with 100k for AC coupling" a bit misleading -- to me "AC coupling" implies AC coupling of a signal, rather than the ground. IMO something like "Replace 0R with 100k to isolate grounds." (or at least "to AC couple grounds.") would be better. If there is room, it might be worth adding "Replace 0R with 100k **and fit insulating washer on SMA** to isolate grounds." (but that's getting a little verbose!).

hartytp, 2017-09-17:
Other than this and the issues raised in #296, I'm happy to sign off on the schematics (although, I haven't looked in detail at the FPGA, RAM, etc).

marmeladapk, 2017-09-17:
> Current consumption for the ADCLK948/ADCLK944 in the power budget table looks wrong - shouldn't it be lower because not all outputs are used, and not all negative outputs are biased?

I made maximum, worst case power budget (for example when user populates DNP resistors). I'll change values for ADCLK944, since one output is not connected.

hartytp, 2017-09-17:
> I made maximum, worst case power budget (for example when user populates DNP resistors). I'll change values for ADCLK944, since one output is not connected.

Okay, that makes sense. Thanks!

hartytp, 2017-09-18:
Thanks for fixing that so quickly!

commit a8966dbd (2017-09-18): Kasli fixes
Fixed #311
Changed LEDs #296
Initial components placement
Initial pin swapping on FPGA
closed 2017-09-18 by hartytp