sinara-hw/sinara #315: Kasli PSUs

hartytp, 2017-09-19:
- [ ] Find and test a high quality PSU, which we can ship with Kasli and the VHDCI carrier (and, maybe clocker as well).

- +12V
- 5A
- High-quality, with proper over temperature and overcorrect protection, decent CMCs etc
- Mates **properly** with the connectors on VHDCI carrier/Kasli (cf ). If we have trouble finding something like this, let's change the connectors on these boards.
- Available from Farnell/DigiKey/etc

dhslichter, 2017-09-19:
The connector should be locking if at all possible.

hartytp, 2017-09-21:
Are there any space ADCs on the Kasli FPGA? If so, we should consider using one to monitor the total +12V current draw and/or the +12V supply voltage (could be useful for debugging issues when users draw too much current from the EEMs).

jordens, 2017-09-21:
there are ADCs for monitoring the FPGA voltages and termperatures. for external stuff there are no pins anymore.

hartytp, 2017-09-21:
Okay. Would have been a nice diagnostic, but not worth adding an external ADC IMO.

jordens, 2017-09-21:
the FPGA has a decent internal ADC which we would have used for that.

hartytp, 2017-09-21:
> the FPGA has a decent internal ADC which we would have used for that.

Yes, but all channels are already used (unless there is something else we're ok to stop monitoring to free up an ADC channel?).

jordens, 2017-09-21:
There are plenty of channels still unused (all the `AD*[PN]` in banks 15 and 35). But their pins are used. You can still use them from EEM3 and EEM5 if somebody needs that.

hartytp, 2017-09-21:
ACK, thanks for the correction.

Still, unless there are some pins currently used for non-essential purposes, which we're happy freeing up, we still can't use this ADC for input supply current/voltage monitoring.

hartytp, 2017-09-22:
@jordens What about adding a low-cost I2C ADC onto Kasli for 12V V/I monitoring?

Motivation: power supply droop due to excessive current draw can lead to some really nasty, hard to debug issues. We have some quite thirsty EEMs, so it would be easy for a user to overload Kasli without realising it. The additional BOM cost on Kasli due to an ADC is pretty tiny, so IMO, this could be worthwhile. I'm suggesting I2C, as it allows us to add the ADC to an existing bus without consuming more pins on Kasli.

jordens, 2017-09-22:
And then do what with that ADC data? To protect against overloads, the proper solution is a fuse.
To measure 12 V precisely, you'd take a Novogorny channel and connect that to the 12 V with a divider, right? That seems to be what Novogorny was designed for. Also, what about all the other overload situations? Like all 4 outputs of one of the DIO module group shorted overloading the 500 mA DC/DC modules? What about transient overloads that fall between ADC samples?
All the coding overhead of handling the ADC and the data properly is significant and not worth the gain compared to the other solutions. The LED on the POWER_GOOD is good enough IMHO. And aren't there more tricky situations that are our attention right now? I expect power supplies to fold back when overloaded. That should trip the DC/DC regulator.

hartytp, 2017-09-22:
> And then do what with that ADC data? ...

Okay. One thing that might be worth doing instead at some point in the future (e.g. when automatic device_db generation is implemented) is to get Kasli to check which EEMs are connected and their maximum (specified) current consumption. This could give a warning if the total max current exceeds 5A...

jordens, 2017-09-22:
ACK.

gkasprow, 2017-09-22:
We can use I2C EEPROMs to hold such information. Then simple EEPROM query will do the job.
This of course does not cover situation where one of the EEM is broken and draws more current than needed.
In case the software detects possible overload, it won't initialise the EEMs which will keep dissipated power minimum.

gkasprow, 2017-09-22:
For quick evaluation of the loading we can use simple threshold comparator which detects voltage drop below let's say 11.5V. POWER GOOD is good for embedded DC/DC converters only and they will work even with 6V at their inputs.

jbqubit, 2017-09-22:
Detecting supply voltage transients is helpful at debugging setups being used in the lab. This is especially true for EEMs where there's a single supply shared by heterogeneous devices not all of which may be 
"official" Sinara hardware. @gkasprow A threshold compactor with a short time constant connected to a latch would be ideal. Latch holds memory of low-voltage state and can be readout (and reset) as needed by FPGA.  Observation of the latched state indicates to the user that there's been a transient condition that warrants more extensive debugging.

gkasprow, 2017-09-22:
the same applies to VHDCI carrier.

jordens, 2017-09-23:
Let's not go beyond what we have (LEDs for power good from the switching regulators) in this version of kasli. The suggestion is too late, this is of uncertain value, not important, or useful enough to me and I estimate the delay needed for discussing, implementing, testing, and coding this to be unacceptably large.

hartytp, 2018-01-28:
Now that we're starting to use EEMs in earnest, I wanted to bring this up again.

@gkasprow Can we get a batch of PSUs made up (e.g. by technosystem) which are:
- Decent (e.g. medical spec) quality, high reliability and not too noisy
- +12V and can produce 5A without problem -- I don't want one that starts screaming or getting noisy at 4.5A, so maybe it's worth aiming for a 75W PSU or something
- Not overly expensive
- Have a sensible IEC on one end (I like the standard 3-pin ones) and the locking barrel connector on the other end.

I assume that all users of Kasli would like to buy some of these -- we'd certainly like to buy some if the price is reasonable.

hartytp, 2018-01-28:
Apart from the convenience of not having to reconnectorize PSUs ourselves, I like the idea of having a standardised PSU to go with Kasli. That way, once these cards start getting more widely used, when issues come up we won't have to start asking "What PSU are you using? Is your issue caused by a dirt cheap and noisy PSU." etc

gkasprow, 2018-01-28:
@hartytp ...and the power output would be via front panel or from rear side or both?

gkasprow, 2018-01-28:
The easiest way to go is to use COTS resonant AC/DC module that has >90% efficiency, attach IEC connector, Power LED and suitable DC jacks.

gkasprow, 2018-01-28:
And the mains should be connected to the IEC connector directly, without using PCB tracks for better safety in the crate.

gkasprow, 2018-01-28:
the module size must be: 
H < 36mm
W < 93mm
L < 160mm including IEC connector.
For power rating > 60W it's hard to find conventional enclosed one that fits such constraints.
We can use one I use in uTCA mini chassis. It is resonant topology (efficiency>90%) but open frame so some mechanical protection must be added.
We can also use completely enclosed 3U box and fit all the stuff there. In this way we limit the switching noise propagation.

gkasprow, 2018-01-28:
[This ](https://www.tme.eu/pl/Document/eac8f1626bde852d15cd1b53d00d1bd8/top100-datasheet.pdf)is the module I mentioned

gkasprow, 2018-01-28:
Such [enclosure]( 24813314) should do the job.

jordens, 2018-01-28:
I have been using a bunch of the modules I mentioned in the wiki for a while now and they are just fine. Low ripple, doubly insulated, small euro plug (no earth) tiny capacitive coupling, beefy and sufficiently long cables, solid choke, right connector.

jordens, 2018-01-28:
And Poe modules are also really nice for smallish Kasli setups for the obvious reasons of monitoring, cabling and remote management. The ones on digikey are fine as well.

hartytp, 2018-01-28:
@gkasprow I'd originally been thinking of a wall wart. But, actually you raise a good point that having a rack-mount module is probably the way to go.

What about:
- Rackmount 3U PSU module: 4HP? 8HP? 160mm deep
- IEC on front pannel
- +12V output on front pannel via locking barrel connector
- Second +12V output at the rear on some connector that can interface with Kasli BP adapter to provide power to Kasli. In the next version, we should make the VHDCI_breakout compatible with the Kasli BP adapter, so the same PSU can be used to supply the VHDCI_breakout as well (in that case, the BP EEM connectors will all be NC apart from power).

hartytp, 2018-01-28:
> I have been using a bunch of the modules I mentioned in the wiki for a while now and they are just fine. Low ripple, doubly insulated, small euro plug (no earth) tiny capacitive coupling, beefy and sufficiently long cables, solid choke, right connector.

Or, do you mean the wall wart on the Wiki. I'm also totally fine with using those. Either way, so long as someone can sell us a bunch of decent PSUs with the right connector, which we can encourage all users to use then I'm happy.

gkasprow, 2018-01-28:
@hartytp with such 3U 8 HP module we can fit 120W supply and 2 DC jack sockets so 2 or more carriers can be supplied...

hartytp, 2018-01-28:
> @hartytp with such 3U 8 HP module we can fit 120W supply and 2 DC jack sockets so 2 or more carriers can be supplied...

Good to know. We will put two carriers in one of our racks. Although, I would have been equally happy just using two PSUs!

In that case, maybe put a few power connectors on the FP so this can also clock things like clocker.

Anyway, having a separate rack-mount PSU module sounds good to me so long as the cost is not too much larger than using a high-quality wall wart. @gkasprow are you happy to produce this kind of PSU?

klickverbot, 2018-01-28:
The rack-mount version indeed sounds nice for the case where a whole 19" rack is taken up by two Kaslis + EEMs.

For smaller free-standing crates, having one of these PSU bricks by Mean Well, XP Power, etc. seems more economical in terms of space.

hartytp, 2018-01-28:
> For smaller free-standing crates, having one of these PSU bricks by Mean Well, XP Power, etc. seems more economical in terms of space.

Probably true. But, even then loosing 4U is generally okay.

If people feel strongly about it, we could offer two options:
- A reconnectorized high-grade wall wart
- A Eurorack mount PSU module

gkasprow, 2018-01-28:
To make it low cost, it needs to be a piece of PCB with all connectors soldered. Manual assembly is quite expensive. So let's take standard 12HP plugin module (+30EUR), good PCB mounted AC/DC and single layer PCB. We will get nicely shielded resonant PSU.
We can fit within 8HP but without inner PCB.

hartytp, 2018-01-28:
hmm... 12HP is a bit large. Certainly prohibitively large use in a small rack. Even in a big rack, it's a bit big. For example take our planned 32 channel AOM driver, we have:
- 2 x Kasli: 2 x 8HP=16HP
- 4 x Sampler: 4 x 8HP =32HP
- 8 x Urukul: 8 x 4HP = 32HP
Total: 80HP

Which only leaves 4HP in a standard rack (which is one reason I originally objected to taking Kasli to 8HP).

hmm...maybe we're best of going for simply picking a COTS PSU brick and getting TechnSystem to reconnectorize them and ship them with Kasli.

gkasprow, 2018-01-28:
We could make Kasli 6 HP and then have 8 HP for common supply.
standard IEC connector won't fit on 4HP panel. We can use laptop-type C5/C5 IEC connector that would fit, but cables are less popular.
I have nice AC/DC modules that are only 15mm high but are quite expensive.

hartytp, 2018-01-28:
I'd be fine with 6HP Kasli. Any objections @jordens?

hartytp, 2018-01-28:
If you can fit a good 120W PSU in 8HP, that sounds good to me.

What would users prefer to buy? 3U PSU modules or PSU bricks? (Or would they buy both in roughly equal quantities?)

cjbe, 2018-01-28:
I vote for PSU bricks only. 

I don't see what problem 3U PSU modules solve that the bricks don't - does anyone really want to advocate for 3U PSU modules?

hartytp, 2018-01-28:
@cjbe for everything I'm planning to do a compact (8HP or less) 3U module is more convenient than zip-typing a PSU brick to my racks. But, if the question is whether that extra convenience is worth the hassle of designing a 3U module, then probably not.

Either way, so long as we have a solution that TechnoSyste can provide that plugs directly into Kasi without users having to get out a soldering iron then I'm happy.

jordens, 2018-01-28:
I'm with @cjbe here. For the ~dozen of Kasli crates that we are setting up we'll use bricks. Not having to deal with mains voltage (testing, regulatory, emissions, grounding the case, fuses, certification) is another factor. Those bricks are not that much bigger than the power cord and the UK power connector.
If you want to go for high-density, then maybe use the backside of the rack (gives you 84 HP). That's also a perfect place to put a power supply.
Or save some space on Sampler? Don't populate the BNC connectors, use 4x4 HP panels, and add a 4-HP-wide 4xIDC-HD68 plus break-out box on the table? That saves you 24 HP in the rack and reduces the arm-thick RG58 trunk to a finger-thick HD68 cable.

hartytp, 2018-01-28:
> I'm with @cjbe here. For the ~dozen of Kasli crates that we are setting up we'll use bricks. Not having to deal with mains voltage (testing, regulatory, emissions, grounding the case, fuses, [...]

Fine. Let's just go for reconnectorized PSU bricks.

@gkasprow are you happy to get TechnoSystems to reconnectorize a bunch of nice PSU bricks, like the one Robert linked to on the Wiki. IIRC, you said that they'd only charge something like 1EUR to do the reconnectorization. 

> Or save some space on Sampler? Don't populate the BNC connectors, use 4x4 HP panels, and add a 4-HP-wide 4xIDC-HD68 plus break-out box on the table? That saves you 24 HP in the rack and reduces the arm-thick RG58 trunk to a finger-thick HD68 cable.

Good point. Our plan is probably to put the samplers above the table in a mini rack and use a pair of VHDCI_carrier cards as IDC <-> VHDCI adapters to allow us to hook these up to a Kasli in a separate rack.

gkasprow, 2018-01-28:
Let's ask Technosystem first. All depends on batch size :)
I didn't think about such use case of VHDCI but will work. Only I2C ID chips won't be accessible that way. We can fix it in next revision.

hartytp, 2018-01-28:
So, the other question then is how many people would actually be interested in buying decent PSU bricks with the right locking barrel connector from TechnoSystem/having them shipped with Kasli. Let's assume for the time being that the overhead charged by TechnoSystem is small compared with the cost of the PSUs.

As I said, apart from being convenient for users, I also see this as having an advantage in terms of making people's setups more standardised and hence somewhat easier to debug.

If everyone just wants to sort out their own PSUs then there's no point worrying about this and we can close the issue.

dtcallcock, 2018-02-02:
> How many people would actually be interested in buying decent PSU bricks with the right locking barrel connector from TechnoSystem/having them shipped with Kasli.

I would definitely want this to ship with Kasli.  For a handful of cards this seems to be the way to go.

Given there will potentially be [many Eurocards](https://github.com/m-labs/sinara/wiki/PlannedHardware) with the same barrel connector though, I'd like to see a beefy Eurocard or 19" PSU with a bunch of outputs eventually.   I'm sure it can wait until these cards see the light of day though.  

> Not having to deal with mains voltage (testing, regulatory, emissions, grounding the case, fuses, certification) is another factor. 

We could just get someone like [Acopian](https://www.acopian.com/single-l-rack-m.html) to make one to our specs and avoid these concerns.

hartytp, 2018-03-16:
AFAICT the locking barrel connectors won't work with the current Kasli (not much thread exposed). That's fine IMHO.

FWIW, we're currently using XP Power AKM65US12, which work well with Kasli with no modifications. [Farnell link](http://uk.farnell.com/xp-power/akm65us12/adaptor-medical-ac-dc-12v-5-42a/dp/2801007?ost=AKm65US12&ddkey=http%3Aen-GB%2FElement14_United_Kingdom%2Fsearch)

Anyone object to us adding these to the BOM for  Kasli/VHDCI_carrier and requesting TechnoSystem to ship them with future purchases?

jordens, 2018-03-16:
I'd very much prefer one with c7/c8 mains connector like the one I proposed elsewhere. Smaller, higher density in many countries, lower capacitive coupling, double insulation (with the class ii)

hartytp, 2018-03-16:
There's a version of the PSU we use with C7/C8. I personally prefer the one we have, since it's more common in the UK. If technosystems ship Kasli with the right mains lead to go with it, I'm not fussed though.

hartytp, 2018-03-17:
@jordens can we agree on [AKM65US12C2](https://www.digikey.co.uk/product-detail/en/xp-power/AKM65US12C2/1470-4458-ND/7897378) as our standard PSU and get @gkasprow to add it to the BOM for Kasli/VHDCI carrier.

- Let's not ship a mains lead with this, as it will go to too many different countries. 
- Let's not bother with locking barrel connectors. If users really care, they can sort it out themselves.

Or, if we can't agree on this, let's just close the issue.

jordens, 2018-03-17:
Hmm. If that's what Technosystem will offer, then we'll decline and replace with it what we have been using so far: the [GSM90B12-P1M](https://www.digikey.de/product-detail/de/mean-well-usa-inc/GSM90B12-P1M/1866-2031-ND/7703586) that was proposed a couple months ago. Has lower ripple, more power margin, C7/C8 connector, and could even be ordered with the locking barrel connector which is usable with the retaining nut in Kasli removed.

hartytp, 2018-03-17:
Is more power margin a good thing? Kasli is only rated to 5A and an 80W psu will put out about 7A when short circuited. May be more likely to cause damage.

That aside, assuming your psu mates well with the barrel connectors I'm fine with it. Let's just pick one and add it to the BOM.

jordens, 2018-03-17:
I'd be surprised if 1.67 A more margin pose a realistic threat. We also don't protect against overcurrent due to short circuit on a single IDC.

Maybe there are other things we should look at which are more important. E.g. as I mentioned in the other issue about the kasli clock noise, the floating nature of kasli and isolated or AC coupled I/os on the eems will lead to an issue with almost every psu due to the nonzero leakage current (i.e. all bricks). I don't know whether it's dangerous (300v peak stress on the coupling caps and/or 100uA through clamp diodes) but it is observed frequently and easily and due to the lack of grounding it's not obvious how to eliminate it. In practice grounding the sfp cages or the barrel connector using dirty tricks will cut it but it's not nice. The only devices that will ground a crate are zotino and grabber AFAICT.

gkasprow, 2018-03-17:
We can connect the screws that hold the panels to GND. In this way all modules would be connected to the chassis providing that panels are equipped with EMC springs.

jordens, 2018-03-17:
Yes. That might be a really good solution. But that's a bit tricky as long as the panels are anodized and don't have those EMC springs.

gkasprow, 2018-03-17:
We will do it with new releases of the boards once the issues are eliminated and there won't be any changes that affect the panels. Schroff panels with EMC gaskets and springs are quite expensive when produced in low quantities.

gkasprow, 2018-03-17:
Obviously the old panels could be replaced providing that holes match.

hartytp, 2018-03-17:
Okay, I confess that I hadn't realised that Kasli isn't grounded to the rack by the fp! We should definitely do that in future.

hartytp, 2018-03-17:
@gkasprow are you happy with the pay @jordens is suggesting? If so can you add it to the BOM for Kasli/VHDCI carrier and close this issue please?

gkasprow, 2018-03-17:
Added to VHDCI carrier and Kasli. @marmeladapk please pull the Kasli design befor you commit any changes.

closed 2018-03-17 by gkasprow

jordens, 2018-03-17:
But connecting Kasli/EEM ground to the chassis is not a complete solution either. If the crate is not grounded explicitly or if the rack that the crate is mounted in is not grounded we are where we started. Connecting Kasli/EEM GND to the crate at every front panel and have the crate connected to other systems via screws that may or may not cut through the anodization might even be bad. Is that the recommended way?
Maybe adding a ground connector to the front panel is the actual well-defined solution there. Then people can decide.
I don't really know what the best solution is...

[7 more replies]