sinara-hw/sinara #316: Kasli power supply decoupling

jordens, 2017-09-20:
[Xilinx UG483](https://www.xilinx.com/support/documentation/user_guides/ug483_7Series_PCB.pdf#page=16) mentiones that you need a bunch of decoupling caps for our FPGA (XA7A100T-FGG484). I can't find that in current schematics (e.g. any of the 330 µF, 100 µF, 470 µF caps and many of the smaller ones). Could you verify the size, quantity amnd ESL/ESR? Maybe this was forgotten in the change to the larger size.

closed 2017-11-02 by marmeladapk

jordens, 2017-11-02:
* VCCAUX should get 3x 4.7 µF
* MGTAVCC should have 1x 4.7 µF
* MGTAVTT should have 1x 4.7 µF

I am not too spooked about those since there are a bunch of 22 µF close by. Just to confirm that you chose to deviate from the datasheet deliberately.

And as mentioned elsewhere C91 is unlikely to survive on the bottom.

For the next revision we should consider either splitting all 47 µF into 22 µF or merging/upgrading all 22 µF to 47 µF to save a BOM line.

marmeladapk, 2017-11-02:
Each one of those is missing one 4,7 uF and has 22 uF nearby, it was intentional.

commit c6d0bec8 (2017-11-02): Kasli - small improvements in routing for PI
Added note about LVDS_0 clock input capability
Changed part number of J17 and J25 (didn't change 3d model yet), fixed #345
Fixed #316