sinara-hw/sinara #318: Kasli - MMCX too dense

gkasprow, 2017-09-20:
placement of MMCX connector is too dense - one won't be able to plug or unplug them.
[image]

jordens, 2017-09-20:
I had assumed that was a preliminary placement. ;)

marmeladapk, 2017-09-20:
@jordens is right, I just moved mmcx out of the way to place other components. :)

jordens, 2017-09-25:
@marmeladapk how is the routing going? Do you think we need to reconsider some of the design for routability or is it doable with 6 layers?
And, can we close this one?

marmeladapk, 2017-09-25:
I'm currently working on sample fpga project to see if it is able to properly route everything (dram controller, ibert cores, etc.) with given pinout and I'm applying changes to it accordingly.

As for trace routing, it should be doable when I space out idc connectors. 

Feel free to close this issue.

jordens, 2017-09-25:
Feel free to space them out horizontally as much as you want/can. That acctually makes mounting the connectors easier. Even moving the SFPs to the top and the SMA/power/USB to the bottom would be fine.

closed 2017-09-25 by jordens

sbourdeauducq, 2017-09-26:
@marmeladapk Considered using Migen/MiSoC? Then we could just reuse your code for the ARTIQ port.
https://www.wdj-consulting.com/blog/migen-port.html

jordens, 2017-09-26:
@marmeladapk I don't think IBERT controllers will be a problem. They are small and independent of other features. The DRAM pin assignment should also be harmless since IIRC you said that you took the schematic from an existing design.

marmeladapk, 2017-09-26:
@sbourdeauducq Yes, however I have already written a script to generate all ports, pins, iostandards etc. in Vivado so I can now just drop some ipcores. TBH switching now to Migen/SoC (and learning it) would be too time consuming, and my priority is layout. However I plan to dive in once I'm done with routing. Thank you for this article, it'll be great for learning!

@jordens But I did some pin swapping. And it turns out Memory Interface Generator imposes strict rules for pinout which I was not aware of (which probably help with timing and routing signals).

gkasprow, 2017-09-26:
@marmeladapk in case of memory, there are several restrictions about tiles and DQS pins. Just copy assignments from Terrasic board.