sinara-hw/sinara #319: Urukul - SMA connector selection

gkasprow, 2017-09-21:
@jordens The SMA connector used in Urukul has good RF characteristics (SMD assembly) but it is too close of the panel edge. If we want to use the insulation washer, the hole will go beyond the panel edge.
[image]

I want to change it to standard right angled ones which does not have such problem. I used them in the clocker module.
I don't think we will notice any issues at 400MHz.
It would look like this:
[image]

jordens, 2017-09-21:
That's perfectly fine with me. The angled ones are good.
Please don't do a staggered layout.
Also, are the SMD and the angled ones the same as those on DIO_SMA?
Why don't we have that problem with the edge SMAs on DIO_SMA?

gkasprow, 2017-09-21:
Staggered layout was to show the difference :)
We did have such problems on DIO SMA, that's why I want to change these connectors.

jordens, 2017-09-21:
ACK. Good!
But are the angled ones the same ones used on four of the SMA DIO channels? Just because I have them in front of me.

gkasprow, 2017-09-21:
Yes.