sinara-hw/sinara #320: Sayma servo monitoring taps

jbqubit, 2017-09-21:
In @jordens [Servo proposal](https://github.com/m-labs/sinara/wiki/Servo) an outstanding question was how to implement monitoring taps within the signal chain for debugging. This is a helpful feature on the Leibrandt NIST [digital servo](https://github.com/nist-ionstorage/digital-servo). Can this now be resolved?

hartytp, 2017-09-21:
> In @jordens Servo proposal 

Was this a proposal (for what we should do), or more of a list of things that could potentially be done with the hardware?

It's important to decide from the outset what we want to achieve with this servo. e.g. do we want to support high-speed logic for demodulation etc? This has a large effect on the rest of the design (implementation of profiles etc).

hartytp, 2017-12-01:
This issue is now obsolete, given the ADC discussions; if we only have 1MSPS ADCs on the AFEs then the only servo that makes sense is basically a port of the NU-servo. Discussion about that should happen via the ARTIQ GitHub page.

closed 2017-12-01 by hartytp