sinara-hw/sinara #321: Urukul - initial routing

gkasprow, 2017-09-21:
@jordens The routing is done. It is still preliminary. Once you accept it, I will finish cosmetics, pour polygons and start PI & SI & Thermal analysis. The bottom side of the PCB below DDS chips is free so one can use it to glue a heatsink.

hartytp, 2017-09-21:
Maybe too late to do anything about this, but it would be nice to standardise FP layout a little bit between the EEMs. In particular, it would be nice if both Clocker and Urukul had their clock input SMAs in similar locations.

jordens, 2017-09-21:
@gkasprow excellent! will do.

hartytp, 2017-09-21:
I'm out of the lab atm with no access to Altium, and it's hard to follow this kind of thing on the PDFs. I'll have a quick look, but can @cjbe or @klickverbot have a more careful look using Altium and sign off on this, please?

jordens, 2017-09-21:
Yes @gkasprow  could you add Gerbers?

gkasprow, 2017-09-21:
@jordens I added Gerbers. I also updated pdfs.

gkasprow, 2017-09-21:
@hartytp It's easier to change the clock input on clocker module than on DDS.

hartytp, 2017-09-22:
@gkasprow Had a quick look over the PDFs, and it generally looks like a good layout.

To do a better check, I'd really need Altium to follow what the traces are connected to etc.

hartytp, 2017-09-22:
Closing, as this seems to have been superseded by #322

jordens, 2017-09-22:
This is the issue for doing the review. It should be closed when done with the review. The other issue(s) are those arriving from the review. Note the different assignees.

jordens, 2017-09-22:
@hartytp I am done at the rough level @gkasprow requested. I'll leave this open for you or others who want to review.

gkasprow, 2017-09-23:
Here is how the voltage drop looks like for 1.8V rail. This is not optimised PCB.
So with existing layout the board would probably not work or work in not reliable way. Instead of 1.8V the DDS chips would get 1.5V in the worst case, under assumption that with lower voltage they still consume same current.
I post this example to show how important design verification is :)
[image]

More critical is current density. Without optimisation we exceed the limit 10 times. It means that sooner or later the copper will overheat and may burn.
[image]

gkasprow, 2017-09-24:
And after optimisation we get 39mV of drop
[image]

gkasprow, 2017-09-24:
I did simple thermal simulation. The conditions are below:
[image]
And the results:
[image]

gkasprow, 2017-09-24:
The same conditions but flow rate reduced to 10cm/s:
The board will switch off the power supply at 80deg.
[image]

Fow rate 200cm/s
[image]

gkasprow, 2017-09-24:
The [24572-407 cooling unit](https://schroff.pentair.com/en/schroff/24572-407) which Joe ordered, generates 36m3/h of flow. After [translation](http://www.comairrotron.com/airflow-unit-conversion)  assuming that we have rectangular duct of 160x480mm, we get 130cm/s of air speed. So we should be fine providing that  we block unused slots. For MTCA there are special dummy modules that provide such function, there should be similar for 3U.
Another issue is supply of the fans (12V, 0.7A), maybe it should be connected to the same power source as the crate to avoid operation of EEMs without cooling.

gkasprow, 2017-09-24:
at 130cm/s the temperature distribution is here:
[image]

jordens, 2017-09-24:
Power (assuming the other rails are similar) and thermal look ok to me.
Is this board/part/copper temperature or air boundary layer?
And all the coupling via exposed pads and copper planes is included?
We could consider rotating the four channels by 90 degrees together. That seems to give about 5-10 degC more headroom. But likely not worth the trouble.

* [ ] We should move IC12 as high up as possible so that it sees more of the relevant peak temperature.

jordens, 2017-09-24:
@gkasprow Also could you review the component choices from the perspective of the testing that you would like to do with the AD9912 prototype boards before you send them to us (testing with ADI software+LVDS driver, etc)?

hartytp, 2017-09-24:
> @gkasprow Also could you review the component choices from the perspective of the testing that you would like to do with the AD9912 prototype boards before you send them to us (testing with ADI software+LVDS driver, etc)?

@jordens @gkasprow What is the plan for testing the AD9910 variant? @gkasprow Can you verify the performance using one of the prototype boards we've paid for (this will need to be done for the NU-servo project).

hartytp, 2017-09-24:
> Another issue is supply of the fans (12V, 0.7A), maybe it should be connected to the same power source as the Kasli/VHDCI to avoid operation of EEMs without cooling.

That's quite a lot of current, given that our PSU is only 5A.

Since these boards have over temperature shutdown, it might be better to leave the fans independent of Kasli.

gkasprow, 2017-09-24:
@jordens This is very rough thermal simulation. It takes into account copper distribution but IC models are very simplified.
For testing I will use 9910 and 9912 devkits together with TTL to LVDS board. Not all features could be tested (i.e. synchronisation) so to make tests complete I'd need ARTIQ driver.
I will need to produce both variants to do the tests. Which means additional tooling cost because component population is quite different - too many of them to do it manually.

jordens, 2017-10-10:
The testing of the AD9910 synchronization only needs setting of the registers. If the ADI tool gives you access to those that should be fine.

The new routing looks good!
* A few stray wires around still.
* I guess the SYNC distribution if matched yet.
* And there is no space for clips to shield IC19? (It's OK if there isn't, just making sure we understood each other).
* Is it OK to power the SYNC distribution IC4, IC9, IC16 from P3V3A and not from P3V3? They are not noise sensitive and more of an aggressor on P3V3A IMHO. (IC19 and OSC1 should definitely stay on P3V3A).

gkasprow, 2017-10-10:
@jordens I didn't finish routing yet.

jordens, 2017-10-10:
I know. ;) Just providing some early feedback.

gkasprow, 2017-10-16:
@jordens I didn't shield IC19 because there is not much to shield. Most tracks go on bottom or on mid layers. Frankly speaking, I believe that we won't need any shields at all.
I replaced names of J1A and J1B. 
The IC4,9,16 are separated by ferrite beads, but it's true, they can be supplied from 3V3. It simplifies routing and makes 3.3VA rail less loaded.

closed 2017-10-17 by gkasprow

jordens, 2017-10-17:
@gkasprow Thanks! Do you want to have us take another look at it (we'd need a git tag and the complete data package uploaded to the release)? Or can we slot it into the production now?

commit 1cf5cf26 (2017-10-17): Production ready - fixed #321 #278 #322