sinara-hw/sinara #323: EEM 2.0

jordens, 2017-09-22:
* IMHO 10 GND is Overkill. The LVDS pairs end up all over the place on the connector (GND, *P, *N etc). And they are differential, i.e. they don't flow current through ground. To return the current from SDA, SCL, 2x12V, 3.3V MP, 3 GND would be sufficient.
* Also the grounds are far away from the power supplies, creating loops.
* The IDC cable seem want the cable come out on the right if pin 1 is down. I.e. the other way around from now.
* 30 pins seems to be a rarer selection on IDC cables and connectors.
* IDC connectors with strain relief (that fold-over thing with the additional wing) are higher than 4 HP

even | odd
--- | ---
0P | 0N
1P | 1N
2P | 2N
3P | 3N
4P | 4N
5P | 5N
6P | 6N
7P | 7N
SCL | SDA
3V3 | GND
12V | 12V
GND | GND

With pin 1 (0P) being at the bottom of the PCB, cable coming out to the right.

gkasprow, 2017-09-22:
@jordens the GNDs are to separate differential pairs.
If you route them without GND, then they are no longer differential, because the return current wont enclose between P and N pins but will propagate to adjacent LVDS pair. This will create mess from SI point of view. 
In such mode, with 2 wires separated by one ground, one can work reliably with digital signals up to 1 at least. Popular PCIe  gen 1 adapters use 2 separation GND wires to ensure good SI.
So let's don't do that.
One can get 30pin IDCs easily from Farnell, Digikey in large variety.
And less popular form is an advantage here - it won't tempt people to plug other pieces of equipment that uses more popular IDC forms :)
I mounted [right angled latching IDCs](https://www.digikey.com/product-detail/en/3m/N3440-5302RB/MHR30N-ND/2183059) on some existing boards and it's easy to do. But anyway we have some boards that use double IDCs so this approach wont work.

jordens, 2017-09-23:
Ack. I forgot the separation argument.

gkasprow, 2017-09-23:
There are also 3M stacked latching IDC connectors, so we can make rev2 of all dual-IDC boards compatible with them. In this way one can install everywhere except Kasli and VHDCI carrier such IDC cables with strain relief.
Moreover, with angled IDC connectors, we can plug them to the backplane directly.

hartytp, 2017-09-24:
@jordens Given Greg's comments, what do you want to do about this? (If we're going to change the EEM Connectors, let's do it before Kasli/Urukul V1, otherwise, let's close this issue)

jordens, 2017-09-24:
No. We can't do this before Kasli/Urukul V1. This should be synchronized more. I want to keep this open for later.
For now I am OK with the current state plus work arounds (filler panels right of Kasli/VHDCI).
For later I'd want to rotate the IDC connector and put the power supply connection to the top where one would reasonably put the power supply as well. That's what remains.

hartytp, 2017-09-24:
Okay.

hartytp, 2018-01-11:
@jordens Do you still want these changes? If so, should we be making them for the DIO_BNC v1.2, etc boards?

gkasprow, 2018-01-11:
Sooner or later we will use backplane. I have an idea how to do it preserving all the modules with IDCs and changing only Kasli.

hartytp, 2018-01-11:
Sounds good. Not a high priority for now, but will be interested to hear what your plan is at some point...

jbqubit, 2018-01-11:
Created new wiki page for [EEM-v2](https://github.com/m-labs/sinara/wiki/EEM-v2) and moved working plans for backplane to that page from [EEM](https://github.com/m-labs/sinara/wiki/EEM).

jordens, 2018-01-12:
No. I don't want those changes for the modules we are doing right now.

closed 2018-06-06 by jordens