sinara-hw/sinara #324: wiki: merge PlannedSupportHW and CommunityHardware

jordens, 2017-09-24:
Could we merge the two pages about community non/rt/driver/monitoring hardware where designs at varying levels of completeness and production stages exist?
https://github.com/m-labs/sinara/wiki/CommunityHardware
https://github.com/m-labs/sinara/wiki/PlannedSupportHW

/cc @hartytp @npisenti

hartytp, 2017-09-24:
My motivation for keeping them separate was that the "planned support hardware" section was for hardware that I envisage being professionally designed, tested and documented (e.g. by WUT) and having supplied as a product by e.g. Creotech. In contrast, I see the "community hardware" as being more "here are some gerbers/eagle files for a project by one of our students. it seems to work, but you have to build it yourself and use at your own risk.". I can also imagine the "community hardware" section including real-time hardware, while the "support hardware" is explicitly non-real time.

I'm happy to make this distinction more clear on the Wiki or even rename the sections (or, merge if you'd strongly prefer that).

jordens, 2017-09-24:
Afaik the idea for the community hardware was explicitly that the designs should be 'taken up', redone, and produced by professionals as well. I didn't see the distinction.

hartytp, 2017-09-24:
> Afaik the idea for the community hardware was explicitly that the designs should be 'taken up', redone, and produced by professionals as well. I didn't see the distinction.

Okay, I hand't realised that. If there are plans to get these boards redone professionally then I'll move them to the support hardware page.

I'd still like to keep the community hardware page as a place for people to post links to hardware that is unlikely to be redone professionally though.

jordens, 2017-09-24:
Sure.

hartytp, 2017-09-24:
Okay, I've had a stab at doing this.

Comments:
- @jbqubit @npisenti I've tried to tidy these up a little bit. Can you have a quick look at what I've done and check you're happy with it.
- AFAICT, some of these projects might be better suited to being EEMs for RT control. @jbqubit @npisenti Feel free to move them to the PlannedEEM section if you'd like.

hartytp, 2017-09-24:
Also: if we decide that there is no realistic prospect of these being professionally redeveloped in the near future, then let's move them back to the community hardware section. (But, let's try to tidy them up before we do that, as I'd like to keep that section tidy rather than just being a brain dump!)

jbqubit, 2017-09-24:
@hartytp Thanks for taking a stab at getting this better organized. Currently there are separate pages for ARTIQ RTIO vs asynchronous and community design vs professional design and EEM vs uTCA. To my eye this way too many separate pages for future/planned hardware. As it stands it's difficult for a prospective Sinara user or contributor to get up to speed quickly on what hardware the community is working on. Its just too spread out. 

## Proposal 1
Consolidate the following pages into a single PlannedHardware page. 
- https://github.com/m-labs/sinara/wiki/uTCAPlanned
- https://github.com/m-labs/sinara/wiki/PlannedEEMs
- https://github.com/m-labs/sinara/wiki/PlannedSupportHW
- https://github.com/m-labs/sinara/wiki/CommunityHardware

## Proposal 2
Define some tags to make distinctions of the sort that @hartytp is after. "planned support hardware" vs  "community hardware" and RTIO vs "support hardware" and EEM vs uTCA. Annotate each entry on the PlannedHardware page using the tags. 

## Proposal 3
Some designs currently in the wiki sidebar are not yet funded. I propose to move designs that are not fully specified and funded to PlannedHardware . This makes it clear which components are solidly on track to being first class Sinara components. AFACT this includes the following:
- Shuttler: DAC
- Baikal: clock distribution module 
- Grabber: Camera Link
- Mirny: microwave synthesiser
- Stamper: TDC

hartytp, 2017-09-25:
> @hartytp Thanks for taking a stab at getting this better organized. Currently there are separate pages for ARTIQ RTIO vs asynchronous and community design vs professional design and EEM vs uTCA. To [...]

That's not quite the way I was looking at things: my thinking was that "community contributed hardware" is for PCBs that *already exist* and which are unlikely to be developed further. Thus, they can be used any time. They probably also aren't either EEMs or uTCA boards (just some random size/interface that their designer chose). In contrast, the planned EEMs section is either for boards that don't exist yet at all, or for boards which we plan to professionalise soon. This is a distinction I'd like to preserve.

For this reason, I'd prefer to keep "community contributed hardware" separate from "planned hardware". We can merge "planned EEMs", "planned uTCA" and 'planned support/non-real time" if you'd prefer (just have sub-headings for these categories within the page).

hartytp, 2017-09-25:
> Some designs currently in the wiki sidebar are not yet funded. I propose to move designs that are not fully specified and funded to PlannedHardware . This makes it clear which components are solidly on track to being first class Sinara components. AFACT this includes the following:

I'd like to keep the following PCBs on the sidebar now because of a combination of reasons (pick the appropriates ones for each board): the PCB is actually finished, even if it's not technically funded yet!; I expect it to be funded relatively soon, so not worth the effort of moving it; I want to keep it's visibility high to help remind people to fund it!

 - **Shuttler: DAC** 
- **Baikal: clock distribution module**
- **Grabber: Camera Link** 
- **Stamper: TDC**

Mirny probably won't be funded any time soon and is low priority, so sure, let's relegate it to the planning section.

jordens, 2017-09-26:
Keeping the visibility of projects that are "hot" in the funding/involvement sense (e.g. Shuttler/Baikal/Grabber and maybe Stamper) high is nice. 

Other than that I don't see the benefit nor the practicality in trying to draw a lines through the different properties of finished/proven/planned/prooposed/to-be-redesigned or EEM/non-EEM or Eurocard/non-Eurocard or uTCA/non-uTCA. Take Clocker which doesn't really fit into non-EEM/EEM or Eurocard/non-Eurocard. Or Sayma which can be used without a uTCA crate as well. Those are all just properties and metadata to me.

jordens, 2017-09-26:
Sidenote: I put a [Makefile](https://raw.githubusercontent.com/wiki/m-labs/sinara/Makefile) into the wiki repo with the commands to make PDFs from the markdown. That works as well as can be expected. This can be tweaked and extended at will.

hartytp, 2017-09-26:
Thanks for uploading that @jordens!

Okay, I've merged the pages and re-organised a bit. @jbqubit @jordens feel free to change things if you're still unhappy with the way I've organised things...

@jordens On a similar note, shall we move the NU-servo page to the NU-servo repository's wiki? Might make more sense to store it there and link to it from Sinara...

closed 2017-09-26 by hartytp

jordens, 2017-09-26:
I was planning on merging that page into the nu servo documentation when we merge nu servo into artiq.

hartytp, 2017-09-26:
Sounds like a good plan.

jbqubit, 2017-09-26:
@hartytp  Thanks for pulling together PlannedHardware. And thanks @jordens for the script.