sinara-hw/sinara #325: Urukul heat sink

jordens, 2017-09-24:
Back side
Exposed copper
See note in #291

dhslichter, 2017-09-25:
Solder mount?

gkasprow, 2017-09-25:
There are tiny copper heatsinks that can be placed by P&P machine and  reflow soldered. I will leave bottom part of thermal pad exposed. Anyway, one should not cover vias from one side because it creates pockets so would leave them open anyway.
So when needed, we can easily solder such heatsink.

dhslichter, 2017-09-25:
Are you not planning to use epoxy-filled vias under the DDS chips?  From what I understand, assembly and exposed paddle solder joint quality are considerably better if the vias are filled, rather than empty or soldermask-tented.

gkasprow, 2017-09-25:
We will use soldering paste on both sides and let it flow in and fill  vias entirely, It has much better characteristics than epoxy. Guys who do assembly know how to control the process to have the vias filled with tin..

gkasprow, 2017-10-09:
I added enlarged, exposed pad on bottom layer, covered with solder paste. This should effectively increase heat dissipation area. In case of troubles, we will solder copper pads that would further increase the area.

commit a18981ba (2017-10-09): fixed #325, #322, #319, #310, #309
closed 2017-10-09 by gkasprow