sinara-hw/sinara #326: Stamper specification

dhslichter, 2017-09-25:
NIST has some interest in an EEM TDC module (currently code-named Stamper), and I'd be interested in getting some draft specs based on potential use cases.  Some questions:

- Who else is interested in a TDC module currently?
- What sort of jitter/channel count/overall event rate is currently targeted?
- Are there preferences for an FPGA-based TDC vs. a commercial TDC ASIC with a Kasli as readout?  The latter would probably be considerably simpler/cheaper to implement.

cjbe, 2017-09-26:
We at Oxford are interested, but not pushing this yet.

For our use case, we need at least 2 channels. We don't have any strong event rate requirements, but we do want a low latency (<<100ns) way of determining that there was an event. For example, sending a copy of the buffered input signal down an LVDS pair to the master FPGA.

Our desired jitter is ideally ~<40ps FWHM, but we are happy to compromise on this.

We don't have any preference between commercial ASICs or FPGA based systems.

hartytp, 2017-09-26:
ACK. My guess that we'd go for a similar front panel to Urukul, giving 4 SMA inputs and 1 SMA clock input.

@dhslichter Are you considering funding this?

gkasprow, 2017-09-26:
As a reference, [here ](https://www.ohwr.org/projects/fmc-delay-1ns-8cha/wiki/wiki) is successful board I designed for CERN a few years ago. [Here](https://www.ohwr.org/documents/304) are some measurements.

dhslichter, 2017-09-27:
@gkasprow the board you have sent looks like the kind of thing I am imagining, at least for now.  In our case, it would be something simpler in various ways, since we'd just look for timestamping, and I think having buffered copies of the inputs sent via LVDS to the FPGA for the low-latency event trigger mentioned by @cjbe would be sensible.  There has since been a second generation of ACAM TDC chips with better performance as well, readout over SPI at ~1 Mevent per second or so.

dhslichter, 2017-09-27:
@hartytp this is still very much in the brainstorm phase out here, but funding might be forthcoming from NIST in the medium term, depending on how things prioritize here.

gkasprow, 2018-08-18:
This issue was moved to sinara-hw/meta#9

closed 2018-08-18 by gkasprow