sinara-hw/sinara #327: Sayma AMC - Version 1 - Ethernet

jbqubit, 2017-09-26:
This is a big enough topic it needs its own atomic Issue. Moving from #209. 
-------
@abyszuk said:

There are a few fixes which can applied to the ethernet PHY:
- [ ] connect JTRST_N to GND
- [ ] connect ALOS to GND
- [ ] pull GPO1 to GND
- [ ] CRS/COMMA is connected to MMC twice: (a) through MII_CRS line (b) via PHY_CFG_DDR line. Just pull it low and delete PHY_CFG_DDR wire.
- [ ] GTXCLK should be pulled low

But this just to get it to work reliably in MII mode (max 100 Mbps).
Schematics suggest that it's intended to work in RGMII mode when connected to FPGA. Following changes are required to enable RGMII mode:
- [ ] connect 125 MHz clock source to GTXCLK (maybe it's possible to use one of GPIO for this, if it's configured as clock output)

With current schematics ethernet will not work in RGMII mode, and will work in MII mode only if reconfigured via MDIO (because it starts in (R)GMII mode due to pin straps).

-----
@gkasprow replied

@abyszuk The GTXCLK fix was already applied to some boards. So it works in RGMII.
I didn't test it in MII, thanks for that. I simply shorted GTXCLK with TXCLK. Did you make it running in MII mode?
Is JTRST_N critical? It works without it. In next revision we can change, now would be difficult.
The same with ALOS. For some reason it works in RMII.
COnnection of CRS/COMMA to MMC via CFG was intentional, just to make sure it is properly configured when PHY is connected to FPGA.

abyszuk, 2017-09-26:
@gkasprow I'm not sure about JTRST_N/ALOS, this is simply what documentation says. Maybe it's minor bug, maybe it works by pure luck with current silicon revision. But it's always better to follow datasheet recommendation.
Shorting GTXCLK with TXCLK seems like a good enough fix for prototype board, but for next revision I'd recommend a more stable source.
I'll test the MII mode when I'll get a chance (maybe next week?). I'll need to push some real data there, as this chip can report LINK UP status even with completely defunct MII/RGMII side.

gkasprow, 2017-09-26:
@abyszuk I agree with you, just want to know which fix is really necessary for this revision. We have 10 bards that need fix and each needs some time...

abyszuk, 2017-09-26:
@gkasprow If RGMII works for you with current GTXCLK workaround, then I'd leave it like that.

jordens, 2017-09-28:
And on the board we had at JQI and on the three we have in HK the changes (gtxclock shorted to txclock and the others) are implemented?

gkasprow, 2017-09-28:
This is good question. Just check if there is a piece ow wire which shorts pins across the ETH PHY chip.

gkasprow, 2017-09-28:
Without this fix the PHY will work at 100Mbit.

sbourdeauducq, 2017-09-29:
Here is what I have on the three HK boards:
[image]

gkasprow, 2017-09-29:
OK, the fix is done then.

jbqubit, 2018-01-12:
@gkasprow can this be closed now?

hartytp, 2018-01-12:
@jbqubit Are the changes required to fix this for Sayma v2.0 recorded in an issue somewhere?

gkasprow, 2018-01-12:
@jbqubit @hartytp this issue is referenced in collective errata.

closed 2018-01-12 by gkasprow