sinara-hw/sinara #328: Upload release data for boards

jordens, 2017-09-27:
* Putting the PDFs, gerbers, drill, XY, BOM, renderings, and .cce files into the repository won't scale and will break down (we pointed this out multiple times before)
* Retrieving actual data files for a given board revision is hard and tedious.
* For example, I have no good way of locating/verifying components on the actual Sayma boards.
* Doing schematics and layout reviews becomes shooting a moving target.

The usual solution taking lessons from software development would be:

* remove all "output" or "generated" files from the repository
* determine and agree to proper versioning scheme and follow it through systematically.
* clean up version numbers in the [milestones](https://github.com/m-labs/sinara/milestone/6), and clean up the milestones
* add [tags](https://github.com/m-labs/sinara/tags) to the commits in the repository that correspond to manufactured board release (back in the days I added one for the original [version numbering scheme](https://github.com/m-labs/sinara/releases/tag/0.1rc1), but that was ignored)
* when there is a schematic review, or a layout review, or a production release, upload complete release data (PDFs, gerbers, drills, XY, BOM, renderings, photos, front panel layouts) to the [releases](https://github.com/m-labs/sinara/releases) that are generated from the [tags](https://github.com/m-labs/sinara/tags)
* do the same with schematic and layout reviews.

gkasprow, 2017-09-27:
We could follow OHWR approach where production and documentation files for certain revision are placed outside repo, in dedicated EDMS database with link to product Wiki. The idea with releases is similar and we should follow it. If you create proper layout, I will fill it with proper files for all designs.

gkasprow, 2017-09-27:
@jordens What is proposed release naming scheme? You wrote about it some time ago but I cannot find itd

jordens, 2017-09-28:
For the boards we have it seems to me that the following would be a versioning scheme that is both compatible with the existing versions and the typical "semantic versioning" that people use in software dev.

* **v1.0** is the first prototype
* **v1.1** is the first "production" run with minor tweaks
* **v2.0** would be the second (sometimes major) iteration of the design, potentially denoting only a prototype run.
* **v2.1**, **v2.2** etc would be production runs including minor tweaks.

Then I would propose that you determine which commits (git hashes) correspond to the versions of the boards that we have (Sayma RTM, AMC, Allaki, VHDCI, 3U*...) and that we (given that they are also in the same repository) add tags like:
* **Sayma_RTM_v1.0rc1** and **Kasli_v1.0rc1** for various iterations of schematic and board layout review and discussions (before and toward the corresponding v1.0 release).
* **Sayma_RTM_v1.0** and **3U_SMA_v1.1** etc. for production data.

* [ ] If you give me the mapping of the commits used for production and what they were used for, 
* [ ] I can add the tags (and delete the old one)
* [ ] Then you can [create releases from the tags](https://github.com/m-labs/sinara/tags) and upload the files to the releases
* [ ] After than you can start removing files.

hartytp, 2017-10-01:
@jordens Added this to the style guide section of the wiki so it's not forgotten about once this issue is closed. Feel free to edit/move as you see fit.

jordens, 2017-10-01:
Sure. This should ultimately find its way as a documented procedure into the wiki or the repo like we have for [ARTIQ](https://github.com/m-labs/artiq/blob/master/RELEASING.rst).

hartytp, 2017-10-04:
- [ ] All schematics should have a hardware version number on them somewhere prominent (either bottom right of each page, or on the top of the first page)!

hartytp, 2017-11-02:
What's the difference between PCB_3U_DDS_9910.pdf and the _a.pdf?

jordens, 2017-11-02:
I dropped channels 1-3 from the PDF to make the scrolling a bit easier.

gkasprow, 2017-11-27:
@hartytp they are already here https://github.com/m-labs/sinara/releases
Project.Outputs.for.PCB_3U_DDS.zip

hartytp, 2017-11-27:
@gkasprow True. That's a snapshot of the entire Sinara repository. I'd hoped for something a little more convenient, like a small zip file that only has the production files for Urukul. But, I guess one can dig what one needs from the Sinara snapshot.

cjbe, 2017-11-27:
@hartytp nope! From that link there is [Project.Outputs.for.PCB_3U_DDS](https://github.com/m-labs/sinara/releases/download/Urukul%2Fv1.0/Project.Outputs.for.PCB_3U_DDS.zip) which contains only gebers, drill files, and PnP files.

gkasprow, 2017-11-27:
@hartytp this is all you need to produce these boards :)

hartytp, 2017-11-27:
> @hartytp nope! From that link there is Project.Outputs.for.PCB_3U_DDS which contains only gebers, drill files, and PnP files.

Thanks, that's just what I was after.

gkasprow, 2017-11-27:
Would it make sense to add also schematic and PCB sources to releases? Theoretically we have them in repo at certain tag, but could be more convenient to have them in releases too.

hartytp, 2017-11-27:
> Would it make sense to add also schematic and PCB sources to releases? Theoretically we have them in repo at certain tag, but could be more convenient to have them in releases too.

I'm not so fussed about that. I'm okay with having the sources left as part of the repository snapshot, but it's nice to keep the production files in a separate zip file (as you have done).

hartytp, 2017-11-28:
Do these releases have the CAD files for the front panels? IMO they should.

jordens, 2017-11-28:
We recommend doing exactly what's described in the top post. Reiterating the warning we have raised so many times: keeping more and more generated binary data (PDFs for board layouts and schematics) in the repository will break and I don't see anybody who will clean that up.

jordens, 2017-11-28:
@gkasprow The releases are mis-named and/or mis-tagged. Please lease first decide on a single name accross the Technosystem part numbers, the wiki, the repository, the direcotry names, the issue labels, issue titles, file names. (is it "EEM_DIO_BNC", "DIO_BNC", "3U_DIO_BNC", "3U_BNC_DIO"?). (Sidenote regarding naming: unique opaque names would have been awesome here, right? Encoding metadata in the name fails once again...) Then make sure that the tag name is equal the title of the release (if the tag is "3U_SMA/1.1rc1" and the release name is "3U_SMA/1.1" nobody knows what's meant). Also make sure that this is actually the correct numbering (i.e. use the "v" prefix). There is a spurious "1.1rc1" tag now. Also make sure the version appears in the tarball filename. If you take a look at [the releases page](https://github.com/m-labs/sinara/releases) you'll see what I mean.

gkasprow, 2017-11-28:
@hartytp there are dxf files which are outputs. Sources are in pcbdoc.

hartytp, 2017-11-28:
> @hartytp there are dxf files which are outputs. Sources are in pcbdoc.

 IMHO, it's much easier if the all data for each PCB is in its own self-contained directory.

That means that, instead of putting all the panels in a single panels directory (and PDF), I'd prefer to put each panel DXF in the directory for that PCB. e.g. Urukul panel DXF in the Urukul directory. Then, put a PDF with just that panel in the Urukul release.

hartytp, 2017-11-28:
>  (is it "EEM_DIO_BNC", "DIO_BNC", "3U_DIO_BNC", "3U_BNC_DIO"?). (Sidenote regarding naming: unique opaque names would have been awesome here, right? Encoding metadata in the name fails once again...)

IIRC, you didn't want opaque names for the DIO breakout PCBs...

In any case, semi-opaque names like grabber are good IMHO. But -- not to reopen a can of worms -- each time I have a conversation with a newcomer to Sinara it goes a bit like this and I die a bit on the inside:

Me: So, we have these PCBs including Urukul the DDS, Zotino the DAC, etc...
Newcomer: wait a second, which board was which again? Uru? Uruk? Uro? Was that the DAC, right?
Me: No, Urukul is the DDS.
Newcomer: aah, so which was the DAC? Novo, right?
Me: No, that's the ADC, Zotino is the DAC.
Newcomer: OOh, right. So... why do these boards have such odd names? Can't we just call them the DAC? I'm going to call it the DAC from now on because I can't be bothered to remember that name.
Me: sigh

You might think that this is bad thinking and that opaque names based on Russian towns are the way forward, but the user reaction so far is pretty close to 100% negative no matter how many times I explain the names to people.

hartytp, 2017-11-28:
In any case, the argument in favor of opaque names seems even weaker given that we now have design variants; "Urukul" does not uniquely identify a product, "Urukul-AD9910" and "Urukul-AD9912" do. If appending the part number of a particular DDS IC to the name isn't encoding meta-data in the name, then what is?

dhslichter, 2017-11-29:
I think the ideal release format involves literally all of the pieces necessary (schematics source and PDF, layouts source and PDF, gerbers, BOM, xy, panel designs source and PDF) in a single folder per board.  You should be able to download one folder and have it contain everything that is required to order complete boards from your favorite vendor(s) already present, as well as the corresponding source files in case you want to make your own little tweaks and re-generate things.  The contents of this folder are generated when you make a release and then it remains completely static thereafter.  Tagging the appropriate commits in the repository for the source files as well gives an alternate way of chasing down source files, but it's more for our purposes than for someone external to the project who may not want to be bogged down trying to dig through the repository, thus the duplication in terms of having a copy of all relevant source files at the appropriate commit included in the release folder.

hartytp, 2017-11-29:
My feeling is that the only good names we've chosen so far are "grabber" and "clocker" because:
- Don't sound like something weird that you probably don't want in your lab ;)
- Easily pronounceable to anyone who speaks english
- Semi-opaque in the sense that the name hints at the function, making it memorable, without being overly prescriptive. In contrast, "shuttler" seems to be too prescriptive -- I can imagine plenty of uses for shuttler other than moving ions around, while (almost) all uses for clocker involve clocking things.

And, @dhslichter is right, I do find that these names make it more difficult to convince people to adopt the hardware (bad names can sink products). Some thoughts:
- I always thought that "phaser" was a cool name, and would be a much better name for "Sayma"
- I also liked the "ingester" suggestion someone bounced around for an ADC, but am mentally reserving it for a fast ADC that deserves such a grandiose title.
- I half thought that "Sampler" could be a good name for Novo. Okay, it breaks the NU-servo name, but I'm not too fussed about that; it won't be NU for that long anyway + we can call it SU-servo for the Super Urukul Servo.

If we went down that path then we start to have more consistency in the names, which is nice and makes the whole project seem cohesive.

hartytp, 2017-11-29:
@jordens is right that we should make the names for the DIO boards consistent. I'd suggest DIO-SMA, DIO-BNC, DIO-RJ45. Unless anyone can come up with an "er" name to replace DIO that is ;)

jordens, 2017-11-29:
> IIRC, you didn't want opaque names for the DIO breakout PCBs...

You are misquoting me. I said non-opaque names are ok IMHO in this case. I said that because I had the hope that despite the metadata in the names, **the names could be used consistently**. But you'll agree that this is not the case and that it is failing in spectacular ways. You even [concluded and summarized the/a discussion of those names](https://github.com/m-labs/sinara/issues/254#issuecomment-326800570) with a name that was exactly used once: in that conclusion. So much for names that have **any** identifying value.
Or are you in fact pointing out that we should have instead insisted on unique and opaque names?

> Newcomer: wait a second, which board was which again? Uru? Uruk? Uro? Was that the DAC, right?

The quotes that you use in your argument and that Daniel so enthusiastically applauds is amusing but irrelevant. The only thing you provide anecdotal evidence for is that said anonymous "Newcomer" has imperfect associative memory. That's a very common trait likely shared by a large fraction of users (myself included). But you again forget that opaque names are not meant to solve the problem of imperfect associative memory. They are meant to solve the problem of providing unique, distinguishable, and reliable identifiers for certain things. Afterwards they can be tuned for recognizability (like the semi-opaque names).

Nobody is claiming that opaque names are perfect. You'd have to argue that metadata-names are significantly more memorable. I have seen zero evidence for that and have pointed out evidence to the contrary above.
Do you have any evidence that EEM_DIO_BNC as an identifier is more memorable than Urukul? Or any evidence that the distinction between EEM_DIO_BNC and 3U_SMA is clearer than the distinction between Urukul and Novogorny?
Looking through my E-Mails and this project alone I can find plenty of evidence that metadata names have yielded more problems. And in my discussions with Sinara users this impression is being reinforced regularly. I can supply you with plenty of anecdotes about metadata names if that is the currency you want to trade in.

Both your arguments about the EEM_DIO_* and Urukul-* patterns are textbook *non sequitur*. The fact that a scheme was not applied to some problem has zero implications for scheme's applicability or validity. You would be in a problematic rhetoric spot because you lobbied against applying the scheme but are now apparently blaming the scheme (or me) for not being applied.
For variants it can be OK to sub-type. Urukul designates that "class of Urukul boards on the same PCB". How else would you do it? Do you have a different proposal for designating population variants (with/without transformer X, resistor bridge Y placed or not, connector Z mounted or not...) in mind or are you just throwing dirt?

I just had to write this.

hartytp, 2017-11-29:
@jordens I think the upshot is that semi-opaque names will make everyone happy -- or close enough to stop complaining publicly. What about remaining Novo to Sampler and Sayma to Phaser?

jordens, 2017-11-29:
Please only for a **major** revision of Sayma (e.g. mechanically completely incompatible with the current mezzanines, different DACs, ADCs). And please never reuse names, i.e. veto Phaser.
I won't hold a Novo->Sampler rename against you but I am also not particularly exited about it.

hartytp, 2017-11-29:
> Please only for a major revision of Sayma (e.g. mechanically completely incompatible with the current mezzanines, different DACs, ADCs).

Well, I think the next revision of Sayma will be fairly major: e.g. there is little chance that it will be mechanically compatible with the current mezzanines due to the issues that Joe and others highlighted; may have different RF connectors etc.

You want to veto Phaser because it currently refers to a piece of gateware? Isn't that gateware about to be obsoleted anyway? In anycase, I'd argue that having Phaser gateware as the gateware that powers Phaser hardware isn't such a bad thing...

@dhslichter @cjbe @klickverbot @gkasprow @jbqubit @dtcallcock or anyone else who cares, happy with the rename to Sampler on the grounds that we agree never to rename this board again?

jordens, 2017-11-29:
Just to set the record straight: those issues (forces, interfering constraints, alignment) were pointed out last year already.
Yes. No matter how obsolete phaser is, please don't recycle or reappropriate those names. See also the RFCs on naming and [RFC2100](http://www.faqs.org/rfcs/rfc2100.html) for a lyric response to your screenplay above.

hartytp, 2017-11-29:
> Just to set the record straight: those issues (forces, interfering constraints, alignment) were pointed out last year already.

Sure, I remember that. Anyway, seems like we might have a better solution now.

Getting wildly off-topic for this issue, but I like having the AFEs on removable mezzanines so long as we can make them cheap enough and solve the mechanical issues. IIRC you were against having the AFE on a mezzanine. Is that correct? If so, would you still object if we get the cost down and solve the mechanical issues?

jordens, 2017-11-29:
Sure. If cost, mechanics, time, and risk are resolved then this would be a no-brainer. I think that can be decided based on numbers. Get a price for the different approaches and batch sizes that would be involved. Plus factors for the different development/debugging/manufacturing/testing time, and risk. We seem to have enough past data from Sayma for time and risk. And to be accurate this is can only be advice and no objection.

jbqubit, 2017-11-30:
>  No matter how obsolete phaser is, please don't recycle or reappropriate those names. 

I agree with Robert on this. Just too confusing. 

> Edit: if you feel strongly about this, we can keep thinking and see if we come up with something better before finalising the designs for Sayma v2

Please create a new Issue if there's interest in discussing. Right now I'd prefer we all focus on getting Sayma v1 working... 

> Getting wildly off-topic for this issue, but I like having the AFEs on removable mezzanines so long as we can make them cheap enough and solve the mechanical issues. IIRC you were against having the [...]

Please create a new Issue for that discussion. Things are already far enough afield from the Issue's stated topic.

gkasprow, 2018-04-17:
https://github.com/sinara-hw/Urukul/releases

closed 2018-04-17 by gkasprow