sinara-hw/sinara #329: Novo v2.0

hartytp, 2017-10-01:
Plan for Novo v2.0, designed to work with the NU-servo Kasli gateware. Main change is allowing *simultaneous sampling* of all ADC channels using a fast, source-synchronous multi-lane serial interface.

Novo 2.0 supports two interface modes:
"Basic" interface mode
- uses a single EEM connector (EEM0) and standard SPI gateware.
- ADCs must be read sequentially in increasing order
- Max sample rate depends on length of cabling etc. With a 20 MHz SPI read clock (25ns round trip plus setup), we get one sample one each channel per: 30ns tCONVH + 450ns tCONV + 8 channels * 16 bits * 50ns = 6.9µs

"Fast" interface mode
- Two EEM connectors and custom gateware for multi-lane source-synchronous serial ADC interface. Data sampled on ADC_CLKOUT using ADC_SDOA to ADC_SDOD. 1.5MSPS on all channels should be possible (200MHz clock).
- Used by Kasli laser servo

- [ ] Fix all v1.0 issues #220 
- Keep Novo v1.0 AFE unchanged (connectors, input filter and InAmp)
- [ ] Replace ADC with LTC2320-16 8
- [ ] Use external reference with ADC, chosen to have better stability than ADC
- [ ] Divide InAmp outputs from +-20V to +-2.5V. Divider's accuracy should be 0.1% or better, TCR should be chosen to avoid degrading the ADC's +-15ppm/K accuracy. Either use precision resistors or a cheap divider, such as AORN1002AT3 (£2.50 in quantity from Farnel). Shunt filter capacitor in parallel with the divider to give a second pole at 1MHz (it's 250kHz current Novo v1.0 design).
- [ ] Offset InAmp outputs by +2.5V (Vref/2) using a cheap divider (2 resistors + filter capacitor, OpAmp buffer if required) and the InAmp ref input. Voltage range after divider is now 0V to +5V to match ADC inputs.
- [ ] Drive ADCs in pseudo-differential configuration as in datasheet figure 4, using the ADCs CMRR to subtract Vref/2 from the signal. Filter pole here should be set to 1MHz. Choose a suitable OpAmp for this: cheap and low power consumption; slew rate should be at least 20V/us / 4=5V/us to avoid degrading the InAmp slew rate; GBP >= 1MHz; +-12V supply. FWIW, when I looked at this the best choice seemed to be the OPA2197 (same used in Zotino).
- [ ] Ensure that the ADC is driven by a suitably low impedance (don't want accuracy to depend on measurement duty cycle!).
- [ ] Confirm that power sequencing is okay (e.g. do we need to do anything about power sequencing here to ensure that the ADC input doesn't exceed Vdd during startup (e.g. if a signal is connected when the device is powered on))?
- Two EEM connectors as follows:

Pin | EEM0 | EEM1
0 | ADC_SCK | ADC_CLKOUT
1 |  NC | ADC_SDOB
2 |  ADC_SDOA | ADC_SDOC
3 | ADC_CNV | ADC_SDOD
4 | SR_SCK | NC
5 | SR_MOSI | NC
6 | SR_MISO | NC
7 | SR_CS | NC

hartytp, 2017-10-01:
We won't start this for a while, as there is currently a backlog of work to be done, but I'd ideally like to agree on what needs to be done for planning purposes. Ideally, I'd like to agree on a specification freeze for this version pretty soon.

@gkasprow @jordens Are you still happy with the above plan for Novo v2.0? In particular, @gkasprow can you double check that there isn't a better ADC (e.g. lower temp co etc) that we could use for this? Given our requirements, I don't think there is, but I'd like to confirm that.

jordens, 2017-10-01:
All good from our (digital) end. I didn't review the analog side though.

gkasprow, 2017-11-14:
@hartytp shall I update existing project or create a new one?

hartytp, 2017-11-14:
Update the existing project; this is a v2, not a new project and it should obsolete the old v1.

The advantage of git is that the old one is still there if we ever want to go back to it.

gkasprow, 2017-11-15:
@hartytp I'm drawing schematic and have several questions:
- Is DIP switch for SDR/DDR option OK?
- To shift the output of AD8253 by 2.5V and preserve +/- 10V range, I'd need to increase the power supply by roughly 2V. Another issue is division of such shifted voltage by 4 because such divider should be connected to stable 2.5V potential. IMHO it's better to divide it by 4 and then add the offset at the ADC driver stage.
- the datasheet says about re-timing of CNV input signal, there is also D flip-flop on reference design. Look at the schematic on p.32. This is necessary when external clock is not in sync with CONV signal. In our case I'd skip it.
- Linear uses LT1819CMS8 as a driver. It is 1.5$ chip, 400MHz, 2500V/μs, 9mA. 
OPA2197 costs the same , consumes much less power and has better offset. But it has 10MHz GPW, 20V/us.
These amps can be supplied with lower voltage (+/-6V) so are not that critical in terms of power consumption
- with pseudo-differential driving, only half of ADC codes are available. Since we have double amplifier anyway, it's better to use differential driving and utilise full ADC scale.

Below is proposed AFE
[image]

hartytp, 2017-11-15:
@gkasprow Thanks for doing that! That's a major improvement on my original design suggestion.

Comments in posts to follow...

hartytp, 2017-11-15:
In the proposed AFE, the reference current draw for IC2A depends on the input voltage, which can lead to worse cross-talk between channels etc (the reference has to source current to 8 channels, which is quite a few). It might be better to connect the reference voltage to a high-impedance node or to do the summing at a summing junction (virtual ground). Maybe the easiest way of doing this is to configure IC2A as an inverting amplifier with gain of 1/4 (e.g. 2*10k=20k between the InAmp and virtual ground, 10k/2=5k between the OpAmp output and virtual ground, feel free to use a resistor network with higher-valued resistors to reduce current draw), then connect the +10/3V reference to the positive terminal. This inverts the polarity of the circuit, which can be compensated by swapping the + and - pins of the InAmps. What do you think (please do check my maths on this, as this was very much pen-paper-and-beer circuit design)?

hartytp, 2017-11-15:
> Is DIP switch for SDR/DDR option OK?

DIP switch is fine (what's the alternative?).

hartytp, 2017-11-15:
> To shift the output of AD8253 by 2.5V and preserve +/- 10V range, I'd need to increase the power supply by roughly 2V. Another issue is division of such shifted voltage by 4 because such divider [...]

You're right. Good catch!

hartytp, 2017-11-15:
> the datasheet says about re-timing of CNV input signal, there is also D flip-flop on reference design. Look at the schematic on p.32. This is necessary when external clock is not in sync with CONV signal. In our case I'd skip it.

Seems fine to me.

hartytp, 2017-11-15:
> with pseudo-differential driving, only half of ADC codes are available. Since we have double amplifier anyway, it's better to use differential driving and utilise full ADC scale. This of course needs precise resistors in the driver. There is cap missing, it's only a sketch.

Good point. Differential is better.

hartytp, 2017-11-15:
> Linear uses LT1819CMS8 as a driver. It is 1.5$ chip, 400MHz, 2500V/μs, 9mA.
> OPA2197 costs the same , consumes much less power and has better offset. But it has 10MHz GPW, 20V/us.
> These amps can be supplied with lower voltage (+/-6V) so are not that critical in terms of power consumption

Let's go for the OPA2197:
- It's nice to use the same OpAmp as Zotino; we already have experience of it, so no surprises lurking hopefully, and this allows us to get better bulk-buy discounts when we manufacture Zotino and Novo at the same time.
- As you say, the offset is better and the power consumption is lower
- 10MHz GBP is fine: we use this in G=1, so 3dB BW is 10MHz, which is large compared with the 0.25MHz input filter
- Max input signal amplitude for this OpAmp is 2.5V, max signal frequency is 0.25MHz, so max signal slew rate is 2.5V * 2*pi*0.25e6/s = 4V/us, so 20V/us is fine. (NB InAmp slew rate is 20V/us, but signal is 4 times bigger, so the InAmp is the limitation here).

You can introduce a +-6V supply for these OpAmps if you want, but I'm not sure the power dissipation due to these amplifiers is sufficient to be worth the hassle/cost of more supply rails. I'll leave the decision to you; I'm happy either way.

hartytp, 2017-11-15:
Did you check for other ADCs which we could use for this design (should be 8 channel, with a sensible interface). I'm happy with the one we have, but it's probably worth double checking that there isn't a better alternative.

hartytp, 2017-11-15:
Since the ADC runs from a different supply to the OpAmps/InAmps, do we need protection diodes to prevent it from being overdriven transiently during startup (which can cause damage/lockup)? Probably it's worth at least having pads for diodes.

hartytp, 2017-11-15:
This ADC can only run with digital supplies of up to 3V, can we use a 2.5V logic supply for the LVDS interface as well? This is something I want to do for the other EEMs to save power.

hartytp, 2017-11-15:
> Is DIP switch for SDR/DDR option OK?

Ignore my previous comment on this.

AFACIT, no DIP switch is needed to change between *fast* and *basic* modes; both use SDR, the only difference is in the software (@jordens correct me if I'm wrong about that).

@jordens Is it worth changing the connector pin-out so that SODA is on both EEM0 and EEM1. That way in fast mode we have CLK_OUT as well as SDOA-D on the same cable and we don't need to worry about length matching between the EEMs.

hartytp, 2017-11-15:
That's me done!

Thanks again for the work on this @gkasprow

gkasprow, 2017-11-15:
@hartytp Below is improved AFE schematic
[image]

But there is nasty thing above 1MHz
This can be partially mitigated by other configuration of ADC filter
[image]

I want to run ADC from 2.5V, anyway, LVDS mode requires that.
I used quad matched reistors AORN1001AT3, 2.5EUR/piece. So this gives 5EUR/channel. Tracking tempco is 5ppm

Below is comparison of 16 bit ADCs

[ADC selection.xlsx](https://github.com/m-labs/sinara/files/1474179/ADC.selection.xlsx)

jordens, 2017-11-15:
Yes. The plan is to use LVDS mode (no LVDS-CMOS converters on SDOx, CLKOUT, SCK, CONV). Please pin-swap the signals so that you need the minimum amount of those quad LVDS line drivers.

NU-Servo uses DDR with a 125 MHz SCK (250 Mb/s/lane). It could be made SDR (250 MHz SCK) but we'd need another MMCM/PLL output. But that's really fast and I am worried about getting that over the IDCs. Also it precludes using an LVDS transcievers on the clock lines (SCK and CLKOUT, maybe relevant for FMC-LVDS-VHDCI). I'd like to use one of the unused pins on EEM1 (let's say EEM1.7P) to decide over DDR/SDR with the well-defined default (when there is not EEM connector plugged in) being SDR. Just for kicks do the same with the CMOS/LVDS pin (put it on EEM1.7N). And for both pins don't bother with the LVDS-CMOS converter as those can be just single ended EEM signals (use EEM1.7P and EEM1.7N for them).

We could also consider using the same technique (single ended EEM lines) on the PGIA SR. Then we have LVDS pairs for SCK,4xSDOx,CLKOUT, a single LVDS-CMOS converter for CNV, and 2 single-ended PGIA SR pins. We could even squeze everything onto one EEM connector (CLKOUT,SCK,SDOA,SDOB,SDOC,SDOD,CNV,SR-SDI/SR-SCK). We wouldn't bother with SR-RCLK and just tie that to CNV (i.e. have the SR latch with each CNV. It's a bit weird but I don't think that will be a problem). Then I'd like the CMOS/LVDS and DDR/SDR selection to be handled with a DIP switch.

Putting SDOA both on EEM0 and EEM1 is probably not easy due to termination. I'd rather take the risk of having to match IDC cable lenghts. But if you can replicate SDOA on EEM1 without significant additional cost, complexity, delays, power consumption, risks, I am fine with it.

gkasprow, 2017-11-15:
@jordens I'm affraid of running SPI signals over diff line as single ended.
Certainly there won't be any problems between Kasli and Novo, but once somebody uses 2m VHDCI cable, the capacitive coupling would be an issue. I'd have to use clock filtering and recovery to get rid of spikes.
LVDS has one advantage - it is compatible with 3.3, 2.5 and 1.8V systems.
If you make it LVCMOS, then issues will arrive. Metlino is 1.8V, Kasli is 2.5V, you won't be able to use FMC-VHDCI adapter.
Don't go this way. For PGIA it's safer to use I2C if we want to limit the number of wires.

jordens, 2017-11-15:
@gkasprow Ack, let's not do it. I am also very hesitant with single ended signals over the IDC connectors or VHDCI cables.
Doing I2C for the PGIAs would work and be nice to save wires and EEM connectors. But we have zero gateware or software support for the I2C tree. And it would make the PGIA settings decidedly non-RT. It would preclude switching gains mid-experiment with controlled timing. Right now we can't do that until we have the I2C tree gateware and software support funded.

gkasprow, 2017-11-15:
Another aspect of on-fly selection between SDR/DDR is that there is no indication in the datasheet when configuration pins are sampled. It probably happens during power on where Power On Reset event takes place. It means that this pin has to be stable during power on which is not the case in Kasli.

gkasprow, 2017-11-15:
@jordens I wouldn't be affraid of running 125MHz signal over EEM connector.

jordens, 2017-11-15:
Could be. But my guess is that this is not sampled but just controls the output SRs directly. Maybe with a bunch of [these](http://www.ralf-hildebrandt.de/publication/pdf_dff/pde_dff.pdf).
125 MHz is fine. That's what I want to do. But we are talking about a 250 MHz SDR clock.

gkasprow, 2017-11-15:
People use these ribbon cables for PCIe so I wouldn't expect troubles at 250MHz neither. But there can be an issue with length matching. If we want to route ADC-SDOA to both EEM connectors, some LVDS buffers would be needed. I plan to add them anyway to isolate the ADC from FPGA and cabling. The noise picked by the ADC output interface will be visible in the signal spectra. And the LVD IFC delivers 300mV@100Ohm so it means that has only 3mA of current which is not much and may cause troubles with longer cables.

jordens, 2017-11-15:
Ok. That sounds reasonable. Then LVDS buffers on SCK,SDOx,CLKOUT, CMOS-LVDS converter on CNV, SDOA on both EEM, and a DIP switch for DDR/SDR. The matching between CLKOUT and SDOx is the only important one. The EEM signals are EEM0: [SCK,CNV,SDOA,nc,PGIA-CLK,PGIA-LATCH,PGIA-MOSI,PGIA-MISO], EEM1: [CLKOUT,SDOA,SDOB,SDOC,SDOD,nc,nc,nc].

hartytp, 2017-11-15:
@gkasprow Thanks for posting the updated AFE design.

With the designs you posted, where are the resistor ladders? AFAICT, you can't implement that using a standard network of 4 identical resistors -- you need 4R for the InAmp voltage, R for the reference and R for the feedback, so 6 resistors which must be in the same resistor network package (not spread between two separate resistor networks or they won't track). Or, am I missing something? This is the reason I suggested applying the reference voltage to the positive input terminal.

hartytp, 2017-11-15:
On the plots you posted, what are the red and green curves? The + and - ADC inputs? If so, can you plot the differential signal instead, since that's what we care about.

I'm happy with you to configure the ADC filter however you think is best. From the plots, it's not obvious which configuration is best to me: we don't have any signal content at 1MHz, so I'd go with whichever gives less peaking in the differential gain.

hartytp, 2017-11-15:
> Below is comparison of 16 bit ADCs

Thanks for posting that. From your table, there are no other ADCs which offer 1MSPS simultaneously on 8 channels, so it looks like we should stick with the ADC we have.

hartytp, 2017-11-15:
Thanks @jordens @gkasprow for the discussion of the digital side. I agree with everything you've said:

- Let's only use LVDS for communication
- Let's not use I2C for the PGIA gain, as I want this to be real-time settable
- Let's mirror SDOA onto EEM1 to avoid having to think about cable length matching (make sure to delay match SDOA-D and the CLK_OUT pins)
- 2V5 LVDS buffers sound like a good idea (can we use 1V8 for this?)
- DIP switch for SDR/DDR sounds like a good idea.

hartytp, 2017-11-15:
@gkasprow I think we've more or less settled all design questions for the schematic. Can you have a go at doing a complete schematic so we can do a final design review. Hopefully we can get this process done quickly.

gkasprow, 2017-11-15:
@hartytp this design can be implemented using 2 quad resistor ladders with 2x10k + 2x 20k values.
With the first one I get 20+20 = 40k and 10  + 10 for first amplifier, for second amplifier I simply use 10 + 10 k.
 It's better to go for lower resistors, let's say 1 and 2k to reduce parasitic time constant.

hartytp, 2017-11-15:
> @hartytp this design can be implemented using 2 quad resistor ladders with 2x10k + 2x 20k values.

Aah, okay. I was assuming that most cheap resistor ladders have all four resistors with the same value. But if you have a R-R-2R-2R part then your way is better.

gkasprow, 2017-11-15:
[AORN2-1AT3](http://datasheet.octopart.com/AORN2-1AT3-Vishay-datasheet-43370278.pdf) does the job. It's 3EUR@100pcs.

hartytp, 2017-11-15:
> It's better to go for lower resistors, let's say 1 and 2k to reduce parasitic time constant.

If you want to decrease the resistance, that's fine by me, but make sure you check that the OpAmps are happy driving the load (and that self-heating won't cause too much of a problem!).

FWIW, I'm not convinced that lowering the impedances to 1k/2k is necessary. The design you have has summing junctions, so the capacitances are bootstrapped out. The OpAmp BW isn't that high, so the bootstrapping not may be great at higher frequencies, but I'd guess it will work fine up to our filter cut-off of 250kHz (the spice model should confirm this).

OpAmp capacitance is 6.4pF, so cutoff frequency (ignoring bootsrapping) with 40k is: fc=1/(2 x pi x 6.4e-12 x 40e3) = 620kHz. Probably that's high enough that we can ignore it given that our AFE filter cutoff frequency is 250kHz.

hartytp, 2017-11-15:
> AORN2-1AT3 does the job. It's 3EUR@100pcs.

Nice!

gkasprow, 2017-11-15:
@hartytp I managed to find only [one](http://www.ti.com/lit/ds/symlink/sn65lvds20.pdf) LVDS repeater that works with 2.5V. I checked DIgikey, Mouser, TI, Onsemi, Maxim. It seems that there is really no alternative. Standard supply is 3.3V.
In case of the translators, standard [ones ](http://www.ti.com/lit/ds/symlink/sn65lvds2.pdf) work with 2.5V.

hartytp, 2017-11-15:
> @hartytp I managed to find only one LVDS repeater that works with 2.5V. I checked DIgikey, Mouser, TI, Onsemi, Maxim. It seems that there is really no alternative. Standard supply is 3.3V.
> In case of the translators, standard ones work with 2.5V.

Okay, so we can run from 2.5V using a translator, but there is no easy way of running from 1V8.

Let's go with 2V5 and the translators then.

gkasprow, 2017-11-15:
@hartytp I mean that there is no way to run LVDS with 1.8V.
The translators are needed for SPI only

hartytp, 2017-11-15:
> @hartytp I mean that there is no way to run LVDS with 1.8V.

Aaah, okay. Let's stick with 2V5 then.

gkasprow, 2017-11-16:
@hartytp @jordens schematics are ready for design review.
The last thing that needs to be discussed is protection of the ADC input against too high voltage from the driver which is quite powerful (65mA).

hartytp, 2017-11-16:
@gkasprow Thank you for doing this so fast!

Let's give people 24hours to submit comments/changes to the schematic before we freeze it. @cjbe @klickverbot @jordens

closed 2017-11-16 by hartytp

jordens, 2017-11-16:
I filed my issues. Digital interface looks ok for nu-servo.

gkasprow, 2017-11-16:
@hartytp I updated the schematics.